margelo/react-native-vision-camera #2918: 🔧 A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction

dipo1, 2024-05-27:
### How were you trying to build the app?

React Native Version: 0.74.1
Kotlin Version: 1.9.22

`.\gradlew assembleRelease --debug`

### Full build logs

```tsx
$ .\gradlew assembleRelease --debug
...
2024-05-27T05:17:52.028+0100 [DEBUG] [org.gradle.internal.operations.DefaultBuildOperationRunner] Build operation 'Cross-configure project :gradle-plugin' started
2024-05-27T05:17:52.030+0100 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Execution worker: acquired lock on worker lease
2024-05-27T05:17:52.030+0100 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Execution worker: released lock on worker lease
2024-05-27T05:17:52.031+0100 [DEBUG] [org.gradle.internal.operations.DefaultBuildOperationRunner] Completing Build operation 'Cross-configure project :gradle-plugin'
2024-05-27T05:17:52.031+0100 [DEBUG] [org.gradle.internal.operations.DefaultBuildOperationRunner] Build operation 'Cross-configure project :gradle-plugin' completed
2024-05-27T05:17:52.031+0100 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Daemon worker Thread 5: released lock on state of build :gradle-plugin
2024-05-27T05:17:52.031+0100 [DEBUG] [org.gradle.internal.operations.DefaultBuildOperationRunner] Completing Build operation 'Execute 'allprojects {}' action'
2024-05-27T05:17:52.031+0100 [DEBUG] [org.gradle.internal.operations.DefaultBuildOperationRunner] Build operation 'Execute 'allprojects {}' action' completed
2024-05-27T05:17:52.032+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.BuildFlowService$Inject] [KOTLIN] Initialize BuildFlowService$Inject
2024-05-27T05:17:52.032+0100 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Execution worker Thread 3: acquired lock on worker lease
2024-05-27T05:17:52.032+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Executing [class org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler.reportGlobalMetrics]
2024-05-27T05:17:52.033+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Executing [class org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler.reportBuildFinish]
2024-05-27T05:17:52.036+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Executing [report metric STATISTICS_COLLECT_METRICS_OVERHEAD]
2024-05-27T05:17:52.036+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Executing [class org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler.reportGlobalMetrics]
2024-05-27T05:17:52.037+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Executing [class org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler.reportBuildFinish]
2024-05-27T05:17:52.037+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Executing [report metric STATISTICS_COLLECT_METRICS_OVERHEAD]
2024-05-27T05:17:52.037+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Executing [class org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler.buildFinished]
2024-05-27T05:17:52.038+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Executing [class org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler.buildFinished]
2024-05-27T05:17:52.038+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.BuildFlowService$Inject] [KOTLIN] Close BuildFlowService$Inject
2024-05-27T05:17:52.033+0100 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Execution worker Thread 3: released lock on worker lease
2024-05-27T05:17:52.039+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.KotlinGradleBuildServices$Inject] [KOTLIN] Disposed KotlinGradleBuildServices
2024-05-27T05:17:52.040+0100 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Execution worker: acquired lock on worker lease
2024-05-27T05:17:52.040+0100 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Execution worker: released lock on worker lease
2024-05-27T05:17:52.041+0100 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Execution worker Thread 2: acquired lock on worker lease
2024-05-27T05:17:52.041+0100 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Execution worker Thread 2: released lock on worker lease
2024-05-27T05:17:52.042+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.BuildFlowService$Inject] [KOTLIN] Initialize BuildFlowService$Inject
2024-05-27T05:17:52.043+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Executing [report metric STATISTICS_COLLECT_METRICS_OVERHEAD]
2024-05-27T05:17:52.043+0100 [INFO] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] Could not execute [report metric STATISTICS_COLLECT_METRICS_OVERHEAD]
2024-05-27T05:17:52.043+0100 [DEBUG] [org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler] org.jetbrains.kotlin.gradle.plugin.statistics:type=StatsService,name=v2
javax.management.InstanceNotFoundException: org.jetbrains.kotlin.gradle.plugin.statistics:type=StatsService,name=v2
        at java.management/com.sun.jmx.interceptor.DefaultMBeanServerInterceptor.getMBean(DefaultMBeanServerInterceptor.java:1088)
        at java.management/com.sun.jmx.interceptor.DefaultMBeanServerInterceptor.invoke(DefaultMBeanServerInterceptor.java:811)
        at java.management/com.sun.jmx.mbeanserver.JmxMBeanServer.invoke(JmxMBeanServer.java:802)
        at org.jetbrains.kotlin.gradle.plugin.statistics.JMXKotlinBuildStatsService.callJmx(KotlinBuildStatsService.kt:207)
        at org.jetbrains.kotlin.gradle.plugin.statistics.JMXKotlinBuildStatsService.access$callJmx(KotlinBuildStatsService.kt:203)
        at org.jetbrains.kotlin.gradle.plugin.statistics.JMXKotlinBuildStatsService$report$2.invoke(KotlinBuildStatsService.kt:222)
        at org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatHandler$Companion.runSafe$kotlin_gradle_plugin_common(KotlinBuildStatHandler.kt:33)
        at org.jetbrains.kotlin.gradle.plugin.statistics.JMXKotlinBuildStatsService.report(KotlinBuildStatsService.kt:221)
        at org.jetbrains.kotlin.statistics.metrics.IStatisticsValuesConsumer$DefaultImpls.report$default(StatisticsValues.kt:11)
        at org.jetbrains.kotlin.gradle.plugin.statistics.KotlinBuildStatsService$Companion.applyIfInitialised(KotlinBuildStatsService.kt:161)
        at org.jetbrains.kotlin.gradle.plugin.statistics.BuildFlowService.recordBuildFinished$kotlin_gradle_plugin_common(BuildFlowService.kt:115)
        at org.jetbrains.kotlin.gradle.plugin.BuildFinishFlowAction.execute(StatisticsBuildFlowManager.kt:75)
        at org.jetbrains.kotlin.gradle.plugin.BuildFinishFlowAction.execute(StatisticsBuildFlowManager.kt:61)
        at org.gradle.configurationcache.flow.FlowScheduler.schedule(FlowScheduler.kt:48)
        at org.gradle.configurationcache.flow.BuildFlowScope.schedulePendingActions(BuildFlowScope.kt:140)
        at org.gradle.configurationcache.flow.BuildFlowScope.buildFinished(BuildFlowScope.kt:118)
        at jdk.internal.reflect.GeneratedMethodAccessor29.invoke(Unknown Source)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:568)
        at org.gradle.internal.dispatch.ReflectionDispatch.dispatch(ReflectionDispatch.java:36)
        at org.gradle.internal.dispatch.ReflectionDispatch.dispatch(ReflectionDispatch.java:24)
        at org.gradle.internal.event.DefaultListenerManager$ListenerDetails.dispatch(DefaultListenerManager.java:472)
        at org.gradle.internal.event.DefaultListenerManager$ListenerDetails.dispatch(DefaultListenerManager.java:454)
        at org.gradle.internal.event.AbstractBroadcastDispatch.dispatch(AbstractBroadcastDispatch.java:83)
        at org.gradle.internal.event.AbstractBroadcastDispatch.dispatch(AbstractBroadcastDispatch.java:69)
        at org.gradle.internal.event.DefaultListenerManager$EventBroadcast$ListenerDispatch.dispatch(DefaultListenerManager.java:443)
        at org.gradle.internal.event.DefaultListenerManager$EventBroadcast$ListenerDispatch.dispatch(DefaultListenerManager.java:431)
        at org.gradle.internal.event.AbstractBroadcastDispatch.dispatch(AbstractBroadcastDispatch.java:43)
[... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

yossularko, 2024-05-28:
same problem with:

- "react": "18.2.0"
- "react-native": "0.74.1"
- "react-native-vision-camera": "^4.0.5"

innocodes, 2024-05-29:
Facing the same challenge. My setup is identical to @yossularko.

psparago, 2024-05-29:
Same issue I think.

expo eas android build failed

expo SDK 51 
react-native 0.74.1 
react 18.2.0 
react-native-vision-camera ^4.0.5

This built with expo SDK 50 yesterday, moved to expo SDK 51 today and it failed. iOS builds properly on both SDKs

FAILURE: Build failed with an exception.
* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details
* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.
BUILD FAILED in 9m 38s

aGORyan, 2024-05-29:
react: 18.3.1
react-native: 0.74.1
react-native-vision-camera: "^4.0.5"

Using Expo/EAS, migrating from expo-camera because FaceDetection no longer supported.

Trying to avoid downgrading react-native version, though I see the other open issue suggested it as a work-around.

`* What went wrong:
Execution failed for task ':react-native-vision-camera:compileReleaseKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction`

BunnyTechnologies, 2024-05-30:
Same Issue 

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

"dependencies": {
    "@notifee/react-native": "^7.8.2",
    "@react-native-async-storage/async-storage": "^1.23.1",
    "@react-native-community/datetimepicker": "^8.0.1",
    "@react-native-voice/voice": "^3.2.4",
    "@react-navigation/bottom-tabs": "^6.5.20",
    "@react-navigation/native": "^6.1.17",
    "@react-navigation/native-stack": "^6.9.26",
    "@react-navigation/stack": "^6.3.29",
    "axios": "^1.6.8",
    "react": "18.2.0",
    "react-native": "0.74.1",
    "react-native-geolocation-service": "^5.3.1",
    "react-native-gesture-handler": "^2.16.2",
    "react-native-linear-gradient": "^2.8.3",
    "react-native-modal": "^13.0.1",
    "react-native-permissions": "^4.1.5",
    "react-native-safe-area-context": "^4.10.1",
    "react-native-screens": "^3.31.1",
    "react-native-svg": "^15.3.0",
    "react-native-vector-icons": "^10.1.0",
    "react-native-vision-camera": "^4.0.5"
  },

Prakhar-Agarwal-byte, 2024-05-30:
Same issue with RN 0.74.1

aGORyan, 2024-05-30:
> react-native-vision-camera does not support react-native 0.74
> 
> [#2904 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/2904#issuecomment-2133304515)

Thank you!

Do you know if there is plan to support 74+ in the future?

RenaudAubert, 2024-05-30:
@aGORyan I'm sorry I don't know. You'd have to ask @mrousavy which is the creator and maintainer of this package. As per his comment you can sponsor this request if you want it sooner.

dipo1, 2024-05-30:
> [2614#issuecomment-2105995734](https://github.com/mrousavy/react-native-vision-camera/issues/2614#issuecomment-2105995734)

^This solution seems to resolve the issue. The build was successful at least.

Just slipped in `@OptIn(FrameworkAPI::class)` on Line 17 and, of course, imported the `FrameworkAPI`.

`node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt`

```
package com.mrousavy.camera.frameprocessors
...
import com.facebook.react.common.annotations.FrameworkAPI//<--import here
...

@OptIn(FrameworkAPI::class)//<--insert here
@Suppress("KotlinJniMissingFunction") // we use fbjni.
class VisionCameraProxy(private val reactContext: ReactApplicationContext) {
...
}
```

Hopefully, we don't have to do this in the next release.

Youssef0603, 2024-06-04:
@dipo1 thanks man it works for me

MrShahzeb-kresus, 2024-06-08:
"react": "18.2.0",
    "react-native": "0.74.1",
    "react-native-vision-camera": "^4.1.0"
    
Adding FrameworkAPI worked for me too

sudeshvs, 2024-09-19:
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-screens:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction

can someOne help me here 

im using 
 "react": "^17.0.2",
 "react-native": "0.65.3",
"react-native-screens": "3.14.0",

Iuriy-Budnikov, 2025-12-19:
0.73 still has this issue