margelo/react-native-vision-camera #2921: 💭 Is there a migration guide from V3 to V4?

tomerh2001, 2024-05-28:
### Question

I'm having difficulties migrating from V3 to V4. The build seems to fail when I enable frame processors, but it works when I disable them.

I don't want to spam the issues here since I'm confident it's a cumulative result of the multiple problems already discussed here. 

I do want to ask if there are any must-dos when upgrading from V3 to V4 that I may have missed, which may cause my errors. Is there an upgrade guide?

For example, is `react-native-skia` or `react-native-reanimated` required? Cause I don't use them, and installing them is a different kind of rabbit hole for my app.

Thank you.

### VisionCamera Version

v4.0.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

tomerh2001, 2024-05-28:
The specific issue I'm having when enabling frame processors:
```
> Task :prepareKotlinBuildScriptModel UP-TO-DATE
C/C++: VisionCamera: Frame Processors: ON!
C/C++: VisionCamera: Linking react-native-worklets...
C/C++: CMake Error in CMakeLists.txt:
C/C++:   Imported target "react-native-worklets-core::rnworklets" includes
C/C++:   non-existent path
C/C++:     "/Users/chagitcharazi/WorkSpace/gama-workspace/projects/gamapaymobile/node_modules/react-native-worklets-core/android/build/headers/rnworklets"
C/C++:   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
C/C++:   * The path was deleted, renamed, or moved to another location.
C/C++:   * An install or uninstall procedure did not complete successfully.
C/C++:   * The installation package was faulty and references files it does not
C/C++:   provide.
C/C++: CMake Generate step failed.  Build files cannot be regenerated correctly.
```

Versions:
```
    "@react-native-community/cli-platform-android": "^12.3.6",
    "react-native": "0.73.6",
    "react-native-vision-camera": "4.0.5",
    "react-native-worklets-core": "^1.3.3",
    "@ismaelmoreiraa/vision-camera-ocr": "^3.0.1",
    "vision-camera-resize-plugin": "^3.1.0",
```

Non of the troubleshooting steps I found (github or official docs) helped

mrousavy, 2024-05-29:
Only react-native-worklets-core is required for Frame Processors.

Try to completely nuke your cache (delete the entire android/ and node_modules/ folder) and try again. That should fix it.

closed 2024-05-29 by mrousavy