margelo/react-native-vision-camera #2922: 🔧 :react-native-vision-camera:compileDebugKotlin FAILED

kssgarcia, 2024-05-30:
### How were you trying to build the app?

# Build Error with react-native-vision-camera on Expo

## Description

I am trying to use the `react-native-vision-camera` package to build a real-time detection app with Expo. However, I encounter build errors that prevent the app from compiling successfully.

## Steps to Reproduce

1. Create a new Expo project:
    ```sh
    npx create-expo-app@latest MyNewApp --template
    cd MyNewApp
    ```

2. Install `react-native-vision-camera` package:
    ```sh
    npm install react-native-vision-camera
    ```

3. Attempt to build the project:
    ```sh
    npx expo run:android
    ```

## Expected Behavior

The project should build successfully and allow me to use the `react-native-vision-camera` package for real-time detection.

## Additional Information

I have tried changing the React Native version and the `react-native-vision-camera` version. I also created a new Expo project and installed the camera package, but I always get the same error.

Any help to resolve this issue would be greatly appreciated.

Thank you!

### Full build logs

```tsx
› Building app...
Configuration on demand is an incubating feature.

> Configure project :expo

Using expo modules
  - expo-asset (10.0.6)
  - expo-constants (16.0.2)
  - expo-file-system (17.0.1)
  - expo-font (12.0.6)
  - expo-keep-awake (13.0.2)
  - expo-modules-core (1.12.12)
  - expo-splash-screen (0.27.4)
  - expo-system-ui (3.0.4)
  - expo-web-browser (13.0.3)

> Configure project :react-native-reanimated
Android gradle plugin: 8.2.1
Gradle: 8.6

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ??
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: http://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at C:\Users\santy\Downloads\myapp\node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core not found, Frame Processors are disabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!

> Task :react-native-vision-camera:compileDebugKotlin FAILED                                                                                                                         
e: file:///C:/Users/santy/Downloads/myapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:31:79 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///C:/Users/santy/Downloads/myapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:36:19 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///C:/Users/santy/Downloads/myapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:36:47 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///C:/Users/santy/Downloads/myapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:75:73 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 47s
308 actionable tasks: 25 executed, 17 from cache, 266 up-to-date
Error: C:\Users\santy\Downloads\myapp\android\gradlew.bat app:assembleDebug -x lint -x test --configure-on-demand --build-cache -PreactNativeDevServerPort=8081 -PreactNativeArchitectures=arm64-v8a exited with non-zero code: 1
Error: C:\Users\santy\Downloads\myapp\android\gradlew.bat app:assembleDebug -x lint -x test --configure-on-demand --build-cache -PreactNativeDevServerPort=8081 -PreactNativeArchitectures=arm64-v8a exited with non-zero code: 1
    at ChildProcess.completionListener (C:\Users\santy\Downloads\myapp\node_modules\@expo\spawn-async\build\spawnAsync.js:42:23)
    at Object.onceWrapper (node:events:632:26)
    at ChildProcess.emit (node:events:517:28)
    at cp.emit (C:\Users\santy\Downloads\myapp\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (node:internal/child_process:1098:16)
    at ChildProcess._handle.onexit (node:internal/child_process:303:5)
    ...
    at spawnAsync (C:\Users\santy\Downloads\myapp\node_modules\@expo\spawn-async\build\spawnAsync.js:7:23)
    at spawnGradleAsync (C:\Users\santy\Downloads\myapp\node_modules\@expo\cli\build\src\start\platforms\android\gradle.js:96:48)
    at assembleAsync (C:\Users\santy\Downloads\myapp\node_modules\@expo\cli\build\src\start\platforms\android\gradle.js:74:18)
    at runAndroidAsync (C:\Users\santy\Downloads\myapp\node_modules\@expo\cli\build\src\run\android\runAndroidAsync.js:45:37)
```

### Project dependencies

```json
"dependencies": {
    "@expo/vector-icons": "^14.0.0",
    "@react-navigation/native": "^6.0.2",
    "expo": "~51.0.9",
    "expo-constants": "~16.0.2",
    "expo-font": "~12.0.6",
    "expo-linking": "~6.3.1",
    "expo-router": "~3.5.14",
    "expo-splash-screen": "~0.27.4",
    "expo-status-bar": "~1.12.1",
    "expo-system-ui": "~3.0.4",
    "expo-web-browser": "~13.0.3",
    "react": "18.2.0",
    "react-dom": "18.2.0",
    "react-native": "^0.74.1",
    "react-native-gesture-handler": "~2.16.1",
    "react-native-reanimated": "~3.10.1",
    "react-native-safe-area-context": "4.10.1",
    "react-native-screens": "3.31.1",
    "react-native-vision-camera": "^4.0.5",
    "react-native-web": "~0.19.10"
  },
```

### VisionCamera Version

4.0.5

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

stewartiee, 2024-05-30:
I get the same here, fresh Expo project with latest version. Tried using previous versions also, but those all result in build errors.

kssgarcia, 2024-05-30:
Ty, that work for me

closed 2024-05-30 by kssgarcia