margelo/react-native-vision-camera #2923: 🐛

P0diceps, 2024-05-31:
### What's happening?

I want to use the CodeScanner with the torch already activated but it doesn't work.
If I use a button to turn the torch on while using the Codescanner then it works but I would like to start the scanner with the torch already on.

### Reproduceable Code

```tsx
<Camera
        key={device.id}
        isActive={isActive}        
        torch="on"
        device={device}
        codeScanner={codeScanner}
        style={StyleSheet.absoluteFillObject}
        enableZoomGesture
      />
```

### Relevant log output

```shell
.
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 4,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "wide-angle-camera",
    "ultra-wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": true,
  "minFocusDistance": 10,
  "minExposure": -4,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Huawei P30 Pro

### VisionCamera Version

4.0.5

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-05-31:
Use titles. Without a title an issue is invalid.

mrousavy, 2024-05-31:
Also I'm not sure why this is not obvious, but for such thingsg I absolutely need logs. If you just ignore that and put a `.` there, I'll ignore the issue and close it.

closed 2024-05-31 by mrousavy

P0diceps, 2024-05-31:
I updated the title and the log output. Would be happy if you could open the ticket again.