margelo/react-native-vision-camera #2927: 🔧

nolife08021, 2024-05-31:
### How were you trying to build the app?

i tried to build an face detect app, but facing error after install "vision-camera-face-detector". I had done install react-native-vision-camera and react-native-worklets-core.
try install "vision-camera-trustee-face-detector-v3" also facing same issue.

### Full build logs

```tsx
/Users/wlm/rn/face-app/node_modules/vision-camera-face-detector/android/src/main/java/com/visioncamerafacedetector/VisionCameraFaceDetectorPlugin.java:29: error: package com.mrousavy.camera.frameprocessor does not exist
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;
                                         ^
/Users/wlm/rn/face-app/node_modules/vision-camera-face-detector/android/src/main/java/com/visioncamerafacedetector/VisionCameraFaceDetectorPlugin.java:35: error: cannot find symbol
public class VisionCameraFaceDetectorPlugin extends FrameProcessorPlugin {
                                                    ^
  symbol: class FrameProcessorPlugin
/Users/wlm/rn/face-app/node_modules/vision-camera-face-detector/android/src/main/java/com/visioncamerafacedetector/VisionCameraFaceDetectorPluginPackage.java:10: error: package com.mrousavy.camera.frameprocessor does not exist
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;
                                         ^
/Users/wlm/rn/face-app/node_modules/vision-camera-face-detector/android/src/main/java/com/visioncamerafacedetector/VisionCameraFaceDetectorPlugin.java:134: error: method does not override or implement a method from a supertype
  @Override
  ^
/Users/wlm/rn/face-app/node_modules/vision-camera-face-detector/android/src/main/java/com/visioncamerafacedetector/VisionCameraFaceDetectorPlugin.java:141: error: cannot access Detector
      Task<List<Face>> task = faceDetector.process(image);
                                          ^
  class file for com.google.mlkit.vision.common.internal.Detector not found
/Users/wlm/rn/face-app/node_modules/vision-camera-face-detector/android/src/main/java/com/visioncamerafacedetector/VisionCameraFaceDetectorPluginPackage.java:19: error: cannot find symbol
    FrameProcessorPlugin.register(new VisionCameraFaceDetectorPlugin());
    ^
  symbol:   variable FrameProcessorPlugin
  location: class VisionCameraFaceDetectorPluginPackage
6 errors
```

### Project dependencies

```json
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.71.17",
    "react-native-vision-camera": "^4.0.5",
    "react-native-worklets-core": "^1.3.3",
    "vision-camera-face-detector": "^0.1.8"
  },
```

### VisionCamera Version

4.0.5

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-06-03:
this is related to the plugin. the plugin needs to update for latest V4. it's just a small change.

also, use titles for issues. without titles your issues just get closed.

closed 2024-06-03 by mrousavy

nolife08021, 2024-06-04:
hi, sorry for the empty title, I tried downgraded vision-camera to v3.7.1 but still issue. 
```
/Users/wlm/rn/face-app/node_modules/vision-camera-trustee-face-detector-v3/android/src/main/java/com/visioncamerafacedetector/VisionCameraFaceDetectorPlugin.java:12: error: package com.mrousavy.camera.frameprocessors does not exist
import com.mrousavy.camera.frameprocessors.Frame;
                                          ^
```

Do you have any documentation about how to use V4 to do for face detection/ recognition?

ibrahim-developes, 2024-06-11:
@nolife08021  did it work?

nolife08021, 2024-06-13:
@ibrahim-developes nope, for android i use back react-native-camera, but having linking issue on ios.

dev-hk22, 2025-07-25:
2025-07-25 13:11:39.851 25014-25014 crash_dump64            crash_dump64                         E  failed to get the guest state header for thread 24967: Bad address
2025-07-25 13:11:39.912 25014-25014 crash_dump64            crash_dump64                         I  obtaining output fd from tombstoned, type: kDebuggerdTombstoneProto
2025-07-25 13:11:39.923 25014-25014 crash_dump64            crash_dump64                         I  performing dump of process 22601 (target tid = 23238)
---------------------------- PROCESS STARTED (25020) for package com.pythonsdk ----------------------------
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  Build fingerprint: 'Nothing/TetrisIND/Tetris:15/AP3A.240905.015.A2/2506092111:user/release-keys'
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  Revision: '0'
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  ABI: 'arm64'
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  Timestamp: 2025-07-25 13:11:39.941850448+0530
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  Process uptime: 607s
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  Cmdline: com.pythonsdk
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  pid: 22601, tid: 23238, name: mqt_js  >>> com.pythonsdk <<<
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  uid: 10505
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
2025-07-25 13:11:42.241 25014-25014 DEBUG                   crash_dump64                         A  Abort message: 'JNI DETECTED ERROR IN APPLICATION: JNI GetObjectRefType called with pending exception java.lang.IllegalStateException: register failed, the sensor listeners size has exceeded the maximum limit 128
                                                                                                      at boolean android.hardware.SystemSensorManager.registerListenerImpl(android.hardware.SensorEventListener, android.hardware.Sensor, int, android.os.Handler, int, int) (SystemSensorManager.java:289)
                                                                                                      at boolean android.hardware.SensorManager.registerListener(android.hardware.SensorEventListener, android.hardware.Sensor, int, android.os.Handler) (SensorManager.java:871)
                                                                                                      at boolean android.hardware.SensorManager.registerListener(android.hardware.SensorEventListener, android.hardware.Sensor, int) (SensorManager.java:778)
                                                                                                      at void android.view.OrientationEventListener.enable() (OrientationEventListener.java:93)
                                                                                                      at void com.visioncamerafacedetector.VisionCameraFaceDetectorOrientation.startDeviceOrientationListener() (VisionCameraFaceDetectorOrientation.kt:39)
                                                                                                      at void com.visioncamerafacedetector.VisionCameraFaceDetectorOrientation.<init>(android.content.Context) (VisionCameraFaceDetectorOrientation.kt:24)
                                                                                                      at void com.visioncamerafacedetector.VisionCameraFaceDetectorPlugin.<init>(com.mrousavy.camera.frameprocessors.VisionCameraProxy, java.util.Map) (VisionCameraFaceDetectorPlugin.kt:35)
                                                                                                      at com.mrousavy.camera.frameprocessors.FrameProcessorPlugin com.visioncamerafacedetector.VisionCameraFaceDetectorPluginPackage._init_$lambda$0(com.mrousavy.camera.frameprocessors.VisionCameraProxy, java.util.Map) (VisionCameraFaceDetectorPluginPackage.kt:13)
                                                                                                      at com.mrousavy.camera.frameprocessors.FrameProcessorPlugin com.visioncamerafacedetector.VisionCameraFaceDetectorPluginPackage.$r8$lambda$m06YTD3kKVG0x0AAsMhXLidaNGQ(com.mrousavy.camera.frameprocessors.VisionCameraProxy, java.util.Map) (VisionCameraFaceDetectorPluginPackage.kt:-1)
                                                                                                      at com.mrousavy.camera.frameprocessors.FrameProcessorPlugin com.visioncamerafacedetector.VisionCameraFaceDetectorPluginPackage$$ExternalSyntheticLambda0.initializePlugin(com.mrousavy.camera.frameprocessors.VisionCameraProxy, java.util.Map) (D8$$SyntheticClass:0)
[... cut ...]
                                                                                                    
                                                                                                        in call to GetObjectRefType
                                                                                                        from void com.facebook.jni.NativeRunnable.run()'

Im getting this issue while face detection it works 1-3 time fine then app get crashed on debug and in apk app shows black screen or blank

please help me to fix this issue.