margelo/react-native-vision-camera #2928: Torch does not work on CodeScanner🐛

P0diceps, 2024-05-31:
### What's happening?

I want to use the CodeScanner with the torch already activated but it doesn't work.
If I use a button to turn the torch on while using the Codescanner then it works but I would like to start the scanner with the torch already on.

### Reproduceable Code

```tsx
<Camera
        key={device.id}
        isActive={isActive}        
        torch="on"
        device={device}
        codeScanner={codeScanner}
        style={StyleSheet.absoluteFillObject}
        enableZoomGesture
      />
```

### Relevant log output

```shell
05-31 12:12:46.049 12943 12943 D BoundBrokerSvc: onUnbind: Intent { act=com.google.android.gms.inappreach.service.START dat=chimera-action: cmp=com.google.android.gms/.chimera.GmsApiService }
05-31 12:12:46.272  3444  3928 I WifiClientModeImpl: fetchRssiLinkSpeedAndFrequencyNative rssi=-63 TxLinkspeed=72 freq=2437 RxLinkSpeed=72
05-31 12:12:46.292  4283  4283 D CompatibilityChangeReporter: Compat change id reported: 150939131; UID 10134; state: DISABLED
05-31 12:12:46.292  4283  4283 D RegisterPublishData: in OnChange, selfChange = false
05-31 12:12:46.299  4283  4283 D RegisterPublishData: Publishing data for the uri
05-31 12:12:46.358  4283  4283 D DcaPublisher: Successfully registered and inserted the data
05-31 12:12:46.358  4283  4283 D RegisterPublishData: in OnChange, selfChange = false       
05-31 12:12:46.360  4283  4283 D RegisterPublishData: Publishing data for the uri
05-31 12:12:46.379  4283  4283 D DcaPublisher: Successfully registered and inserted the data
05-31 12:12:46.379  4283  4283 D RegisterPublishData: in OnChange, selfChange = false       
05-31 12:12:46.380  4283  4283 D RegisterPublishData: Publishing data for the uri
05-31 12:12:46.393  4283  4283 D DcaPublisher: Successfully registered and inserted the data
05-31 12:12:46.393  4283  4283 D RegisterPublishData: in OnChange, selfChange = false       
05-31 12:12:46.394  4283  4283 D RegisterPublishData: Publishing data for the uri
05-31 12:12:46.409  4283  4283 D DcaPublisher: Successfully registered and inserted the data
05-31 12:12:46.617  3444  4348 D BatteryService: level : 80 / status : 2/Health : 2/Decommission : 0
05-31 12:12:46.622  4508  4508 D QtiCarrierConfigHelper: WARNING, no carrier configs on phone Id: 0
05-31 12:12:46.625  4283  4283 D com.symbol.dataanalytics.a.a.d.b: Received battery event
05-31 12:12:46.626  4283  4283 D com.symbol.dataanalytics.a.a.d.b: BatteryFactory event action = android.intent.action.BATTERY_CHANGED
05-31 12:12:46.629 21999 21999 D HighTemperatureReceiver: Receive action: android.intent.action.BATTERY_CHANGED 
05-31 12:12:46.633  5503  5503 I OSUService: handle_ab_resume -1 3
05-31 12:12:46.633  5503  5503 I OSUService: handle_ab_resume state 1 last_state 0
05-31 12:12:46.633  5503  5503 E OSUService: Can't resume - there are no update in suspended state
05-31 12:12:46.633  3999  3999 D KeyguardUpdateMonitor: received broadcast android.intent.action.BATTERY_CHANGED
05-31 12:12:46.649  3999  3999 D KeyguardUpdateMonitor: handleBatteryUpdate
05-31 12:12:46.683 27691 27691 I CameraSession: Camera Lifecycle changed to CREATED!
05-31 12:12:46.685 27691 27691 I CameraView: Updating CameraSession...
05-31 12:12:46.708 27691 27691 I CameraView: Updating CameraSession...
05-31 12:12:46.709 27691 27691 I CameraSession: configure { ... }: Waiting for lock...
05-31 12:12:46.709 27691 27691 I CameraView: A new configure { ... } call arrived, aborting this one...
05-31 12:12:46.712 27691 27691 I CameraSession: configure { ... }: Waiting for lock...
05-31 12:12:46.712 27691 27691 I CameraView: A new configure { ... } call arrived, aborting this one...
05-31 12:12:46.712 27691 27691 I CameraSession: configure { ... }: Waiting for lock...
05-31 12:12:46.713 27691 27691 I CameraSession: configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true, locationChanged=true)
05-31 12:12:46.713 27691 27691 I CameraSession: Creating new Outputs for Camera #0...
05-31 12:12:46.713 27691 27691 I CameraSession: Using FPS Range: null
05-31 12:12:46.713 27691 27691 I CameraSession: Creating Preview output...
05-31 12:12:46.714 27691 27691 I CameraSession: Creating CodeScanner output...
05-31 12:12:46.715 27691 27691 I CameraSession: Successfully created new Outputs for Camera #0!
05-31 12:12:46.715 27691 27691 I CameraSession: Binding Camera #0...
05-31 12:12:46.716 27691 27691 I CameraSession: Binding 2 use-cases...
05-31 12:12:46.721 27691 27691 D CameraOrientationUtil: getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
05-31 12:12:46.722 27691 27691 D CameraOrientationUtil: getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
05-31 12:12:46.724 27691 31902 D TransportRuntime.SQLiteEventStore: Storing event with priority=DEFAULT, name=FIREBASE_ML_SDK for destination cct
05-31 12:12:46.727 27691 27691 D CameraOrientationUtil: getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
05-31 12:12:46.729 27691 27691 D DynamicRangeResolver: Resolved dynamic range for use case androidx.camera.core.Preview-21363c61-42e2-4b74-b37a-e11a098734e8 to no compatible HDR dynamic ranges.
05-31 12:12:46.729 27691 27691 D DynamicRangeResolver: DynamicRange@1862b5b{encoding=UNSPECIFIED, bitDepth=0}
05-31 12:12:46.729 27691 27691 D DynamicRangeResolver: ->
[... 412 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Huawei P30 Pro

### VisionCamera Version

4.0.5

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-06-13:
Can you try if this also happens in the Example app here?

P0diceps, 2024-06-18:
I tried it but the example app crashes instantly on multiple devices

` BUNDLE  ./index.js

 LOG  Running "VisionCameraExample" with {"rootTag":11}
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 LOG  Re-rendering Navigator. Camera: denied | Microphone: denied
 LOG  Requesting camera permission...
 LOG  Camera permission status: granted
 LOG  Requesting microphone permission...
 LOG  Microphone permission status: granted
 LOG  Camera: 0 (BACK) androidx.camera.camera2 | Format: (3840x2160 photo / 1920x1080@30 video @ 30fps)
 LOG  Camera initialized!
 LOG  Camera started!
 LOG  Preview orientation changed to portrait!
 LOG  Output orientation changed to portrait!
 LOG  Camera stopped!
 LOG  56856238571394: 1280x720 yuv Frame (landscape-right)
 LOG  Camera started!
 LOG  Preview started!
 BUNDLE  ./index.js

 LOG  Running "VisionCameraExample" with {"rootTag":11}
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 LOG  Re-rendering Navigator. Camera: granted | Microphone: granted
 LOG  Camera: 0 (BACK) androidx.camera.camera2 | Format: (3840x2160 photo / 1920x1080@30 video @ 30fps)
 LOG  Camera: 0 (BACK) androidx.camera.camera2 | Format: (3840x2160 photo / 1920x1080@30 video @ 30fps)
 LOG  Camera initialized!
 LOG  Preview orientation changed to portrait!
 LOG  Output orientation changed to portrait!
 LOG  Camera started!
 LOG  Preview started!
 LOG  56888207548108: 1280x720 yuv Frame (landscape-right)`

mrousavy, 2024-06-18:
format your logs correctly. Three backticks (```), otherwise I cannot help you.

P0diceps, 2024-06-18:
Done

mrousavy, 2024-07-04:
Hi, no I did not look into this.

mrousavy, 2024-07-12:
> I tried it but the example app crashes instantly on multiple devices.

@P0diceps I fixed this error in the example app. Can you try this again? in example

P0diceps, 2024-07-15:
@mrousavy 
Yeah the example app works now.
The issue with the torch is still the same. I set it to 'on' but the torch stays off when starting the QR-Code Scanner.

'''07-15 07:44:39.769  1041  2241 I HwPerformEx: HwPerformEx(0xee0900c0), camera 0.
07-15 07:44:39.769  1041  2241 I HwFoldDisplayModeEx: HwFoldDisplayModeEx(0xee108098).
07-15 07:44:39.769  1041  2241 I CameraDeviceClient: CameraDeviceClient 0: Opened
07-15 07:44:39.769  1041  2241 I CameraService: connectHelper(), make client for 'com.mrousavy.camera.example'.
07-15 07:44:39.769  2698  3552 I PGServer: report state:10019 event type:1 pid:0 uid:0 pkg:com.mrousavy.camera.example to pid: 2698
07-15 07:44:39.769  1073  1073 I SensorHub: Activate, handle: 5, enabled:0, err: 0
07-15 07:44:39.770  1547  2301 I SensorService: disableSensor handle:5, packageName:hidl_client_pid_1059
07-15 07:44:39.770  1547  2301 I SensorService: disable activate end
07-15 07:44:39.770  1547  2301 I SensorService: disable end
07-15 07:44:39.770  1547  1687 I SensorService: disable begin
07-15 07:44:39.770  1547  1687 I SensorService: disable activate begin
07-15 07:44:39.770  1073  1073 I Sensors@1.2-impl: activate-2 enabled-0 fwkOals-1 camOals-1 fwkOcolor-0 camOcolor-1 fwkOtof-0 camOtof-0
07-15 07:44:39.770  1073  1073 I Sensors@1.2-impl: IsOpenNeedUpdate: service: 1547, sensor: 2, open: 0, openCnt: 1
07-15 07:44:39.770  1073  1073 I Sensors@1.2-impl: SetTotalOpenCnt: sensor: 2 has been close, erase it's map
07-15 07:44:39.770  1073  1073 I Sensors@1.2-impl: EraseSensorConfig: sensor: 2
07-15 07:44:39.770  1073  1073 I Sensors@1.2-impl: EraseSensorConfig: erase the config of sensorId: 2
07-15 07:44:39.770  1073  1073 I Sensors@1.2-impl: SetTotalOpenCnt: erase the config of sensorId: 2
07-15 07:44:39.771  1044  1151 D dubaid  : [BufferedLogServer.cpp] ReceiveMessage# Received buffered log from dubai_driver-0
07-15 07:44:39.771  1073  1073 I SensorHub: Activate, handle: 2, enabled:0, err: 0
07-15 07:44:39.771  1044  1151 D dubaid  : [BufferedLogServer.cpp] ReceiveMessage# Received buffered log from dubai_driver-0
07-15 07:44:39.771  1547  1687 I SensorService: disableSensor handle:2, packageName:hidl_client_pid_1059
07-15 07:44:39.771  1041  2241 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=0
07-15 07:44:39.771  1547  1687 I SensorService: disable activate end
07-15 07:44:39.771  1547  1687 I SensorService: disable end
07-15 07:44:39.771  1073  1073 I Sensors@1.2-impl: alsActivate-3 enabled-0 fwkOals-1 camOals-1
07-15 07:44:39.771  1073  1073 I Sensors@1.2-impl: IsOpenNeedUpdate: service: 1059, sensor: 3, open: 0, openCnt: 2
07-15 07:44:39.771  1073  1073 I Sensors@1.2-impl: GetSensorConfig: sensor: 3, Can't get param
07-15 07:44:39.771  1073  1073 I Sensors@1.2-impl: UpdateSensorConfig: sensor: 3, optimize period 0
07-15 07:44:39.771  1041  2241 V Camera3-Device: initialize: Initializing HIDL device for camera 0
07-15 07:44:39.771  1059  1860 I FlickerSensorImpl: disableSensor: sensor_handle:700, enabled:false, sampling_period_ns:500000
07-15 07:44:39.772  1073  1073 I Sensors@1.2-impl: extendSensorActivate
07-15 07:44:39.772  1547  2758 D HwCameraServiceProxy: It is not the fold camera device.
07-15 07:44:39.772  1547  2758 I HwCameraServiceProxy: notifyCameraStateChangeForMediaAssistant cameraId 0, newCameraState 0, facing 0
07-15 07:44:39.772  1059  6754 I ServerTranslator: open: enter
07-15 07:44:39.772  1059  6754 I CfgServer: open: open device (/dev/video22)
07-15 07:44:39.772  1059  6754 D ImagingSystem: cancelPrelaunchCheckPoint(2904): enter. instance[0x766a359000].
07-15 07:44:39.772  1059  6754 D HwaPerfRecorderImpl: [HWA_CAM3]enter perfInitialize(), id(0):rear camera
07-15 07:44:39.772  1059  6754 I CPipelineTranslator: CPipelineTranslator: ctor instance[0x76d169b218].
07-15 07:44:39.772  2517  4279 I HwMediaAssistantService: notifyCameraStateChange
07-15 07:44:39.772  2517  4279 I HwMediaAssistantService: sendCameraChangeBroadcast to com.huawei.harmonyos.foundation cameraId(0), facing(0), newCameraState(0), clientName(com.mrousavy.camera.example)
07-15 07:44:39.765  1547  1730 I PowerManagerService: handleSandman: wakefulness: 1,startDreaming: false
07-15 07:44:39.772  2517  4279 W ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:1183 android.content.ContextWrapper.sendBroadcast:453 com.huawei.systemserver.mediaassist.HwMediaAssistantService.sendCameraChangeBroadcast:518 com.huawei.systemserver.mediaassist.HwMediaAssistantService.notifyCameraStateChange:246 com.huawei.systemserver.mediaassist.IMediaAssist$Stub.onTransact:67
07-15 07:44:39.773  1059  6754 I VirCam_StreamChange: OnSessionChangedEvent: receive camera session changed event = 0
07-15 07:44:39.773  1041  2241 I CameraProviderManager: Camera device device@3.4/legacy/0 torch status is now NOT_AVAILABLE
07-15 07:44:39.773  1041  2241 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=0
07-15 07:44:39.773  1547  1547 I HwDualSensorEventListenerImpl: onTorchModeUnavailable, mFlashlightMode = HwDualSensorData.FlashlightMode.CAMERA_MODE
07-15 07:44:39.773  2517  4279 E HwMediaAssistantService: ignore camera state 0
07-15 07:44:39.773  2517  2707 I OllieMsgCenter: publishEvt=2130903200
07-15 07:44:39.773  1547  2758 I HwCameraServiceProxy: notifyCameraStateChangeForRotate begin cameraId 0, newCameraState 0, facing 0
[... cut ...]
07-15 07:44:40.292  1059 30876 D OisMgr  : [dc23155_64D] enter onStart()
07-15 07:44:40.292  1059 30845 E HwaUtils: [dc23155_64E] getAttributeValue() mFirstLevelElement Is NULL
07-15 07:44:40.292  1059 30876 D OisMgr  : [dc23155_64D] onStart, mWidth=768, mHeight=576, mSize=667648
07-15 07:44:40.292  1059 30845 D Misc____: [dc23155_64D] enter onStart()
07-15 07:44:40.292  1059 30876 D OisMgr  : [dc23155_64D] onStart(), mHwCam2Flag=0
07-15 07:44:40.292  1059 30845 I Misc____: [dc23155_64I] dynamicVideoFpsMode = 0'''

kolodezny, 2024-07-21:
I also came across this problem. When starting/waking up and torch='on' it doesn't work. After any screen update, torch starts working as expected.

Hejifei, 2024-09-06:
> I also came across this problem. When starting/waking up and torch='on' it doesn't work. After any screen update, torch starts working as expected.

Sorry, I can't undestand "After any screen update"

Hejifei, 2024-09-06:
> I also came across this problem. When starting/waking up and torch='on' it doesn't work. After any screen update, torch starts working as expected.

Sorry, I can't understand "After any screen update"

Hejifei, 2024-09-06:
It works fine on Xiaomi, Google, iQOO and other phones, but it is abnormal on Huawei phones

YUOrz233, 2025-04-15:
> It works fine on Xiaomi, Google, iQOO and other phones, but it is abnormal on Huawei phones

The 'hasTorch' of the cameraDevice I obtained using 'useCameraDevice ('back') 'returned' false '. But when I used 'Camera. getAvailableCameraDevices()', I found that the returned result contained devices with 'hasTorch===true', so I optimized the code for obtaining camera devices

`

const [cameraDevice, setCameraDevice] = useState(null)

const initCameraDevice = () => {
    const devices = Camera.getAvailableCameraDevices()
    console.log('AvailableCameraDevices', devices);
    // Priority should be given to rear devices with flash. If none are available, the default method will be used to obtain available rear devices. If none are available, the page will prompt "No available camera devices found"
    const device = devices.find(item => item.position === 'back' && item.hasTorch)
    if (device) {
      setCameraDevice(device)
    } else {
      setCameraDevice(useCameraDevice('back'))
    }
  }

if (cameraDevice) {
        return (
          <Camera
            ref={cameraRef}
            style={[styles.flex1]}
            device={cameraDevice} 
            isActive={true}
            codeScanner={codeScanner}
            torch={torch ? 'on' : 'off'}
          />
        )
      } else {
        return (
          <View style={[styles.flex1, styles.rowJustifyCenter, styles.bgColorBlack]}>
            <Text style={[styles.fontWhite20]}>No available rear camera device found</Text>
          </View>
        )
      }
`