margelo/react-native-vision-camera #2930: 🐛  Error when switching front camera on iPhone

seereal-beomjin, 2024-06-03:
### What's happening?

I use isSelfee to switch cameras, but when switching from the rear to the front, an error occurs only in iOS.
```
err: Rendered fewer hooks than expected. This may be caused by an accidental early return statement.
```

### Reproduceable Code

```tsx
const device: any = useCameraDevice(isSelfee ? 'front' : 'back', {
    physicalDevices: isSelfee ? [] : ['ultra-wide-angle-camera', 'wide-angle-camera', 'telephoto-camera'],
  });
  const format = useCameraFormat(device, [{photoAspectRatio: ios ? ratio : 9 / 16}, {photoResolution: 'max'}]);
```

### Relevant log output

```shell
VisionCamera.onCameraStarted(): Camera started!
VisionCamera.onSessionInitialized(): Camera initialized!
VisionCamera.configure(_:): Beginning AudioSession configuration...
VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
VisionCamera.configure(_:): Committed AudioSession configuration!
Error: Rendered fewer hooks than expected. This may be caused by an accidental early return statement.

This error is located at:
    in Zoom (created by CellRenderer)
    in RCTView (created by View)
    in View (created by CellRenderer)
    in RCTView (created by View)
    in View (created by CellRenderer)
    in VirtualizedListCellContextProvider (created by CellRenderer)
    in CellRenderer (created by VirtualizedList)
    in RCTScrollContentView (created by ScrollView)
    in RCTScrollView (created by ScrollView)
    in ScrollView (created by ScrollView)
    in ScrollView (created by NativeViewGestureHandler)
    in NativeViewGestureHandler (created by ScrollView)
    in ScrollView
    in Unknown (created by VirtualizedList)
    in VirtualizedListContextProvider (created by VirtualizedList)
    in VirtualizedList (created by FlatList)
    in FlatList
    in Unknown (created by CameraScreen)
    in CameraScreen (created by SceneView)
    in StaticContainer
    in EnsureSingleNavigator (created by SceneView)
    in SceneView (created by BottomTabView)
    in RCTView (created by View)
    in View (created by Screen)
    in RCTView (created by View)
    in View (created by Background)
    in Background (created by Screen)
    in Screen (created by BottomTabView)
    in RNSModalScreen
    in Unknown (created by InnerScreen)
    in Suspender (created by Freeze)
    in Suspense (created by Freeze)
    in Freeze (created by DelayedFreeze)
    in DelayedFreeze (created by InnerScreen)
    in InnerScreen (created by Screen)
    in Screen (created by MaybeScreen)
    in MaybeScreen (created by BottomTabView)
    in RNSScreenNavigationContainer (created by ScreenContainer)
    in ScreenContainer (created by MaybeScreenContainer)
    in MaybeScreenContainer (created by BottomTabView)
    in RCTView (created by View)
    in View (created by SafeAreaProviderCompat)
    in SafeAreaProviderCompat (created by BottomTabView)
    in BottomTabView (created by BottomTabNavigator)
    in PreventRemoveProvider (created by NavigationContent)
    in NavigationContent
    in Unknown (created by BottomTabNavigator)
    in BottomTabNavigator (created by TabNav)
    in TabNav (created by SceneView)
    in StaticContainer
    in EnsureSingleNavigator (created by SceneView)
    in SceneView (created by CardContainer)
    in RCTView (created by View)
    in View (created by CardContainer)
    in RCTView (created by View)
    in View (created by CardContainer)
    in RCTView (created by View)
    in View
    in CardSheet (created by Card)
    in RCTView (created by View)
    in View
    in Unknown (created by PanGestureHandler)
    in PanGestureHandler (created by PanGestureHandler)
    in PanGestureHandler (created by Card)
    in RCTView (created by View)
    in View
    in Unknown (created by Card)
    in RCTView (created by View)
    in View (created by Card)
    in Card (created by CardContainer)
    in CardContainer (created by CardStack)
    in RNSModalScreen
    in Unknown (created by InnerScreen)
    in Suspender (created by Freeze)
    in Suspense (created by Freeze)
    in Freeze (created by DelayedFreeze)
    in DelayedFreeze (created by InnerScreen)
    in InnerScreen (created by Screen)
    in Screen (created by MaybeScreen)
    in MaybeScreen (created by CardStack)
    in RNSScreenContainer (created by ScreenContainer)
    in ScreenContainer (created by MaybeScreenContainer)
    in MaybeScreenContainer (created by CardStack)
    in RCTView (created by View)
    in View (created by Background)
    in Background (created by CardStack)
    in CardStack (created by HeaderShownContext)
    in RCTView (created by View)
    in View (created by SafeAreaProviderCompat)
    in SafeAreaProviderCompat (created by StackView)
    in RCTView (created by View)
    in View (created by GestureHandlerRootView)
    in GestureHandlerRootView (created by StackView)
    in StackView (created by StackNavigator)
    in PreventRemoveProvider (created by NavigationContent)
    in NavigationContent
    in Unknown (created by StackNavigator)
    in StackNavigator (created by NavigationStacks)
    in NavigationStacks (created by MainStack)
    in MainStack (created by App)
    in StickerProvider (created by App)
    in PortalProviderComponent (created by App)
    in EnsureSingleNavigator
    in BaseNavigationContainer
    in ThemeProvider
    in NavigationContainerInner (created by App)
    in Provider (created by App)
    in RNCSafeAreaProvider (created by SafeAreaProvider)
    in SafeAreaProvider (created by App)
    in RCTView (created by View)
    in View (created by GestureHandlerRootView)
    in GestureHandlerRootView (created by App)
    in App (created by CodePushComponent)
    in CodePushComponent
    in RCTView (created by View)
    in View (created by AppContainer)
    in RCTView (created by View)
    in View (created by AppContainer)
    in AppContainer
    in SeeRealBox(RootComponent), js engine: hermes
```

### Camera Device

```json
{
  "hardwareLevel": "full",
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:6",
  "minFocusDistance": 12,
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "isMultiCam": true,
  "sensorOrientation": "landscape-right",
  "formats": [],
  "position": "back",
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 2,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera"
  ],
  "hasFlash": true,
  "name": "Back Dual Wide Camera",
  "minExposure": -8,
  "maxZoom": 123.75,
  "hasTorch": true,
  "minZoom": 1
}
```

### Device

iPhone 12 Mini

### VisionCamera Version

3.9.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-06-03:
> ```
> Error: Rendered fewer hooks than expected. This may be caused by an accidental early return statement
> ```

this is not VisionCamera's fault. You are using hooks wrong, I recommend you read about React hooks to make sure you understand how to properly use them.

Also, this is how to use it:

```ts
const front = useCameraDevice('front')
const back = useCameraDevice('back', { physicalDevices: [...] })
```

closed 2024-06-03 by mrousavy

seereal-beomjin, 2024-06-11:
@mrousavy 
```
const [isSelfee, setIsSelfee] = useState(false)

const front = useCameraDevice('front')
const back = useCameraDevice('back', { physicalDevices: ['ultra-wide-angle-camera', 'wide-angle-camera'] })

const backFormat = useCameraFormat(backDevice, [{photoAspectRatio: ios ? ratio : 9 / 16}, {photoResolution: 'max'}]);
const frontFormat = useCameraFormat(frontDevice, [{photoAspectRatio: ios ? ratio : 9 / 16}, {photoResolution: 'max'}]);
  
 const device = isSelfee ? frontDevice : backDevice;
 const format = isSelfee ? frontFormat : backFormat;
 
 
 return (
 <ReanimatedCamera
                          ref={cameraRef}
                          style={[styles.camera, {aspectRatio: 1 / ratio, width: '100%'}]}
                          device={device}
                          format={format}
                          isActive={true}
                          photo={isPhoto}
                          video={!isPhoto}
                          audio={!isPhoto}
                          animatedProps={animatedProps}
                          onTouchEnd={handleFocus}
                          zoom={device.neutralZoom}
                          exposure={exposure}
                          enableHighQualityPhotos
                        />
 )
 ```
 
 In this code, when i change isSelfee true to false 
 
 Rendered fewer h ooks than expected. This may be cause dby an accidental early return statement.
 
 but without physical device, it is work well