margelo/react-native-vision-camera #2944: 🐛  ERROR  Camera.onError(session/recoverable-error): An unknown error occurred while creating the Camera Session, but the Camera can recover from it. [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]

bigg-S, 2024-06-06:
### What's happening?

Hello,

Everything is functioning properly when I use useFrameProcessor but when I apply useSkiaFrameProcessor to the camera. I get an error that says:

    ERROR Camera.onError(session/recoverable-error): An unknown error occurred while creating the Camera Session, but the Camera can recover from it. [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]

This is the relevant part of my package.json:
"react": "18.2.0",
"react-native": "0.72.4",
"@shopify/react-native-skia": "^1.2.3",
"react-native-reanimated": "^3.11.0",
"react-native-vision-camera": "^4.0.3",
"react-native-worklets-core": "^1.3.0",

When I comment out the 'frameprocessor' attribute from the camera component or use 'useFramProcessor' hook, everything works  just fine

I appreciate any help, Thanks

### Reproduceable Code

```tsx
`const frameProcessor = useSkiaFrameProcessor(
		frame => {
		"worklet"
		runAtTargetFps(5, () => {
			"worklet"
			if (templateAligned) {
				try {
					const data = resize(frame, {
						crop: {
							x: 0,
							y: 0,
							width: frame.width,
							height: frame.height,
						},
						scale: {
							width: 256,
							height: 256,
						},
						pixelFormat: 'rgb',
						dataType: 'float32',
						rotation: rotation
					});

					// Run the model inference
					const result = plugin.model?.runSync([data]);
					const locations = result ? result[0] : []; // detections
					const scores = result ? result[1] : []; // confidence scores

					const numDetections = locations.length / 4;
					const numClasses = 3;

					let bowlerDetected = false;
					
					outerLoop:for (let i = 0; i < numDetections; i++) {
						const offsetDetections = i * 4;
						const offsetScores = i * numClasses;

						const detectionSlice = locations.slice(offsetDetections, offsetDetections + 4);
						const scoreSlice = scores.slice(offsetScores, offsetScores + numClasses);

						for (let j = 0; j < numClasses; j++) {
							const confidence = scoreSlice[j];
							if (confidence > MIN_CONFIDENCE) {
								console.log('Confidence: ', confidence);
								const bbox = {
									x_min: Number(detectionSlice[1]),
									y_min: Number(detectionSlice[0]),
									x_max: Number(detectionSlice[3]),
									y_max: Number(detectionSlice[2]),
								};
								
								console.log('Bbox', bbox);

								const rect = Skia.XYWHRect(bbox.x_min, bbox.y_min, bbox.x_max, bbox.y_max);
								frame.drawRect(rect, paint);

								const classLabel = CLASS_LABELS[j];
								if (classLabel === "bowlers") {
									console.log('detected a bowler!');
									bowlerDetected = true;
									break outerLoop;
								}
							}
						}
					}

					if (bowlerDetected && !recording) {
						startVideoRecordingJS();
					}
				} catch (error: any) {
					console.log("An error occurred during frame processing: ", error.message);
					setErrorJS("Error processing the frame");
				}
			}
		});
	}, [templateAligned]);
`
`<Camera
  style={StyleSheet.absoluteFill}
  ref={camera}
  fps={format?.maxFps}
  video={true}
  audio={false}
  device={device}
  enableFpsGraph
  isActive={true}
  frameProcessor={recording ? undefined : frameProcessor}
  format={format}
  pixelFormat={pixelFormat}
  />`
```

### Relevant log output

```shell
ERROR  Camera.onError(session/recoverable-error): An unknown error occurred while creating the Camera Session, but the Camera can recover from it. [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]
```

### Camera Device

```json
{
  "hardwareLevel": "limited",
  "minExposure": -16,
  "neutralZoom": 1,
  "minZoom": 1,
  "supportsFocus": true,
  "formats": [],
  "supportsLowLightBoost": true,
  "hasTorch": true,
  "supportsRawCapture": false,
  "minFocusDistance": 0,
  "sensorOrientation": "landscape-left",
  "maxZoom": 2,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "isMultiCam": false,
  "position": "back",
  "id": "0",
  "hasFlash": true,
  "maxExposure": 16,
  "name": "0 (BACK) androidx.camera.camera2"
}
```

### Device

Android

### VisionCamera Version

4.0.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bigg-S, 2024-06-10:
I was able to pinpoint the cause of the error. I don't know about other devices but the 'useSkiaFrameProcessor', for some reason doesn't work with 'back' cameras. I was able to reproduce that with the example app on an android platform. Unless this was apparent and am missing something.

mrousavy, 2024-06-13:
Hey! Can you try with 4.3.2 again? I fixed a bunch of issues and upgraded CameraX to the latest version - all is running perfectly stable on my phones now :)

mrousavy, 2024-06-13:
I _think_ this is fixed in 4.3.2, I'll close this issue and if you still experience this issue then we can re-open.

Either way, I'd need native `adb logcat` logs to investigate this.

closed 2024-06-13 by mrousavy

usamaabutt, 2024-12-15:
Hi @mrousavy I am experiencing the black/blank screen issue for a few android's devices back camera  using v4.5.2 
https://github.com/mrousavy/react-native-vision-camera/issues/3297

i7soft, 2025-03-14:
> I _think_ this is fixed in 4.3.2, I'll close this issue and if you still experience this issue then we can re-open.
> 
> Either way, I'd need native `adb logcat` logs to investigate this.

4.6.3 has the same error on my phone.

`
26627 SyncCaptureSessionImpl  cn.coollashare.draw                  D  [androidx.camera.camera2.internal.SynchronizedCaptureSessionImpl@1eb85f7] Session onConfigured()
2025-03-14 12:38:59.185 20061-26627 CaptureSession          cn.coollashare.draw                  D  Attempting to send capture request onConfigured
2025-03-14 12:38:59.185 20061-26627 CaptureSession          cn.coollashare.draw                  D  Issuing request for session.
2025-03-14 12:38:59.185 20061-26627 Camera2Cap...estBuilder cn.coollashare.draw                  D  createCaptureRequest
2025-03-14 12:38:59.188 20061-26627 CaptureSession          cn.coollashare.draw                  D  CameraCaptureSession.onConfigured() mState=OPENED
2025-03-14 12:38:59.189 20061-26627 CaptureSession          cn.coollashare.draw                  D  CameraCaptureSession.onReady() OPENED
2025-03-14 12:38:59.189 20061-26516 qdgralloc               cn.coollashare.draw                  D  Falling back to default YUV format - no camera/video specific format defined, usage 0x3
2025-03-14 12:38:59.192 20061-26623 DeferrableSurface       cn.coollashare.draw                  D  use count-1,  useCount=0 closed=false androidx.camera.core.impl.ImmediateSurface@bf6c2a4
2025-03-14 12:38:59.192 20061-26623 DeferrableSurface       cn.coollashare.draw                  D  Surface no longer in use[total_surfaces=3, used_surfaces=1](androidx.camera.core.impl.ImmediateSurface@bf6c2a4}
2025-03-14 12:38:59.192 20061-26623 DeferrableSurface       cn.coollashare.draw                  D  use count-1,  useCount=1 closed=false androidx.camera.core.SurfaceRequest$2@7fb1579
2025-03-14 12:38:59.193 20061-26623 Camera2CameraImpl       cn.coollashare.draw                  D  {Camera@e6b322a[id=1]} Camera receive onErrorCallback
2025-03-14 12:38:59.193 20061-26623 Camera2CameraImpl       cn.coollashare.draw                  D  CameraDevice.onError(): 1 failed with ERROR_CAMERA_DEVICE while in OPENED state. Will attempt recovering from error.
2025-03-14 12:38:59.194 20061-26623 Camera2CameraImpl       cn.coollashare.draw                  D  Attempt to reopen camera[1] after error[ERROR_CAMERA_DEVICE]
2025-03-14 12:38:59.194 20061-26623 Camera2CameraImpl       cn.coollashare.draw                  D  {Camera@e6b322a[id=1]} Transitioning camera internal state: OPENED --> REOPENING
2025-03-14 12:38:59.197 20061-26623 CameraStateRegistry     cn.coollashare.draw                  D  Recalculating open cameras:
                                                                                                    Camera                                       State                 
                                                                                                    -------------------------------------------------------------------
                                                                                                    Camera@e6b322a[id=1]                         OPENING               
                                                                                                    Camera@1b25659[id=0]                         UNKNOWN               
                                                                                                    -------------------------------------------------------------------
                                                                                                    Open count: 1 (Max allowed: 1)
2025-03-14 12:38:59.197 20061-26623 CameraStateMachine      cn.coollashare.draw                  D  New public camera state CameraState{type=OPENING, error=StateError{code=3, cause=null}} from OPENING and StateError{code=3, cause=null}
2025-03-14 12:38:59.197 20061-26623 CameraStateMachine      cn.coollashare.draw                  D  Publishing new public camera state CameraState{type=OPENING, error=StateError{code=3, cause=null}}
2025-03-14 12:38:59.198 20061-26623 Camera2CameraImpl       cn.coollashare.draw                  D  {Camera@e6b322a[id=1]} Resetting Capture Session
2025-03-14 12:38:59.198 20061-26623 Camera2CameraImpl       cn.coollashare.draw                  D  {Camera@e6b322a[id=1]} Skipping Capture Session state check due to current camera state: REOPENING and previous session status: true
2025-03-14 12:38:59.199 20061-26623 SyncCaptureSessionImpl  cn.coollashare.draw                  D  [androidx.camera.camera2.internal.SynchronizedCaptureSessionImpl@1eb85f7] Session call close()
2025-03-14 12:38:59.199 20061-26623 Camera2CameraImpl       cn.coollashare.draw                  D  {Camera@e6b322a[id=1]} Releasing session in state REOPENING
2025-03-14 12:38:59.199 20061-20061 CameraSession           cn.coollashare.draw                  I  Camera State: OPENING (has error: true)
2025-03-14 12:38:59.200 20061-20061 CameraView              cn.coollashare.draw                  I  invokeOnStopped()
2025-03-14 12:38:59.200 20061-26623 SyncCaptureSessionImpl  cn.coollashare.draw                  D  [androidx.camera.camera2.internal.SynchronizedCaptureSessionImpl@1eb85f7] onClosed()
2025-03-14 12:38:59.200 20061-26623 CaptureSession          cn.coollashare.draw                  D  onSessionFinished()
[... cut ...]
2025-03-14 12:39:00.166 20061-26627 CameraStateMachine      cn.coollashare.draw                  D  Publishing new public camera state CameraState{type=OPENING, error=null}
2025-03-14 12:39:00.166 20061-20061 CameraSession           cn.coollashare.draw                  I  Camera State: OPENING (has error: false)
2025-03-14 12:39:00.168 20061-26627 UseCaseAttachState      cn.coollashare.draw                  D  All use case: [androidx.camera.core.ImageAnalysis-10652708-3681-4189-ac85-212b23520a5f23221913, androidx.camera.video.VideoCapture-454e46e2-5164-439b-8e52-33a8038ece955456864] for camera: 1
`