margelo/react-native-vision-camera #2949: Disable Frame processor on route change💭

kssgarcia, 2024-06-07:
### Question

I am making an app that use expo-router, what I want to know if there is a way to disable the `useFrameProcessor` because when I go to another route the processor still is running. By the way, I am using `active` property when I change the route. This is my `Camera` component:
```
<Camera
  style={StyleSheet.absoluteFillObject}
  ref={cameraRef}
  device={device}
  isActive={isActive}
  photo={true}
  format={format}
  fps={fps}
  orientation="portrait"
  photoQualityBalance="speed"
  // enableFpsGraph={true}
  frameProcessor={frameProcessor}
/>
```

### VisionCamera Version

4.0.5

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

gwendall, 2024-06-07:
You could only run the camera if the screen is active. If the camera doesn't run, the frame processor should not either. So something like:

```
import { useNavigation, usePathname } from 'expo-router';
...
const navigation = useNavigation();
const isScreenFocused = navigation.isFocused();
...
<Camera
   ...
   isActive={isScreenFocused}
/>
```

kssgarcia, 2024-06-07:
I understood you that I should set the `active` property `false` when the screen is not focus, but I am already doing this,

```
<Camera
  style={StyleSheet.absoluteFillObject}
  ref={cameraRef}
  device={device}
  isActive={isActive}
  photo={true}
  format={format}
  fps={fps}
  orientation="portrait"
  photoQualityBalance="speed"
  // enableFpsGraph={true}
  frameProcessor={frameProcessor}
/>
```
the  ` isActive={isActive}` set this to `false` when the screen is not focus or when the app is in second plane. This seems to work fine, but when I use `frameProcessor={frameProcessor}` this is still running when the screen is not focus.

gwendall, 2024-06-07:
Strange, maybe it's a regression on a new version? In this case you could pass the isActive as a dependency of the frame processor and not run what's inside if it is not?

like

```
useFrameProcessor((frame) => {
  if (!isActive) return;
}, [isActive]);
```

kssgarcia, 2024-06-07:
@gwendall you were right, ty. It seems that I was setting the `isActive` in a wrong way. Ty man

closed 2024-06-07 by kssgarcia