margelo/react-native-vision-camera #2950: 🐛 No onOutputOrientationChanged event on v4.1.0

RenatoFerracini, 2024-06-08:
### What's happening?

According to the documentation at https://react-native-vision-camera.com/docs/api/interfaces/CameraProps#onoutputorientationchanged this event should be defined at types/CameraProps.ts:324 but it isn't. I even checked the repository, it is missing.

### Reproduceable Code

```tsx
<ReanimatedCamera
              style={StyleSheet.absoluteFill}
              device={device}
              isActive={isActive}
              ref={camera}
              outputOrientation="device"
              onOutputOrientationChanged={onOutputOrientationChanged}
              onInitialized={onInitialized}
              onError={onError}
              format={format}
              photoQualityBalance="quality"
              lowLightBoost={device.supportsLowLightBoost && enableNightMode}
              enableZoomGesture={false}
              animatedProps={cameraAnimatedProps}
            />
```

### Relevant log output

```shell
Type '{ style: RegisteredStyle<AbsoluteFillStyle>; device: CameraDevice; isActive: boolean; ref: RefObject<Camera>; ... 10 more ...; photo: true; }' is not assignable to type 'IntrinsicAttributes & IntrinsicClassAttributes<Component<AnimateProps<CameraProps>, any, any>> & Readonly<...>'.
  Property 'onOutputOrientationChanged' does not exist on type 'IntrinsicAttributes & IntrinsicClassAttributes<Component<AnimateProps<CameraProps>, any, any>> & Readonly<...>'.ts(2322)
(property) onOutputOrientationChanged: () => void
```

### Camera Device

```json
N/A
```

### Device

Any

### VisionCamera Version

v4.1.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-06-08:
Docs aren't versioned. The event is on main, but not yet released.

mrousavy, 2024-06-08:
> Yes, I can reproduce the same issue in the Example app here

Lie. The orientation event is part of the example app and fires accordingly.

closed 2024-06-08 by mrousavy