margelo/react-native-vision-camera #2951: 🐛  useSkiaFrameProcessor renders black screen using front cameras

khlling, 2024-06-08:
### What's happening?

When calling frame.render() within useSkiaFrameProcessor on an Android phone, the captured frames render as expected when the camera position is set to 'back'. However, setting the camera position to 'front' results in a black screen.

Omitting useSkiaFrameProcessor (i.e., not passing a frame processor as a prop to ReanimatedCamera) renders as expected
Recording and playing back the video output from the front camera works fine, indicating the frames are captured correctly but not rendered properly in real-time.

### Reproduceable Code

```tsx
let device = useCameraDevice('front');
  const format = useCameraFormat(device, Templates.FrameProcessing);
  const microphone = useMicrophonePermission();
  const location = useLocationPermission();

  const frameProcessor = useSkiaFrameProcessor((frame) => {
    'worklet';
    frame.render();
  }, []);

<ReanimatedCamera
                style={StyleSheet.absoluteFill}
                device={device}
                ref={camera}
                format={format}
                enableFpsGraph={true}
                photo={true}
                video={true}
                audio={microphone.hasPermission}
                enableLocation={location.hasPermission}
                frameProcessor={frameProcessor}
              />
```

### Relevant log output

```shell
2024-06-09 00:28:56.362 10089-10089 VRI[MainActivity]       com.toggled.wuji                     D  Start draw after previous draw not visible
2024-06-09 00:28:56.389 10089-10089 CameraView              com.toggled.wuji                     I  Updating CameraSession...
2024-06-09 00:28:56.396 10089-10089 CameraSession           com.toggled.wuji                     I  configure { ... }: Waiting for lock...
2024-06-09 00:28:56.396 10089-10089 CameraSession           com.toggled.wuji                     I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=false, sidePropsChanged=false, isActiveChanged=true, orientationChanged=false, locationChanged=true)
2024-06-09 00:28:56.397 10089-10089 CameraSession           com.toggled.wuji                     I  Camera Lifecycle changed to STARTED!
2024-06-09 00:28:56.397 10089-10183 UseCaseAttachState      com.toggled.wuji                     D  Active and attached use case: [] for camera: 1
2024-06-09 00:28:56.397 10089-10089 CapabilitiesByQuality   com.toggled.wuji                     D  profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=22000000, frameRate=30, width=1920, height=1080, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
2024-06-09 00:28:56.397 10089-10089 CapabilitiesByQuality   com.toggled.wuji                     D  profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=12000000, frameRate=30, width=1280, height=720, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
2024-06-09 00:28:56.397 10089-10089 CapabilitiesByQuality   com.toggled.wuji                     D  profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=6000000, frameRate=30, width=720, height=480, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
2024-06-09 00:28:56.397 10089-10089 CapabilitiesByQuality   com.toggled.wuji                     D  Using supported quality of ConstantQuality{value=5, name=HD, typicalSizes=[1280x720]} for size 1280x720
2024-06-09 00:28:56.397 10089-10089 VideoConfigUtil         com.toggled.wuji                     D  MediaSpec video mime matches EncoderProfiles. Using EncoderProfiles to derive VIDEO settings [mime type: video/avc]
2024-06-09 00:28:56.397 10089-10089 VidEncVdPrflRslvr       com.toggled.wuji                     D  Resolved frame rate 30fps [Video profile frame rate: 30fps, Expected operating range: <UNSPECIFIED>]
2024-06-09 00:28:56.397 10089-10089 VidEncVdPrflRslvr       com.toggled.wuji                     D  Resolved VIDEO frame rate: 30fps
2024-06-09 00:28:56.397 10089-10089 VidEncVdPrflRslvr       com.toggled.wuji                     D  Using resolved VIDEO bitrate from EncoderProfiles
2024-06-09 00:28:56.397 10089-10089 VideoConfigUtil         com.toggled.wuji                     D  Base Bitrate(12000000bps) * Bit Depth Ratio (8 / 8) * Frame Rate Ratio(30 / 30) * Width Ratio(1280 / 1280) * Height Ratio(720 / 720) = 12000000
2024-06-09 00:28:56.397 10089-10089 VideoConfigUtil         com.toggled.wuji                     W  Unsupported mime type video/avc or profile level -1. Data space is unspecified.
2024-06-09 00:28:56.397 10089-10089 CameraOrientationUtil   com.toggled.wuji                     D  getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=270, isOppositeFacing=false, result=270
2024-06-09 00:28:56.397 10089-10089 VideoCapture            com.toggled.wuji                     D  Surface processing is enabled.
2024-06-09 00:28:56.400 10089-10206 NativeCrypto            com.toggled.wuji                     V  Read error: ssl=0xb400007a33915798: I/O error during system call, Software caused connection abort
2024-06-09 00:28:56.400 10089-10206 NativeCrypto            com.toggled.wuji                     V  SSL shutdown failed: ssl=0xb400007a33915798: I/O error during system call, Broken pipe
2024-06-09 00:28:56.406 10089-12027 OpenGlRenderer          com.toggled.wuji                     D  EGLContext created, client version 2
2024-06-09 00:28:56.407 10089-10089 VideoCapture            com.toggled.wuji                     D  camera timebase = REALTIME, processing timebase = REALTIME
2024-06-09 00:28:56.407 10089-10089 DeferrableSurface       com.toggled.wuji                     D  Surface created[total_surfaces=3, used_surfaces=0](androidx.camera.core.processing.SurfaceEdge$SettableSurface@4b889a5}
2024-06-09 00:28:56.407 10089-10089 DeferrableSurface       com.toggled.wuji                     D  Surface created[total_surfaces=4, used_surfaces=0](androidx.camera.core.processing.SurfaceEdge$SettableSurface@823a588}
2024-06-09 00:28:56.407 10089-10089 DeferrableSurface       com.toggled.wuji                     D  Surface created[total_surfaces=5, used_surfaces=0](androidx.camera.core.SurfaceRequest$2@9301107}
2024-06-09 00:28:56.407 10089-10089 DeferrableSurface       com.toggled.wuji                     D  New surface in use[total_surfaces=5, used_surfaces=1](androidx.camera.core.SurfaceRequest$2@9301107}
2024-06-09 00:28:56.407 10089-10089 DeferrableSurface       com.toggled.wuji                     D  use count+1, useCount=1 androidx.camera.core.SurfaceRequest$2@9301107
2024-06-09 00:28:56.407 10089-10089 DeferrableSurface       com.toggled.wuji                     D  Surface created[total_surfaces=6, used_surfaces=1](androidx.camera.core.SurfaceRequest$2@1a0f3cc}
2024-06-09 00:28:56.408 10089-10089 DeferrableSurface       com.toggled.wuji                     D  New surface in use[total_surfaces=6, used_surfaces=2](androidx.camera.core.SurfaceRequest$2@1a0f3cc}
2024-06-09 00:28:56.408 10089-10089 DeferrableSurface       com.toggled.wuji                     D  use count+1, useCount=1 androidx.camera.core.SurfaceRequest$2@1a0f3cc
2024-06-09 00:28:56.408 10089-10089 Recorder                com.toggled.wuji                     D  Surface is requested in state: CONFIGURING, Current surface: 0
2024-06-09 00:28:56.408 10089-10089 CameraOrientationUtil   com.toggled.wuji                     D  getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=270, isOppositeFacing=false, result=270
2024-06-09 00:28:56.408 10089-10386 CapabilitiesByQuality   com.toggled.wuji                     D  profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{code [... cut ...]
### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DarkShtir, 2024-06-10:
Same issue.
But mine version is "4.2.0"

khlling, 2024-06-10:
I've updated my version to "4.2.0" also. By chance, I tipped my phone from portrait to landscape whilst using the front camera and it started working. As soon as I tip it back to portrait it goes black. Try that out @DarkShtir. 

I assume therefore it's something to do with the new orientation changes

DarkShtir, 2024-06-11:
@khlling this works, but the same time is not easy and comfortable to use.
I Downgraded to "4.0.5" and use custom patch to transform, rotate and align canvas in skia fame processor
I'll try get time for additional investigation of this issue

khlling, 2024-06-11:
@DarkShtir Agreed I'm sure it's not the intended behaviour. 

What changes did you have to make in 4.0.5 to get it to work? Can you share your patch files?

DarkShtir, 2024-06-11:
[react-native-vision-camera+4.0.5.patch](https://github.com/user-attachments/files/15787294/react-native-vision-camera%2B4.0.5.patch)
No problem, is local fix for project, to use skiaFrameProcessor
But it isn't great, and has bugs, for example reverse rotations when using front camera. But now this fix is resolving our issues.
Hope this help you.

tsnguyenducphuong, 2024-06-12:
I had the same blank screen issue on Android 14 (Samsung A53) and resolved the issue by changing minSdkVersion from 24 to 26 in android/build.gradle.

tsnguyenducphuong, 2024-06-12:
Also add the following would help: 
 const pixelFormat = Platform.OS === 'ios' ? 'rgb' : 'yuv';

mrousavy, 2024-06-12:
@tsnguyenducphuong wait I thought VisionCamera/Skia will throw an error if `minSdkVersion` is not 26 or higher? Need to double-check with @wcandillon, but we should definitely throw very concise errors in such cases instead of returning null and just logging the errors to the console..

mrousavy, 2024-06-12:
Either way, the issue from OP (@khlling) is not related to minSdkversion as it renders fine on back camera, but not on front. I think this has to do with the orientation/matrix code - @khlling can you play around with this function to figure out where it goes wrong? https://github.com/mrousavy/react-native-vision-camera/blob/90be06b5534ecddb9b8157881aaea0ed49173cb5/package/src/skia/useSkiaFrameProcessor.ts#L61-L82

I unfortunately don't have the free time to look into this right now, but if you find a solution in there I'd appreciate a PR! Thanks

mrousavy, 2024-06-12:
Maybe it'd also be helpful if you posted `device.sensorOrientation` as well as `frame.orientation` of **both** the back and the front camera here just so we know what values we're dealing with.

khlling, 2024-06-12:
> @tsnguyenducphuong wait I thought VisionCamera/Skia will throw an error if `minSdkVersion` is not 26 or higher? Need to double-check with @wcandillon, but we should definitely throw very concise errors in such cases instead of returning null and just logging the errors to the console..

I can confirm that VisionCamera/Skia does throw an error for this. I had to update the `minSdkVersion`on the back of the build error. 

Post this update I'm getting a blank front camera preview.

khlling, 2024-06-12:
> Either way, the issue from OP (@khlling) is not related to minSdkversion as it renders fine on back camera, but not on front. I think this has to do with the orientation/matrix code - @khlling can [...]

No worries @mrousavy let me do a bit of digging. Thanks for giving me a hint of where to start looking!

Will update this issue with logs and investigation.

tsnguyenducphuong, 2024-06-12:
In my case, I had first the blank screen issue then the black screen and solved with the following steps: (I am using Vision Camera version 4.2.1)

1. minSdkVersion = 26 (solved blank issue).

Solve black screen:

2. Set pixelFormat
3. Set outputOrientation = "device"
3. Set photoQualityBalance = "speed"

wcandillon, 2024-06-13:
@mrousavy it's interesting because in the video use-case the error is clean and concise, I assume it would be the same for the vision-camera integration? Could it be because of an older version of Skia (I did improve this recently I think)

mrousavy, 2024-06-13:
True, maybe an old RN Skia version! I'll do more research thanks

mfkrause, 2024-06-14:
I'm experiencing the same bug, but on iOS. When a Skia frame processor is connected, the camera stays black as long as the iPhone is in portrait. Some observations:

When I rotate the iPhone to landscape (which our UI doesn't support), the camera appears, but is rotated by 90 degrees clockwise. On iPad, where we support both portrait and landscape mode for the UI, everything works as expected, though.

When I set `outputOrientation` to `preview`, the camera frame stays visible, but stays locked in its 90 degrees rotated position (no matter whether or not the device's screen orientation lock is enabled). Setting it to `portrait` makes the camera preview stay black permanently (in every device orientation).

I'm not sure if that's expected behavior for apps that don't support landscape mode, but I've also noticed `onPreviewOrientationChanged` never sends any events and stays locked on `portrait`. `onOutputOrientation` first sends an event with `landscape-left` once the camera mounts (which matches what the camera preview is showing when setting `outputOrientation` to `preview`), then immediately one with `portrait`, too, and stays locked on that as well.

I just rebuilt the app with the newest react-native-skia release (1.3.4) which didn't fix it, unfortunately.

mfkrause, 2024-06-28:
After I digged a little into the different orientation values provided by the library, I realized most of my previous comment is actually the expected behavior and the problem must indeed lie in [the frame rotation in useSkiaFrameProcessor.ts](https://github.com/mrousavy/react-native-vision-camera/blob/90be06b5534ecddb9b8157881aaea0ed49173cb5/package/src/skia/useSkiaFrameProcessor.ts#L53-L90). I played around a little and came up with this:

```ts
switch (frame.orientation) {
  case 'portrait':
  case 'portrait-upside-down':
    canvas.scale(frame.height / frame.width, frame.height / frame.width)
    canvas.rotate(90,0,0)
    canvas.translate(0, -(frame.height * 2))
    break
  case 'landscape-left':
  case 'landscape-right':
    canvas.translate(frame.height, 0)
    canvas.rotate(90, 0, 0)
    break
  default:
    throw new Error(`Invalid frame.orientation: ${frame.orientation}!`)
}
```

However, two caveats:

1. There's a minor glitch when rotating the device from portrait to either landscape direction, where the frame visibly "jumps" by a few pixels. I wasn't able to reliably fix that unfortunately (it's 11pm and the matrix translation stuff isn't quite getting in my head right now). I suppose the `scale()` values aren't quite right yet, as the frame still seems to extend beyond the device borders when the device is in landscape.
2. This doesn't work when the UI supports landscape mode and orientation lock is disabled. When rotating to landscape, the orientation is wrong again. The _original / current_ rotation code seems to apply the correct rotation in _most_ scenarios in this case, however 1. the image quality in landscape mode is incredibly bad and 2. when the device is portrait upside down, there's still a black screen.

So, this isn't even close to a full fix or PR ready yet, just some observations.

WillyGV, 2024-07-15:
Hello, I have the same problem, and you can solve it by changing the logic of the landscape-left case:

```
diff --git a/package/src/skia/useSkiaFrameProcessor.ts b/package/src/skia/useSkiaFrameProcessor.ts
index 10888109..16b667a2 100644
--- a/package/src/skia/useSkiaFrameProcessor.ts
+++ b/package/src/skia/useSkiaFrameProcessor.ts
@@ -95,8 +95,8 @@ function withRotatedFrame(frame: Frame, canvas: SkCanvas, previewOrientation: Or
         break
       case 'landscape-left':
         // rotate two flips on (0,0) origin and move X + Y into view again
-        canvas.translate(frame.height, frame.width)
-        canvas.translate(270, 0)
+        canvas.translate(0, frame.width)
+        canvas.rotate(270, 0)
         break
       case 'portrait-upside-down':
         // rotate three flips on (0,0) origin and move Y into view again
```
but I run into the problem that the camera creates a mirror effect, and I can't solve it.
I already tried to add the attribute to the 
```
<camera
isMirrored={true}
/>
```

but the image cannot be reversed

mrousavy, 2024-07-16:
> but I run into the problem that the camera creates a mirror effect, and I can't solve it. I already tried to add the attribute to the
> 
> ```
> <camera
> isMirrored={true}
> />
> ```
> 
> but the image cannot be reversed

You want to disable mirroring, but pass `isMirrored={true}`? If you want to **disable** it you need to set `isMirroring={false}`...

WillyGV, 2024-07-17:
I'm sorry @mrousavy.
What I wanted to say is that the mirror effect happens to me in the preview, I saw that in the code and in the documentation that isMirrored does not work for the preview, also I am only working on Android for now and in the Google documentation, the Preview class, Does not support [SetMirrorMode ](https://developer.android.com/reference/androidx/camera/core/Preview.Builder#setMirrorMode(int))for android versions prior to API 32,
I think the problem is in Skia's canvas.

I tried isMirrored={false} and isMirrored={true} in the camera component, I haven't found out if it can be changed in the preview

mrousavy, 2024-07-18:
Ah, so the Preview is not mirrored with a Skia Frame Processor, but is properly mirrored without a Skia FP - is that what you're saying?

Azmye, 2024-07-20:
is there a workaround for this bug ?

JulianKingman, 2024-08-16:
Seeing this in iOS, running in an expo development build
```
"react-native-vision-camera": "^4.3.2",
"@shopify/react-native-skia": "1.2.3",
"react-native-reanimated": "~3.10.1",
```

I'm in early days of development, so it's not a show-stopper at the moment, let me know if I can provide any useful info.

mrousavy, 2024-08-16:
you can play around with this function to figure out where it goes wrong: https://github.com/mrousavy/react-native-vision-camera/blob/90be06b5534ecddb9b8157881aaea0ed49173cb5/package/src/skia/useSkiaFrameProcessor.ts#L61-L82

minhaz19, 2024-08-18:
@mrousavy change this in useSkiaFrameProcessor. It's working on my device.

`case 'landscape-right':
          canvas.translate(frame.height, 0)
          canvas.rotate(90, 0, 0)
          canvas.scale(-1, 1); // Flip horizontally
          canvas.translate(-frame.width, 0); // Adjust translation after flipping
          break`

raslankiwan, 2024-08-20:
I was facing this issue on iPad, the screen was cutting off in both landscape left and right, and in upside down the whole camera was giving black screen, I made this adjustment and it worked perfectly in all cases (no need to change rotation configuration)
```typescript
 switch (orientation) {
      case 'portrait':
          canvas.translate( 0,frame.height / 2)
        break
      case 'landscape-left':
        // rotate two flips on (0,0) origin and move X + Y into view again
        canvas.translate(0, frame.width)
        canvas.rotate(270, 0, 0)
        break
      case 'portrait-upside-down':
        // rotate three flips on (0,0) origin and move Y into view again
        canvas.translate(frame.width, frame.height * 1.5)
        canvas.rotate(180, 0, 0)
        break
      case 'landscape-right':
        // rotate one flip on (0,0) origin and move X into view again
        canvas.translate(frame.height, 0)
        canvas.rotate(90, 0, 0)
        break
      default:
        throw new Error(`Invalid frame.orientation: ${frame.orientation}!`)
    }

```

sjaiswalgit, 2024-08-21:
Face the same issue, found the cause that front camera actaully do a lateral inversion beacuse its need to show picture just like a mirror . You can verify this by drawing something on canvas and changing  camera from back to front

Viktor-as, 2024-08-23:
> I was facing this issue on iPad, the screen was cutting off in both landscape left and right, and in upside down the whole camera was giving black screen, I made this adjustment and it worked perfectly in all cases (no need to change rotation configuration)
> 
> ```ts
>  switch (orientation) {
>       case 'portrait':
>           canvas.translate( 0,frame.height / 2)
>         break
>       case 'landscape-left':
>         // rotate two flips on (0,0) origin and move X + Y into view again
>         canvas.translate(0, frame.width)
>         canvas.rotate(270, 0, 0)
>         break
>       case 'portrait-upside-down':
>         // rotate three flips on (0,0) origin and move Y into view again
>         canvas.translate(frame.width, frame.height * 1.5)
>         canvas.rotate(180, 0, 0)
>         break
>       case 'landscape-right':
>         // rotate one flip on (0,0) origin and move X into view again
>         canvas.translate(frame.height, 0)
>         canvas.rotate(90, 0, 0)
>         break
>       default:
>         throw new Error(`Invalid frame.orientation: ${frame.orientation}!`)
>     }
> ```

thanks @raslankiwan , I had also this issue on Android phone, in portrait view I was getting black screen. In landscape it was ok. 

I added your changes in node_modules\react-native-vision-camera\src\skia\useSkiaFrameProcessor.ts and now in portrait I also get camera view

softgenicsShubham, 2024-10-08:
@raslankiwan  same issues in the --version 4.5.3 also, but will fixed with your changes.

```
 switch (orientation) {
      case 'portrait':
          canvas.translate( 0,frame.height / 2)
        break
      case 'landscape-left':
        // rotate two flips on (0,0) origin and move X + Y into view again
        canvas.translate(0, frame.width)
        canvas.rotate(270, 0, 0)
        break
      case 'portrait-upside-down':
        // rotate three flips on (0,0) origin and move Y into view again
        canvas.translate(frame.width, frame.height * 1.5)
        canvas.rotate(180, 0, 0)
        break
      case 'landscape-right':
        // rotate one flip on (0,0) origin and move X into view again
        canvas.translate(frame.height, 0)
        canvas.rotate(90, 0, 0)
        break
      default:
        throw new Error(`Invalid frame.orientation: ${frame.orientation}!`)
    }
```

softgenicsShubham, 2024-10-08:
```
const frameProcessor = useSkiaFrameProcessor((frame) => {
    'worklet';
    frame.render();
    const centerX = frame.width / 2;
    const centerY = frame.height / 2;
    const rect = Skia.XYWHRect(centerX, centerY, 150, 150);
    const paint = Skia.Paint();
    paint.setColor(Skia.Color('red'));
    frame.drawRect(rect, paint);
  }, []);
  ```
  
With the above changes, I tried drawing using Skia. However, I am seeing some unexpected behavior on the screen. When the camera is set to the front position, the drawing appears in a different place, and when the camera is set to the back position, the drawing appears in yet another different position. I even tried using isMirror with both true and false values.

  
[image]
[image]

rachellauyui, 2024-10-26:
Hey, any update? Same problem here, black screen when using front camera, but all fine with back camera, both in android and ios.

But what is crazy is that once I transform it { scaleX: 1} in style, meaning did not transform anything, it works in iOS, but not android.... so it seems something in the native code of front camera that cause the issue, probably because front camera needs to mirror (flip left to right) in order to make it looks and works like mirror so that people can really use it to do make-up /shave, etc.

My code in case anybody interested (only work in iOS):
 <Camera
        style={{
          ...StyleSheet.absoluteFillObject,
          transform: [{ scaleX: 1 }],
        }}
        device={device}
        isActive={isActive}
        frameProcessor={frameProcessor}
        camera={true}
        ref={camera}
      />

frodriguez-hu, 2024-11-04:
> > ```ts
> >  switch (orientation) {
> >       case 'portrait':
> >           canvas.translate( 0,frame.height / 2)
> >         break
> >       case 'landscape-left':
> >         // rotate two flips on (0,0) origin and move X + Y into view again
> >         canvas.translate(0, frame.width)
> >         canvas.rotate(270, 0, 0)
> >         break
> >       case 'portrait-upside-down':
> >         // rotate three flips on (0,0) origin and move Y into view again
> >         canvas.translate(frame.width, frame.height * 1.5)
> >         canvas.rotate(180, 0, 0)
> >         break
> >       case 'landscape-right':
> >         // rotate one flip on (0,0) origin and move X into view again
> >         canvas.translate(frame.height, 0)
> >         canvas.rotate(90, 0, 0)
> >         break
> >       default:
> >         throw new Error(`Invalid frame.orientation: ${frame.orientation}!`)
> >     }
> > ```
> 
> Hi.
> 
> Are you guys planning update the package with this fix? Because it worked very well on my device...

It worked well on mine too, I made a patch on my app. But I am no t sure if it is the proper way of handling it

frodriguez-hu, 2024-11-14:
I am not sure about other orientations but on `landscape-left` which is the one coming for Android when you use it on portrait (on ios is coming on landscape-right). On the original code the pivot point is at the top right corner of the Surface when you do 
```
        canvas.translate(frame.height, frame.width)
        canvas.rotate(270, 0, 0)
```

[image]

The coordinate system (yellow) is like that the orange is where the frame is gonna be renderer, and the surface (blue) is what are you gonna see on the screen. If you do a `canvas.scale(1, -1)` after the code above, it will work, it works fine on android cause the image is being flipped (I am not sure why it is coming flipped or if that is a bug). Or if it is coming flipped on just some android devices

uasghar-smile, 2024-11-25:
I am also getting the same issue here https://github.com/mrousavy/react-native-vision-camera/issues/3297

Any update?

shaugland, 2024-12-31:
Also getting the same issue, any update on this?

saeidtkh, 2025-01-02:
Any updates here?

frodriguez-hu, 2025-01-03:
@shaugland @saeidtkh Doing what I explained on the comment, works fine. I am not sure if it is well explained or if that is correct, @mrousavy could you clarify on this? I'll be glad on making a pr testing every possibility on the transform function.

PriyaJainDev, 2025-01-08:
Also getting the same issue, any update on this?

Aycom366, 2025-01-18:
> [@shaugland](https://github.com/shaugland) [@saeidtkh](https://github.com/saeidtkh) Doing what I explained on the comment, works fine. I am not sure if it is well explained or if that is correct, [...]

your code works, I'm talking about the switch orientation, however, the camera is not getting mirrored again(it's getting flipped), I'm on android. is there anything you could help us with?

gonzaloripa, 2025-01-21:
If **frame.orientation** is 'landscape-left', try with this code. This fixed the camera getting flipped in my case:

```
case 'landscape-left':
        // rotate two flips on (0,0) origin and move X + Y into view again
        canvas.scale(-1, 1); // Flip the canvas horizontally
        canvas.translate(-frame.height, frame.width); // Adjust the canvas position after flipping
        canvas.rotate(270, 0, 0)
        break
```

rachellauyui, 2025-03-08:
hm.... I get even more confused by all these solutions. So, does it work at all?

AilsonFreire, 2025-03-17:
Strangely, I put console.log in the frame.orientation and it gave me landscape-left even tho the mobile and the app are in portrait mode.  Adding this to the code fixed the black screen on android 
`
case 'landscape-left':
        // rotate two flips on (0,0) origin and move X + Y into view again
        canvas.scale(-1, 1); // Flip the canvas horizontally
        canvas.translate(-frame.height, frame.width); // Adjust the canvas position after flipping
        canvas.rotate(270, 0, 0)
`