margelo/react-native-vision-camera #2952: I am getting react-native-vision-camera:compileDebugKotlin'. error🔧

mgonekar, 2024-06-09:
### How were you trying to build the app?

I have freshly created the react native project and installed it(react-native-vision-camera) I have lowered the Gradle version 8.6 to 8.5 even tried lower version of react-native-vision-camera 3.9.2 still the same error and have imported any single line in App.tsx

### Full build logs

```tsx
> Task :react-native-vision-camera:extractDeepLinksDebug
> Task :react-native-vision-camera:compileDebugLibraryResources
> Task :react-native-vision-camera:processDebugManifest
> Task :react-native-vision-camera:parseDebugLocalResources
> Task :app:processDebugMainManifest
> Task :app:processDebugManifest
> Task :react-native-vision-camera:checkKotlinGradlePluginConfigurationErrors
> Task :react-native-vision-camera:generateDebugRFile
> Task :react-native-vision-camera:generateDebugBuildConfig
> Task :app:mergeDebugResources
> Task :app:javaPreCompileDebug UP-TO-DATE
> Task :app:mergeDebugShaders UP-TO-DATE
> Task :app:compileDebugShaders NO-SOURCE
> Task :app:generateDebugAssets UP-TO-DATE
> Task :react-native-vision-camera:javaPreCompileDebug
> Task :react-native-vision-camera:mergeDebugShaders
> Task :react-native-vision-camera:compileDebugShaders NO-SOURCE
> Task :react-native-vision-camera:generateDebugAssets UP-TO-DATE
> Task :react-native-vision-camera:packageDebugAssets
> Task :app:mergeDebugAssets UP-TO-DATE
> Task :app:compressDebugAssets UP-TO-DATE
> Task :app:desugarDebugFileDependencies
> Task :app:mergeDebugJniLibFolders UP-TO-DATE
> Task :app:checkDebugDuplicateClasses

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: VisionCamera: Frame Processors: OFF!

> Task :react-native-vision-camera:compileDebugKotlin

> Task :react-native-vision-camera:compileDebugKotlin FAILED
> Task :app:processDebugManifestForPackage
> Task :app:mergeExtDexDebug
43 actionable tasks: 28 executed, 15 up-to-date

info 💡 Tip: Make sure that you have set up your development environment correctly, by running npx react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor

e: file:///D:/codes/pos/gonsPos/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:31:79 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///D:/codes/pos/gonsPos/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:36:19 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///D:/codes/pos/gonsPos/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:36:47 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///D:/codes/pos/gonsPos/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:72:73 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 54s
info Run CLI with --verbose flag for more details.
```

### Project dependencies

```json
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.74.2",
    "react-native-vision-camera": "^3.9.2"
  },
```

### VisionCamera Version

3.9.2

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

flessard, 2024-06-09:
@mgonekar Not compatible with RN .74

https://github.com/mrousavy/react-native-vision-camera/pull/2859

mrousavy, 2024-06-10:
```
e: file:///D:/codes/pos/gonsPos/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:31:79 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///D:/codes/pos/gonsPos/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:36:19 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///D:/codes/pos/gonsPos/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:36:47 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///D:/codes/pos/gonsPos/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:72:73 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
```

Did you read the errors?

There are multiple issues with those errors posted here in the repo, VisionCamera is not yet fully RN 0.74 compatible.

There are workarounds, but a fully featured support will use a different technology/implementation for the JSI bindings.

closed 2024-06-10 by mrousavy

mgonekar, 2024-06-12:
what version does it support of RN

mrousavy, 2024-06-12:
RN 0.73