margelo/react-native-vision-camera #2954: Camera preview doesn't show the full image even with resizeMode='contain'

tomerh2001, 2024-06-09:
### What's happening?

<Uploading video from mobile>

Notice the "Razer" logo on my laptop; in the preview, it's almost at the bottom, but when I grab a photo, I can see the sticker below it and nearly the keyboard

### Reproduceable Code

```tsx
import ImageUtils from 'react-native-photo-manipulator';

const [isCameraInitialized, setIsCameraInitialized] = useState(Platform.OS !== 'ios');
const [cameraLayout, setCameraLayout] = useState({});
const [isCameraActive, setIsCameraActive] = useState(false);
const [reviewImage, setReviewImage] = useState(null);

const {hasPermission, requestPermission} = useCameraPermission();
const device = useCameraDevice('back');
const format = useCameraFormat(device, Templates.FrameProcessing);
const camera = useRef();

const frameProcessor = useSkiaFrameProcessor(frame => {
	'worklet';

	frame.render();
	const paint = Skia.Paint();
	paint.setColor(Skia.Color(Colors.yellow));
	paint.setStyle(PaintStyle.Stroke);
	paint.setStrokeWidth(8);

	frame.drawRect({
		x: RECT_PERCENTAGE.x1 * frame.width,
		y: RECT_PERCENTAGE.y1 * frame.height,
		width: (RECT_PERCENTAGE.x2 - RECT_PERCENTAGE.x1) * frame.width,
		height: (RECT_PERCENTAGE.y2 - RECT_PERCENTAGE.y1) * frame.height,
	}, paint);
},
[
	isCameraInitialized,
	isCameraActive,
	setIsCameraActiveWorklet,
]);

async function grabImage(autoDetected) {
	if (!isCameraInitialized || !isCameraActive || !camera?.current) {
		return;
	}

	try	{
		const photo = await camera?.current.takePhoto({enableShutterSound: true});
		const croppedPhotoPath = await ImageUtils.crop('file://' + photo.path, {
			x: RECT_PERCENTAGE.x1 * photo.width,
			y: 0,
			width: photo.width,
			height: photo.height,
		});
		const base64 = await Image.compress(croppedPhotoPath, {
			input: 'uri',
			returnableOutputType: 'base64',
			quality: 0.9,
		});

		setIsCameraActiveWorklet(false);
		setReviewImageWorklet({autoDetected, base64, ...photo});
	} catch (error) {
		console.error(error);
		crashlyticsRecordErrorWorklet(new Error(error));
	}
}

[... 223 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 7 Pro

### VisionCamera Version

4.1.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

j-jonathan, 2024-06-11:
Hi!
Probably related to https://github.com/mrousavy/react-native-vision-camera/pull/2964
You can test it to see if it solves your problem.

mrousavy, 2024-06-12:
This has been fixed by #2964 👍

closed 2024-06-12 by mrousavy

mlhtnc, 2024-06-20:
Upgraded to 4.3.2 (latest now) and the preview issue is fixed for me 👍🏻