margelo/react-native-vision-camera #2966: 🐛

chrispader, 2024-06-11:
### What's happening?

aoishdawd

### Reproduceable Code

```tsx

```

### Relevant log output

```shell
.,>>../,.,.
```

### Camera Device

```json
12345667788990
```

### Device

macbook pro xs

### VisionCamera Version

-2.6°

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-06-11 by maintenance-hans[bot]

mrousavy, 2024-06-11:
Thank you Hans for your service

chrispader, 2024-06-11:
Thank you Hans! You're such a valuable part of our community ❤️

mrousavy, 2024-06-11:
Thank you @maintenance-hans

mrousavy, 2024-06-11:
thank you hans 👍

mrousavy, 2024-06-12:
danke hans