margelo/react-native-vision-camera #2967: 🐛

Szymon20000, 2024-06-11:
### What's happening?

.

### Reproduceable Code

```tsx
n/a
```

### Relevant log output

```shell
.
```

### Camera Device

```json
n/a
```

### Device

.

### VisionCamera Version

.

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-06-11 by maintenance-hans[bot]

mrousavy, 2024-06-11:
Thank you Hans for your service

mrousavy, 2024-06-11:
thank you hans

mrousavy, 2024-06-11:
thank you hans 👍