margelo/react-native-vision-camera #2969: 🐛 onStatusChanged override function missing for android versions before api 29

markgray123, 2024-06-12:
### What's happening?

The app crashes while trying to take picture with enableLocation={true} on android api version 24. I did some research and found the cause and fix. This is caused due to onStatusChanged function getting removed in android api version 29. To support legacy api versions (before android 10), we need to override the onStatusChanged function. Please add the following code to fix it. Thanks.

The fix: (MetadataProvider.kt)

import android.location.LocationProvider
import android.os.Bundle

override fun onStatusChanged(provider: String, status: Int, extras: Bundle) {
    var statusStr = "available."
    if (status == LocationProvider.OUT_OF_SERVICE) {
      statusStr = "out of service."
    } else if (status == LocationProvider.TEMPORARILY_UNAVAILABLE) {
      statusStr = "temporarily unavailable."
    }
    Log.i(TAG, "Provider $provider is $statusStr")
  }

### Reproduceable Code

```tsx
<Camera enableLocation={true} />
```

### Relevant log output

```shell
2022-02-03 13:05:26.458 2383-2383/? E/AndroidRuntime: FATAL EXCEPTION: main Process: com.casontek.farmconnect, PID: 2383 java.lang.AbstractMethodError: abstract method "void android.location.LocationListener.onStatusChanged(java.lang.String, int, android.os.Bundle)" at android.location.LocationManager$ListenerTransport._handleMessage(LocationManager.java:299) at android.location.LocationManager$ListenerTransport.-wrap0(Unknown Source:0) at android.location.LocationManager$ListenerTransport$1.handleMessage(LocationManager.java:237) at android.os.Handler.dispatchMessage(Handler.java:106) at android.os.Looper.loop(Looper.java:164) at android.app.ActivityThread.main(ActivityThread.java:6524) at java.lang.reflect.Method.invoke(Native Method) at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:451) at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:888) 2022-02-03 14:05:09.685 13052-13052/? E/AndroidRuntime: FATAL EXCEPTION: main
```

### Camera Device

```json
{}
```

### Device

Samsung galaxt s6

### VisionCamera Version

4.2.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

markgray123, 2024-06-12:
@mrousavy  Please review and merge the above fix.

mrousavy, 2024-06-12:
Please create a PR instead of posting code in issues.

closed 2024-06-12 by mrousavy