margelo/react-native-vision-camera #2981: How to get preview size (height and width) on resizeMode="contain"

MilanRadon, 2024-06-12:
### Question

How to get preview size (height and width) on resizeMode="contain"?

### VisionCamera Version

"^3.9.2"

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-06-12:
Milan - please read the docs before posting issues here in the repo. This is the third issue you created within 3 hours that could've been answered immediately if you just read the docs.

MilanRadon, 2024-06-13:
@mrousavy you don't write anywhere how to get the size of the preview... how to determine the size, yes...

MarianAlexandruAlecu, 2024-08-19:
Yep, no way to get the size of the preview without using a frame processor