margelo/react-native-vision-camera #2982: 💭 Get "Missing purpose string" Apple violation when VCEnableLocation is false

ppanwin10, 2024-06-13:
### Question

Hi,

I still receive the 'Missing purpose string' violation notice even after disabling the location service as per the documentation.

The Apple review notice:
_"ITMS-90683: Missing purpose string in Info.plist - Your app’s code references one or more APIs that access sensitive user data, or the app has one or more entitlements that permit such access. The Info.plist file for the “TalkingCamera.app” bundle should contain a NSLocationWhenInUseUsageDescription key with a user-facing purpose string explaining clearly and completely why your app needs the data. If you’re using external libraries or SDKs, they may reference APIs that require a purpose string. While your app might not use these APIs, a purpose string is still required."_

### What I tried

Podfile:

  ...
  target 'PackageName' do
    config = use_native_modules!
  
    use_frameworks! :linkage => :static
    $RNFirebaseAsStaticFramework = true
    $RNGoogleMobileAdsAsStaticFramework = true
  
    $VCEnableLocation = false
  
  ...

### VisionCamera Version

4.0.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-06-13:
Can you ask Apple which key/method signature is offending their tests? I compile out the `CLLocationManager.authorizationStatus` API which I thought was enough, but maybe they even check for more than that?

This is the code I have:

https://github.com/mrousavy/react-native-vision-camera/blob/0a74a84f3dd52c3aeae06002c9c4dcb8f620c1af/package/ios/Core/LocationProvider.swift#L8-L53

Can you ask them which part of that should be compiled out so that it doesn't offend their tests anymore?

ppanwin10, 2024-06-14:
Great news! My app just passed the release review. It seems that the "Missing purpose string" warning only appears during the TestFlight review. So, if anyone else encounters this issue, you can ignore it and proceed with your release.

closed 2024-06-14 by ppanwin10