margelo/react-native-vision-camera #2984: 🐛 onOutputOrientationChanged wrong value on tablets

jaumereg, 2024-06-13:
### What's happening?

This event emits `"portrait"` when it should be `"landscape"` and viceversa. (See screenshots and console warns)

[image]
[image]

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

<VisionCamera
    style={StyleSheet.absoluteFill}
    device={device}
    isActive
    ref={cameraRef}
    photo
    onOutputOrientationChanged={setOrientation}
/>
```

### Relevant log output

```shell
.
```

### Camera Device

```json
{                                                 
    "formats": [],                                         
    "sensorOrientation": "landscape-left",                 
    "hardwareLevel": "limited",                            
    "maxZoom": 1,                                          
    "minZoom": 1,                                          
    "maxExposure": 6,                                      
    "supportsLowLightBoost": false,                        
    "neutralZoom": 1,                                      
    "physicalDevices": [                                   
        "ultra-wide-angle-camera"                            
    ],                                                     
    "supportsFocus": false,                                
    "supportsRawCapture": false,                           
    "isMultiCam": false,                                   
    "minFocusDistance": 999.999985098839,                  
    "minExposure": -6,                                     
    "name": "10 (BACK) androidx.camera.camera2",           
    "hasFlash": false,                                     
    "hasTorch": false,                                     
    "position": "back",                                    
    "id": "10"                                             
}
```

### Device

Pixel Tablet API 34

### VisionCamera Version

4.2.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-06-13:
> No relevant outputs on console.

I need the log output. I need to know what happens on the native side and what orientations are computed. If you leave this out, I cannot help you. If you add this, we can reopen the issue.

jaumereg, 2024-06-13:
The log output is what happens when going from portrait (selfie camera left) to landscape (selfie camera top)

mrousavy, 2024-06-13:
When you say "Preview" in your issue, you mean "Captured Photo", not "Preview" right? The term "Preview" means live preview view for me.

jaumereg, 2024-06-14:
@mrousavy Exactly. I've updated the labelling to avoid confusion.

mrousavy, 2024-07-12:
Hm, to be honest I don't think there's anything I'm doing wrong. I think this is either a bug of the Android Emulator you are using, or the Camera2 HAL implementation of that Tablet, or CameraX - but there's no explicit portrait handling on my end.

closed 2024-07-12 by mrousavy