margelo/react-native-vision-camera #2991: 🐛 system/camera-is-restricted Camera functionality is not available because it has been restricted by the operating system, possibly due to a device policy.

iliapnmrv, 2024-06-14:
### What's happening?

Not sure if it is related to library but got an error in [Sentry](https://sentry.vibeapp.ru/share/issue/c277872cd5a940e4a2ddccacba42830b/)

> system/camera-is-restricted: Camera functionality is not available because it has been restricted by the operating system, possibly due to a device policy.

Any suggestions why can this happen?

### Reproduceable Code

```tsx
const device = useCameraDevice('back', {
    physicalDevices: [
      'ultra-wide-angle-camera',
      'telephoto-camera',
      'wide-angle-camera',
    ],
  });

<ReanimatedCamera
  format={format}
  zoom={device.neutralZoom}
  device={device}
  isActive={isCameraActive}
  onError={onError}
  codeScanner={codeScanner}
  animatedProps={cameraAnimatedProps}
  ref={cameraRef}
/>
```

### Relevant log output

```shell
no logs, got from sentry
```

### Camera Device

```json
{
"id": 0,
"devices": "wide-angle-camera",
"name": "0 (BACK) androidx.camera.camera2"
}
```

### Device

Samsung SM-G996B, Android 14

### VisionCamera Version

4.3.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

iliapnmrv, 2024-06-14:
could not add logs because its not a local bug

jkaufman, 2024-06-16:
I’d expect this from either MDM or parental controls.

mrousavy, 2024-06-18:
> I’d expect this from either MDM or parental controls.

Exactly. Nothing I can do.

closed 2024-06-18 by mrousavy

rohit-jindal018, 2024-07-18:
@mrousavy We are also facing the issue in Samsung devices and it gets corrected on killing the Application.

iliapnmrv, 2024-08-15:
@rohit-jindal018 hey! did you manage to solve it? I use barcode scanner exactly from example app.

in sentry app just went from background state to foreground. In sentry its also mostly samsung devices

https://sentry.vibeapp.ru/share/issue/c277872cd5a940e4a2ddccacba42830b/

ngocleexuan, 2024-08-20:
I think this problem is due to AppState, you only need to handle it when the application is in active mode for the camera to be activated.

import React, { useCallback, useEffect } from 'react';
import { NativeEventSubscription, AppState, AppStateStatus } from 'react-native';

interface useAppStateCheckProps {
  setAppStateStatus: React.Dispatch<React.SetStateAction<AppStateStatus>>;
}

export default function useAppStateCheck(props: useAppStateCheckProps) {
  const { setAppStateStatus } = props;

  const handleAppStateChange = useCallback(async (nextAppState: AppStateStatus) => {
    setAppStateStatus(nextAppState);
  }, [setAppStateStatus]);

  useEffect(() => {
    let eventListener: NativeEventSubscription;
    eventListener = AppState.addEventListener('change', handleAppStateChange);

    return () => {
      eventListener && eventListener.remove();
    };
  }, [handleAppStateChange]);
}

(hasPermission && appStateStatus) && (
                <Camera
                  photo
                  ref={cameraRef}
                  device={devices}
                  outputOrientation={"device"}
                  preview
                  isActive
                  zoom={0}
                  style={StyleSheet.absoluteFillObject}
                />
              )

rohit-jindal018, 2024-08-20:
> @rohit-jindal018 hey! did you manage to solve it? I use barcode scanner exactly from example app.
> 
> in sentry app just went from background state to foreground. In sentry its also mostly samsung devices
> 
> https://sentry.vibeapp.ru/share/issue/c277872cd5a940e4a2ddccacba42830b/

Hi, Yes we were able to fix it temporarily by unmounting the component whenever the app went into background. issue occured only when app went into background with scanner in opened state. So if user was on scanner page and he took the app into background, we unmounted the component and took him to the previous screen

ngocleexuan, 2024-08-20:
> > @rohit-jindal018Này! Bạn đã giải quyết được chưa? Tôi sử dụng máy quét mã vạch chính xác từ ứng dụng ví dụ.
> > trong ứng dụng Sentry chỉ chuyển từ trạng thái nền sang trạng thái nền trước. Trong Sentry, nó cũng chủ yếu là các thiết bị Samsung
> > https://sentry.vibeapp.ru/share/issue/c277872cd5a940e4a2ddccacba42830b/
> 
> Xin chào, Vâng, chúng tôi có thể sửa lỗi tạm thời bằng cách hủy gắn thành phần bất cứ khi nào ứng dụng chuyển sang chế độ nền. Sự cố chỉ xảy ra khi ứng dụng chuyển sang chế độ nền với máy quét ở trạng thái mở. Vì vậy, nếu người dùng đang ở trang máy quét và anh ta đưa ứng dụng vào chế độ nền, chúng tôi đã hủy gắn thành phần và đưa anh ta đến màn hình trước đó

After testing on Samsung and Xiaomi devices, I use the isActive field, to handle the above error, combined with AppState to capture the application state.
isActive={appStateStatus === "active"}

iliapnmrv, 2024-08-22:
@rohit-jindal018 could you please also share your code?
@ngocleexuan I have the same logic, i copied it from this library. It fixed for me on almost all devices except some samsungs
https://github.com/mrousavy/react-native-vision-camera/blob/main/package/example/src/hooks/useIsForeground.ts

jordanhippotec, 2024-09-18:
> After testing on Samsung and Xiaomi devices, I use the isActive field, to handle the above error, combined with AppState to capture the application state. isActive={appStateStatus === "active"}

Work for me! thank you

Tobbe, 2025-02-10:
This hook works great with the `isActive` prop on the `<Camera ... />` component

```tsx
import {useEffect, useState} from 'react'
import {AppState, AppStateStatus} from 'react-native'

export function useAppState() {
  const currentState = AppState.currentState
  const [appState, setAppState] = useState(currentState)

  useEffect(() => {
    function onChange(newState: AppStateStatus) {
      setAppState(newState)
    }

    const subscription = AppState.addEventListener('change', onChange)

    return () => {
      subscription.remove()
    }
  }, [])

  return appState
}
```
From https://github.com/react-native-community/hooks/blob/main/src/useAppState.ts

I use it like this
```tsx
<Camera
  style={StyleSheet.absoluteFill}
  device={device}
  isActive={appState === "active"}
  codeScanner={codeScanner}
/>
```

Tobbe, 2025-02-10:
No, I'm still on old arch. And I've only tested this with my own Pixel 8 Pro and Pixel 6A phones. Not any Samsung devices.

iliapnmrv, 2025-02-10:
Somehow I also have errors in old arch but I've had the same logic for months 🤷‍♂️
But in logs I have firstly this:
08:33:32 
```js
{
  state: background
}
```
and then
08:33:32 
```
system/camera-is-restricted: Camera functionality is not available because it has been restricted by the operating system, possibly due to a device policy.
```

useIsForeground looks like [this](https://github.com/mrousavy/react-native-vision-camera/blob/main/example/src/hooks/useIsForeground.ts) 

Here are the logs from Sentry

| Timestamp                        | Level | Message                                                                                                           | Category     | Info                                             |
|----------------------------------|-------|-------------------------------------------------------------------------------------------------------------------|--------------|--------------------------------------------------|
| 2025-02-10T14:24:19.802000Z       | info  | Output orientation changed! PORTRAIT                                                                              | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:19.803000Z       | info  | invokeOnOutputOrientationChanged(PORTRAIT)                                                                        | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:19.862000Z       | info  | Output orientation changed! LANDSCAPE_LEFT                                                                        | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:19.863000Z       | info  | invokeOnOutputOrientationChanged(LANDSCAPE_LEFT)                                                                  | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:19.883000Z       | info  | Output orientation changed! PORTRAIT                                                                              | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:19.884000Z       | info  | invokeOnOutputOrientationChanged(PORTRAIT)                                                                        | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:19.944000Z       | info  | Output orientation changed! LANDSCAPE_RIGHT                                                                       | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:19.945000Z       | info  | invokeOnOutputOrientationChanged(LANDSCAPE_RIGHT)                                                                 | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:20.003000Z       | info  | Output orientation changed! PORTRAIT                                                                              | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:20.003000Z       | info  | invokeOnOutputOrientationChanged(PORTRAIT)                                                                        | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:20.083000Z       | info  | Output orientation changed! LANDSCAPE_LEFT                                                                        | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:20.084000Z       | info  | invokeOnOutputOrientationChanged(LANDSCAPE_LEFT)                                                                  | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:20.123000Z       | info  | Output orientation changed! PORTRAIT                                                                              | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:20.124000Z       | info  | invokeOnOutputOrientationChanged(PORTRAIT)                                                                        | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:20.145000Z       | info  | Output orientation changed! LANDSCAPE_LEFT                                                                        | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:20.146000Z       | info  | invokeOnOutputOrientationChanged(LANDSCAPE_LEFT)                                                                  | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:20.262000Z       | info  | Output orientation changed! PORTRAIT                                                                              | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:20.262000Z       | info  | invokeOnOutputOrientationChanged(PORTRAIT)                                                                        | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:20.502000Z       | info  | Output orientation changed! LANDSCAPE_LEFT                                                                        | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:20.503000Z       | info  | invokeOnOutputOrientationChanged(LANDSCAPE_LEFT)                                                                  | Logcat       | tag: CameraView         [... cut ...]
| 2025-02-10T14:24:28.418000Z       | info  | Camera State: CLOSING (has error: true)                                                                           | Logcat       | tag: CameraSession                               |
| 2025-02-10T14:24:28.418000Z       | error | invokeOnError(...):                                                                                               | Logcat       | tag: CameraView                                  |
| 2025-02-10T14:24:28.420000Z       | info  | Camera State: CLOSED (has error: false)                                                                           | Logcat       | tag: CameraSession                               |

Tobbe, 2025-02-10:
@iliapnmrv You're making me worried we'll start seeing the exact same problem once we get more users of our app 😬 
I think the only way you're going to get any help with this is to pay @mrousavy to help you.

iliapnmrv, 2025-02-10:
Sure, our company tries to support Marc as much as possible. I've asked to help, for now we should just wait. This is strange that I'm the only one getting this error
Anyways this issue is not fatal, users only report that you need to reenter the camera screen. If camera is not a big part of your app, your users wouldn't even notice it

mrousavy, 2025-02-10:
Hey @iliapnmrv thanks - what company do you work for?

Tobbe, 2025-02-11:
> Sure, our company tries to support Marc as much as possible.

Thank you @iliapnmrv! You and you're company are doing us all a favor!

iliapnmrv, 2025-02-11:
Hi! After an entire day of testing these are my results
- I can consistently get this warning on my `Samsung S9 (Android 10)`, even on basic example app
- Sentry has a bunch of different manufacturers but QA with Xiaomi phone was unable to reproduce the bug'

The steps are
1. Open camera, give all permissions
2. Send app to background
3. Use phone as normal (messengers, streaming platforms etc.) for around a minute
4. Go back to app, there you'll see the warning
5. But for some reason (not even related to library as I understand) in my production app I get black screen, while in example app, camera reloads and works just fine

For now I'm trying to figure out what module in my app causes this error

@Tobbe I don't think this issue is related to your app, so no worries 😊

sayan-avantai, 2025-04-19:
I am getting this error when I am trying to grant the two different permissions, one is camera and another is location. then this error is coming. 
if i am asking for one permission like the camera, then using useEffect I can turn on the camera by passing to isActive. 
No issue for one permission.