margelo/react-native-vision-camera #2993: CodeScanner does not recognize Datamatrix on a black background

vkoktashev, 2024-06-14:
### Question

I have a Datamatrix with inverted colors - black background and white characters. CodeScanner cannot recognize this. However, if you invert the colors in the photo, everything works fine. Original photo:
[image]
Inverted photo:
[image]

Can I use any FrameProccessor? And in this case, will CodeScanner work with an already processed image?
Or maybe there is a way to configure the recognition of codes of this type using the react-native-vision-camera configuration?

### VisionCamera Version

4.3.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-06-18:
Hey - the CodeScanner is just the system API for code scanning, it is very basic and lightweight.

If you need something more advanced, use a custom Frame Processor Plugin.

KarimovAbdullo, 2024-08-21:
@mrousavy adnorid doesn't read these codes for me, ok on ios