margelo/react-native-vision-camera #2994: 🐛 photo.orientation is wrong when outputOrientation = preview on Android

Titozzz, 2024-06-14:
### What's happening?

Unexpected orientation on Android using outputOrientation = preview
iOS works fine

https://github.com/Titozzz/debug-orientation-vision-camera
I kept the example to a minimum
Just run the app and take a picture.

### Reproduceable Code

```tsx
import { StatusBar } from 'expo-status-bar';
import React, { useEffect, useRef } from 'react';
import { StyleSheet, Text, View, Button } from 'react-native';
import {
  Camera,
  useCameraDevice,
  useCameraPermission,
} from 'react-native-vision-camera';
import Exif from 'react-native-exif';

// Exif mapping:
// 1: 0 degrees, no rotation
// 2: 0 degrees, mirrored
// 3: 180 degrees, upside down
// 4: 180 degrees, upside down and mirrored
// 5: 90 degrees, mirrored and rotated to the right
// 6: 90 degrees, rotated to the right
// 7: 90 degrees, mirrored and rotated to the left
// 8: 90 degrees, rotated to the left

export default function App() {
  const device = useCameraDevice('front');
  const camera = useRef(null);

  const { hasPermission, requestPermission } = useCameraPermission();
  useEffect(() => {
    requestPermission();
  }, []);

  return (
    <View style={styles.container}>
      {hasPermission && device && (
        <Camera
          isActive
          device={device}
          photo
          ref={camera}
          style={StyleSheet.absoluteFill}
          outputOrientation="preview"
        ></Camera>
      )}
      <Button
        title="take picture"
        onPress={async () => {
          const photo = await camera.current.takePhoto();
          const exif = await Exif.getExif(photo.path);
          console.log(photo.orientation);
          console.log(exif.Orientation);
          // photo.orientation is portrait but exif.Orientation is 7 on Android. !!! Error !!!
          // on iOS it works fine: Photo.orientation is landscape-left and exif.Orientation is 5.
        }}
      ></Button>
    </View>
  );
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
  },
});
```

### Relevant log output

```shell
// Not relevant
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 0.8958333134651184,
  "maxExposure": 24,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "wide-angle-camera",
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": true,
  "minFocusDistance": 0,
  "minExposure": -24,
  "name": "1 (FRONT) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
 LOG  {
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 0.8958333134651184,
  "maxExposure": 24,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "wide-angle-camera",
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": true,
  "minFocusDistance": 0,
  "minExposure": -24,
  "name": "1 (FRONT) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Pixel 7

### VisionCamera Version

4.3.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

PanamaDonne, 2024-06-19:
Yes, I see the same issue on Android +  iPhone8 & down. Downgraded and it worked.

USAMA327, 2024-07-23:
> Yes, I see the same issue on Android + iPhone8 & down. Downgraded and it worked.

Can you please tell me about the exact version?

frodriguez-hu, 2024-08-15:
> Yes, I see the same issue on Android + iPhone8 & down. Downgraded and it worked.

To what version have you downgraded it?

PanamaDonne, 2024-08-15:
> > Yes, I see the same issue on Android + iPhone8 & down. Downgraded and it worked.
> 
> To what version have you downgraded it?

4.0.1

PanamaDonne, 2024-08-15:
> > Yes, I see the same issue on Android + iPhone8 & down. Downgraded and it worked.
> 
> Can you please tell me about the exact version you are using?

4.0.1

david-littlefield, 2025-10-20:
any update on this?