margelo/react-native-vision-camera #2998: 💭 Obtaining camera calibration data

gameveloster, 2024-06-15:
### Question

How can I access values for intrinsic/extrinsic matrix and lens distortions? Similar to those exposed by `AVCameraCalibrationData`. 

Thank you

### VisionCamera Version

4.3.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

jkaufman, 2024-06-16:
I believe this requires the use of depth image APIs. Not currently exposed, but something I’ve been considering adding.

mrousavy, 2024-06-18:
Why do you need that?

jkaufman, 2024-06-18:
It's valuable for computer vision tasks and, specifically, pose estimation.

mrousavy, 2024-06-20:
Okay well I played around with it just a week ago and actually exposed the calibration matrix in the frame, but I decided to scrap that again since I wasn't sure why you'd need it and it was a lot of code.

Either way, I very likely won't build this unless there is more demand (or $$$), but please create a feature request.

closed 2024-06-20 by mrousavy