margelo/react-native-vision-camera #2999: 🐛  Crash in Android release build

malkawi00, 2024-06-16:
### What's happening?

when I try to run the app from react native metro work fine , but when I make release APK , I have an issue app crash when open the camera , is there any solution 

### Reproduceable Code

```tsx
export const CustomCamera = forwardRef((props, ref) => {
  const {
    isVisible = false,
    frameProcessor = null,
    cameraType = 'back',
    zoom = 1,
    torch = 'off',
  } = props;
  const device = useCameraDevice(cameraType);
  if (device == null) {
    return (
      <View style={styles.loadingView}>
        <Text style={styles.loadingText}>Loading</Text>
      </View>
    );
  }
  return (
    <Camera
      device={device}
      isActive={isVisible}
      zoom={zoom}
      style={EStyleSheet.absoluteFill}
      frameProcessor={frameProcessor}
      ref={ref}
      pixelFormat={'yuv'}
      photo={true}
      video={true}
      torch={isVisible ? torch : 'off'}
      photoQualityBalance="balanced">
      {props.children}
    </Camera>
  );
});

this code use custom camera 
 const frameLogicJS = Worklets.createRunOnJS((scannedData, type) => {
    const result = scannedData.data || {};
    setSdkResult(scannedData);

    // console.log('scannedData--', JSON.stringify(scannedData, null, 2));
   
  });

 

  const frameProcessor = useFrameProcessor(
    frame => {
      'worklet';
      runAsync(frame, () => {
        'worklet';
        let scannedData = scanDocument(frame);
        if (isFrontFrameProcessor.value) {
          frameLogicJS(scannedData, 'front');
        } else {
          frameLogicJS(scannedData, 'back');
        }
      });
    },
    [isFrontFrameProcessor],
  );
I try to remove runAsync it work fine but with lag in camera 

const frameProcessor = useFrameProcessor(
    frame => {
      'worklet';
        let scannedData = scanDocument(frame);
        if (isFrontFrameProcessor.value) {
          frameLogicJS(scannedData, 'front');
        } else {
          frameLogicJS(scannedData, 'back');
        }
    },
    [isFrontFrameProcessor],
  );
```

### Relevant log output

```shell
"react-native-vision-camera": "4.3.1",
"react-native-worklets-core": "^1.3.3",
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 0.6000000238418579,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": true,
  "minFocusDistance": 0,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung s23 , OPPO A5s , Honor 90 REA-NX9

### VisionCamera Version

4.3.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

aguedob, 2024-06-18:
Why are you nesting runAsync with runAtTargetFps? I think you should use one or the other.

malkawi00, 2024-06-19:
@aguedob  if I use runAsync without runAtTargetFps , the camera lagging , when I run application form react-native work fine there is no issue , just when I make APK I have crash issue

mrousavy, 2024-06-24:
> Why are you nesting runAsync with runAtTargetFps? I think you should use one or the other.

No you can also nest them.

closed 2024-07-12 by mrousavy