margelo/react-native-vision-camera #3004: 🐛 Crash in iOS release build when using SkiaFrameProcessor

fnKode, 2024-06-17:
### What's happening?

I'm trying to use the Skia frame processor to draw a rectangle in the camera preview. However, my app crashes when I try to open the camera.

This only happens if I create a release build of my app and then download it via TestFlight. Everything works fine if I build directly to my device and then run the app using Expo. 

Also, this issue seems to only occur when I'm using the `SkiaFrameProcessor`. If I change to `useFrameProcessor` and remove the Skia related code, then the app works fine.

### Reproduceable Code

```tsx
export default function App() {
    const device = useCameraDevice("back");
    const { hasPermission, requestPermission } = useCameraPermission();

    const frameProcessor = useSkiaFrameProcessor(frame => {
        "worklet";
        frame.render();
        const rect = Skia.XYWHRect(frame.width / 2, frame.height / 2, 150, 150);
        const paint = Skia.Paint();
        paint.setColor(Skia.Color("red"));
        frame.drawRect(rect, paint);
    }, []);

    if (!hasPermission) {
        requestPermission();
        return <Text>No permission</Text>;
    }
    if (device == null) return <Text>No device</Text>;

    return (
        <Camera
            style={{ flex: 1 }}
            device={device}
            isActive
            frameProcessor={frameProcessor}
            onError={e => console.error(e)}
            outputOrientation="portrait"
        />
    );
}
```

### Relevant log output

```shell
Exception Type:  EXC_BAD_ACCESS (SIGBUS)
Exception Subtype: KERN_PROTECTION_FAILURE at 0x000000016b677ff0
Exception Codes: 0x0000000000000002, 0x000000016b677ff0
VM Region Info: 0x16b677ff0 is in 0x16b674000-0x16b678000;  bytes after start: 16368  bytes before end: 15
      REGION TYPE                 START - END      [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      Stack                    16b5ec000-16b674000 [  544K] rw-/rwx SM=PRV  thread 19
--->  STACK GUARD              16b674000-16b678000 [   16K] ---/rwx SM=NUL  stack guard for thread 5
      Stack                    16b678000-16b700000 [  544K] rw-/rwx SM=PRV  thread 5
Termination Reason: SIGNAL 10 Bus error: 10
Terminating Process: exc handler [905]

Triggered by Thread:  5

Thread 5 name:
Thread 5 Crashed:
0   JavaScriptCore                	0x00000001b0a15e1c JSC::initialize() + 8 (InitializeThreading.cpp:68)
1   JavaScriptCore                	0x00000001b0023f60 JSStringCreateWithUTF8CString + 40 (JSStringRef.cpp:45)
2   MyApp                 	0x00000001050709fc 0x104ad4000 + 5884412
3   MyApp                 	0x00000001053284c0 0x104ad4000 + 8733888
4   MyApp                 	0x000000010533e9bc 0x104ad4000 + 8825276
5   MyApp                 	0x00000001053467dc 0x104ad4000 + 8857564
6   MyApp                 	0x00000001053459c4 0x104ad4000 + 8853956
7   MyApp                 	0x000000010533e928 0x104ad4000 + 8825128
8   MyApp                 	0x00000001053467dc 0x104ad4000 + 8857564
9   MyApp                 	0x00000001053459c4 0x104ad4000 + 8853956
10  MyApp                 	0x000000010533e928 0x104ad4000 + 8825128
11  MyApp                 	0x00000001053467dc 0x104ad4000 + 8857564
12  MyApp                 	0x00000001053459c4 0x104ad4000 + 8853956
13  MyApp                 	0x000000010533e928 0x104ad4000 + 8825128
14  MyApp                 	0x00000001053467dc 0x104ad4000 + 8857564
15  MyApp                 	0x00000001053459c4 0x104ad4000 + 8853956
16  MyApp                 	0x000000010533e928 0x104ad4000 + 8825128
17  MyApp                 	0x00000001053467dc 0x104ad4000 + 8857564
18  MyApp                 	0x00000001053459c4 0x104ad4000 + 8853956
19  MyApp                 	0x000000010533e928 0x104ad4000 + 8825128
20  MyApp                 	0x00000001053467dc 0x104ad4000 + 8857564
... repeats for 500 lines
```

### Camera Device

```json
{
    "hasTorch": true,
    "minZoom": 1,
    "minFocusDistance": 12,
    "sensorOrientation": "landscape-left",
    "hardwareLevel": "full",
    "maxExposure": 8,
    "supportsLowLightBoost": false,
    "physicalDevices": [
      "wide-angle-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "neutralZoom": 1,
    "maxZoom": 16,
    "hasFlash": true,
    "minExposure": -8,
    "name": "Back Camera",
    "formats": [],
    "isMultiCam": false,
    "position": "back"
  }
```

### Device

iPad mini (6th generation) (iOS 17.5.1)

### VisionCamera Version

4.3.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

razr130, 2024-06-19:
Is there any updated on this issue?

tomerh2001, 2024-08-22:
+1 Also having this issue.

It works with regular frame processor, but Skia frame processor (without any logic inside) crashes the app)

tomerh2001, 2024-09-08:
Any updates on this?

tomerh2001, 2024-09-12:
@mrousavy any chance you could take a look please?

mrousavy, 2024-09-16:
Don't really have a lot of free time at the moment.

yasirwahid, 2025-01-01:
@tomerh2001 same happening with me, i'm trying to add color filters. Did you resolve the issue?

tomerh2001, 2025-01-01:
> @tomerh2001 same happening with me, i'm trying to add color filters. Have you resolve the issue?

Unfortunately no, this had costed me a lot of time and in the end I just migrated back to an old working version, with the hope that @mrousavy would eventually have time to look at it.

GrandChieftain, 2025-01-02:
Having the same problem, just using the regular useFrameProcessor + Skia Canvas as sibling to Camera + state for canvas size (because DOM pixels differ from frame pixels) and detection bounding boxes (mapped to Rects). Convoluted but appears to work fine since state updates are reflected with every frame re-render. Good to know older versions still work in case this fails later.