margelo/react-native-vision-camera #3005: 🐛 Cached apps freezer is freezing whole app on Google Pixel with Android 14

medonja, 2024-06-17:
### What's happening?

I have an app that runs a foreground process (enabled with react-native-background-actions) which runs just fine on a Google Pixel device (7 pro, 8 pro, tablet) with Android 14. The react-native-background-actions has been patched to properly work with the new permissions needed for Android 14 foreground services. Thus, my background process just counts every seconds and outputs it to the log output, and it does so until forever. All works as expected, thus - so far, so good.

Now, I add react-native-vision-camera to my package.json. This is the only step I do. I do not initialize the library, I do not make any calls. In fact, the only thing I do is add the library to my package.json, run npm i and build my app for Android and run it on my device. After starting the app, I switch to another app and leave the foreground service to output the second count to the log.

It seems that on some Android 14 devices, such as the Google Pixel phones and tablet, the app gets freezed after about 30 seconds by the cached apps freezer (see logcat output), and the foreground service stops counting up the seconds to the logcat. I have tried an Android 14 emulator as well as an OnePlus with Android 14 phone, where this problem does not occur. A Pixel 7 running Android 13 does not have this problem either. When I move the app's activity to the foreground, the foreground service continues to count where it left off. 

This is very obscure, as it seems that when react-native-vision-camera is installed, the Pixel devices running Android 14 do not notice that there is a foreground service active (even if it clearly is - the app is even visible in the notification bar and the active apps section), and their cached app freezer decides to freeze the app as it incorrectly deems the app to be in a cached state. Even more obscure is that I only installed the package, but did not make any calls to it from within my app's code. Starting the camera as shown in the package/example does not make any change, too.

Any thoughts on this? I have no idea if this happens on other devices than the Pixel with Android 14.

### Reproduceable Code

```tsx
N/A
```

### Relevant log output

```shell
2024-06-17 16:34:16.959 15642-15760 ReactNativeJS           com.umfragetoolandroid14             I  27
2024-06-17 16:34:17.975 15642-15760 ReactNativeJS           com.umfragetoolandroid14             I  28
2024-06-17 16:34:18.989 15642-15760 ReactNativeJS           com.umfragetoolandroid14             I  29
2024-06-17 16:34:20.016 15642-15760 ReactNativeJS           com.umfragetoolandroid14             I  30
2024-06-17 16:34:20.146  1622-2159  ActivityManager         system_server                        D  freezing 15642 com.umfragetoolandroid14

2024-06-17 16:34:26.927 22723-22739 android.vending         com.android.vending                  W  ApkAssets: Deleting an ApkAssets object '<empty> and /data/app/~~0N7SODlW0M28dc-sEi9CIg==/com.umfragetoolandroid14-95f6ke1UqlA3-SdpSib8vw==/base.apk' with 1 weak references
2024-06-17 16:34:46.668  2358-2387  WindowManagerShell      com.android.systemui                 V  Transition requested (#2192): android.os.BinderProxy@b84779 TransitionRequestInfo { type = TO_FRONT, triggerTask = TaskInfo{userId=0 taskId=6705 displayId=0 isRunning=true baseIntent=Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 cmp=com.umfragetoolandroid14/.MainActivity } baseActivity=ComponentInfo{com.umfragetoolandroid14/com.umfragetoolandroid14.MainActivity} topActivity=ComponentInfo{com.umfragetoolandroid14/com.umfragetoolandroid14.MainActivity} origActivity=null realActivity=ComponentInfo{com.umfragetoolandroid14/com.umfragetoolandroid14.MainActivity} numActivities=1 lastActiveTime=117767030 supportsMultiWindow=true resizeMode=1 isResizeable=true minWidth=-1 minHeight=-1 defaultMinSize=220 token=WCT{android.window.IWindowContainerToken$Stub$Proxy@4d37cbe} topActivityType=1 pictureInPictureParams=null shouldDockBigOverlays=false launchIntoPipHostTaskId=-1 lastParentTaskIdBeforePip=-1 displayCutoutSafeInsets=Rect(0, 113 - 0, 0) topActivityInfo=ActivityInfo{702311f com.umfragetoolandroid14.MainActivity} launchCookies=[] positionInParent=Point(0, 0) parentTaskId=-1 isFocused=false isVisible=false isVisibleRequested=false isSleeping=false locusId=null displayAreaFeatureId=1 isTopActivityTransparent=false appCompatTaskInfo=AppCompatTaskInfo { topActivityInSizeCompat=false topActivityEligibleForLetterboxEducation= false isLetterboxDoubleTapEnabled= false topActivityEligibleForUserAspectRatioButton= false topActivityBoundsLetterboxed= false isFromLetterboxDoubleTap= false topActivityLetterboxVerticalPosition= -1 topActivityLetterboxHorizontalPosition= -1 topActivityLetterboxWidth=-1 topActivityLetterboxHeight=-1 isUserFullscreenOverrideEnabled=false cameraCompatControlState=hidden}}, pipTask = null, remoteTransition = null, displayChange = null, flags = 0, debugId = 2192 }
2024-06-17 16:34:46.669  2358-2387  WindowManagerShell      com.android.systemui                 D  onActivityRestartAttempt: ComponentInfo{com.umfragetoolandroid14/com.umfragetoolandroid14.MainActivity}
2024-06-17 16:34:46.670 15642-15642 Choreographer           com.umfragetoolandroid14             I  Skipped 3181 frames!  The application may be doing too much work on its main thread.
2024-06-17 16:34:46.672 15642-15760 ReactNativeJS           com.umfragetoolandroid14             I  31
2024-06-17 16:34:46.672  1622-1859  ActivityManager         system_server                        D  sync unfroze 15642 com.umfragetoolandroid14 for 1
```

### Camera Device

```json
{
                                                                                                      "formats": [],
                                                                                                      "sensorOrientation": "landscape-left",
                                                                                                      "hardwareLevel": "full",
                                                                                                      "maxZoom": 30,
                                                                                                      "minZoom": 0.4944782853126526,
                                                                                                      "maxExposure": 24,
                                                                                                      "supportsLowLightBoost": true,
                                                                                                      "neutralZoom": 1,
                                                                                                      "physicalDevices": [
                                                                                                        "wide-angle-camera",
                                                                                                        "wide-angle-camera",
                                                                                                        "ultra-wide-angle-camera",
                                                                                                        "telephoto-camera",
                                                                                                        "telephoto-camera",
                                                                                                        "telephoto-camera"
                                                                                                      ],
                                                                                                      "supportsFocus": true,
                                                                                                      "supportsRawCapture": false,
                                                                                                      "isMultiCam": true,
                                                                                                      "minFocusDistance": 10.500000100135805,
                                                                                                      "minExposure": -24,
                                                                                                      "name": "0 (BACK) androidx.camera.camera2",
[... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-06-24:
Maybe a security measure because Camera permission code is used?

medonja, 2024-06-24:
Could be, but I don't think so. Eventually I exchanged for https://github.com/teslamotors/react-native-camera-kit , and the described issue did not occur anymore with that library.

closed 2024-06-24 by mrousavy

medonja, 2024-06-24:
Well, this problem is maybe solved on my side by switching to another library, but, technically, it still persists.

mrousavy, 2024-06-24:
I have never seen this issue before and I don't know how this can happen, nor do I have the free time to investigate this.

If you switched to a different library I'm sure this issue will just go stale, so I'm gonna close it and wait for someone else to encounter the same issue.

medonja, 2024-06-24:
> I have never seen this issue before and I don't know how this can happen, nor do I have the free time to investigate this.
> 
> If you switched to a different library I'm sure this issue will just go stale, so I'm gonna close it and wait for someone else to encounter the same issue.

I understand, I am also quite baffled. Oh well...

ouabing, 2024-08-02:
Same issue here, our app plays sound in background. After I switch our camera implementation from react-native-camera to react-native-vision-camera, on Pixel 7 Pro(Android 14), the sound stopped after I switch to background for about 20 seconds.

gsgary, 2024-09-09:
FYI, same problem here too. We came to the same solution by removing the react-native-vision-camera package.

hibinoA, 2024-09-10:
I guess it's the matter of permission for sdk version 34. Have you checked below article?
https://developer.android.com/about/versions/14/changes/fgs-types-required
I think you needs that permission, because react-native-vision-camera uses [compileSdkVersion 34 and targetSdkVersion 34](https://github.com/mrousavy/react-native-vision-camera/blob/06b6384c545a04ef904a19e3f3c6de7b6e6867b1/package/android/build.gradle#L125-L126).
The latest release 13.0.0 of react-native-camera-kit uses [compileSdkVersion 31 and targetSdkVersion 30](https://github.com/teslamotors/react-native-camera-kit/blob/edfa42380f344ffaa92880d3255ea249325f0078/android/build.gradle#L5C5-L8), and react-native-camera uses [compileSdkVersion 29 and targetSdkVersion 29](https://github.com/react-native-camera/react-native-camera/blob/7cdfb6ec127decf1ac5686635afb7f07557becc5/android/build.gradle#L32C5-L32C21).

Ejswarrior, 2024-09-10:
Here is another Android article explaining android 14 behavior changes with cached apps. https://developer.android.com/about/versions/14/behavior-changes-all#cached-apps

hibinoA, 2024-09-11:
Thanks Ejswarrior! 
https://source.android.com/docs/core/perf/cached-apps-freezer#handling-custom-features
> As a workaround, if an app has a process that needs to perform activities while cached, change the process status to non-cached (such as Bound ForeGround Service (BFGS) or foreground) 

It might be important about "Bound", not only starting Foreground Service. We also should bind foreground service. Have you guys ever tried?

hibinoA, 2024-10-01:
I resolve this by workaround. The workaround for me is removing all the code about `ExtensionsManager`. `ExtensionManager` is used for HDR, night mode or some other device-specific features. I guess this is the reason why this problem occurs only on Pixel. This workaround is good for me because I don't need HDR and night mode. I hope this helps you!

medonja, 2025-02-07:
> I resolve this by workaround. The workaround for me is removing all the code about `ExtensionsManager`. `ExtensionManager` is used for HDR, night mode or some other device-specific features. I guess [...]

Honestly, I was so frustrated that I have replaced the library with another one that did not cause this issue, so I can't give feedback if this problem persists @hibinoA, but I hope that your solution is indeed the remedy :)