margelo/react-native-vision-camera #3008: 💭 iOS photo quality vs native camera

rodrigobertolottizinsp, 2024-06-18:
### Question

Fist of all, thanks for the latest updates, they were great. 

I'm working on an app that has big requirements in terms of picture quality and when comparing pictures taken with the example app of the repository against native images, I can see a difference on quality. This difference is not so noticeable without zooming, but once you zoom into some object, there is quite noticeable difference between the two pictures. 

Im attaching a screenshots with the comparison when zooming on the lock and also the picture metadata for both cases. Left if Vision Camera Example, right is iPhone Native. 

There is any way to get the same result on vision camera and in native? 
Or maybe iOS is doing some post processing after capturing? 

Im using back triple camera with iPhone 15 Pro
Although Camera Devices Back Triple Camera says picture is being taken in 4032x3024, as you can see in the screenshot, but actual image size is 2160x3840. 

[image]

Thanks!

### VisionCamera Version

4.1.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

j-jonathan, 2024-06-24:
After capturing, iOS applies post-processing (which can be seen, as the photo takes a little time to load/improve in the gallery right after being taken).

If you want to reproduce the same thing with react-native-vision-camera, you would need to apply algorithms yourself to enhance the quality of the image after taking the shot.

mrousavy, 2024-06-24:
1. There are two different formats used. iOS native Camera seems to shoot in 4284x5712 (5k+), and you selected a 2160x3840 (4k) format. This is a HUGE difference. You should be able to also select the 5k+ format in VisionCamera.
2. I can see that the iOS native Camera shot in P3, and VisionCamera shoots in sRGB. There is currently no option to change that, but I could add it. A bit of effort though.

mrousavy, 2024-06-24:
And also; yes exactly what Jonathan said; I also think that they are doing some post-processing. That's not something a Camera library should do imo.

closed 2024-06-24 by mrousavy