margelo/react-native-vision-camera #3012: 🔧 Failed to archive

nolife08021, 2024-06-20:
### How were you trying to build the app?

i tried run in simulator without any issue, but failed to archive. I got run below actions before archive

minimum iOS version : 14.0
IPHONEOS_DEPLOYMENT_TARGET : 14.0
SWIFT_VERSION: 5.2

```
   rm -rf ./package-lock.json
   rm -rf ./ios/Podfile.lock
   rm -rf ./ios/Pods
   rm -rf ./node_modules
    cd ios
    pod cache clean --all
    pod deintegrate
    pod repo update
    pod update
    pod install

    xcodebuild archive -workspace $project.xcworkspace -scheme $envName -archivePath $outPutDir/${app}.xcarchive CODE_SIGN_IDENTITY="" CODE_SIGNING_REQUIRED=NO CODE_SIGNING_ALLOWED=NO

```

### Full build logs

```tsx
The following build commands failed:
	CompileSwift normal arm64 (in target 'VisionCamera' from project 'Pods')
	CompileSwiftSources normal arm64 com.apple.xcode.tools.swift.compiler (in target 'VisionCamera' from project 'Pods')
(2 failures)
```

### Project dependencies

```json
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.71.17",
    "react-native-reanimated": "^2.12.0",
    "react-native-vision-camera": "^4.0.4",
    "react-native-vision-camera-face-detector": "^1.6.3",
    "react-native-worklets-core": "^1.3.3"
  },
```

### VisionCamera Version

4.0.4

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-06-24:
It looks like you're using an old Xcode/Swift compiler version, because variable bindings are not supported.

This is the new way:

```swift
if let something {
```

This is the old way:

```swift
if let something = something {
```

But I won't change the whole codebase for the old way, so please upgrade your Xcode/Swift compiler to latest. Xcode 15.x.x, Swift 5.9 at least

closed 2024-06-24 by mrousavy

nolife08021, 2024-07-01:
@mrousavy Is there any way or or which version of vision-camera to do it with xcode 13?  plan to use it for face detection. Thanks

usamaabutt, 2025-01-14:
Can we use Xcode 14.2 to run the latest version of react-native-vision-camera lib?