margelo/react-native-vision-camera #3014: 🐛 Landscape mode camera issue

sw-tt-hitendrachavda, 2024-06-21:
### What's happening?

we have implemented the camera in portrait and landscape mode, but when we change the orientation in landscape then camera is not showing proper. 

In android it is working fine in landscape and portrait mode.

### Reproduceable Code

```tsx
<Camera
            ref={cameraRef}
            style={
              orientation === 'PORTRAIT'
                ? styles.flexCameraStyle
                : currentOrientation === 'LANDSCAPE-LEFT'
                ? StyleSheet.absoluteFillObject
                : StyleSheet.absoluteFillObject
            }
            video={true}
            device={cameraType === 'back' ? devicesBack : devicesFront}
            isActive={true}
          />
```

### Relevant log output

```shell
2024-06-21 12:04:29.196576+0530 utrapp[2837:225160] [native] Running application utrapp ({
    initialProps =     {
    };
    rootTag = 1;
})
2024-06-21 12:04:29.232967+0530 utrapp[2837:225160] [native] Initializing SQLitePlugin
2024-06-21 12:04:29.233206+0530 utrapp[2837:225160] [native] Detected docs path: /var/mobile/Containers/Data/Application/D57ECA84-BFB0-4D69-9A3E-C82C8C92709B/Documents
2024-06-21 12:04:29.233344+0530 utrapp[2837:225160] [native] Detected Library path: /var/mobile/Containers/Data/Application/D57ECA84-BFB0-4D69-9A3E-C82C8C92709B/Library
2024-06-21 12:04:29.233466+0530 utrapp[2837:225160] [native] no cloud sync at path: /var/mobile/Containers/Data/Application/D57ECA84-BFB0-4D69-9A3E-C82C8C92709B/Library/LocalDatabase
2024-06-21 12:04:29.233509+0530 utrapp[2837:225160] [unspecified] A NULL string is not a valid group container identifier.
2024-06-21 12:04:29.233558+0530 utrapp[2837:225160] [unspecified] container_create_or_lookup_app_group_path_by_app_group_identifier: client sent invalid parameters
2024-06-21 12:04:30.128022+0530 utrapp[2837:225453] SocketStream read error [0x280e90c80]: 1 54
2024-06-21 12:04:33.158539+0530 utrapp[2837:225449] SocketStream read error [0x280e88140]: 1 54
2024-06-21 12:04:34.699759+0530 utrapp[2837:225462] [javascript] ViewPropTypes will be removed from React Native, along with all other PropTypes. We recommend that you migrate away from PropTypes and switch to a type system like TypeScript. If you need to continue using ViewPropTypes, migrate to the 'deprecated-react-native-prop-types' package.
2024-06-21 12:04:34.707490+0530 utrapp[2837:225462] [javascript] ViewPropTypes will be removed from React Native, along with all other PropTypes. We recommend that you migrate away from PropTypes and switch to a type system like TypeScript. If you need to continue using ViewPropTypes, migrate to the 'deprecated-react-native-prop-types' package.
2024-06-21 12:04:34.845051+0530 utrapp[2837:225462] [native] Unbalanced calls start/end for tag 19
2024-06-21 12:04:34.847175+0530 utrapp[2837:225462] [javascript] Running "utrapp" with {"rootTag":1,"initialProps":{}}
2024-06-21 12:04:34.999714+0530 utrapp[2837:225462] [javascript]  Database Created ::::::::::::::::
2024-06-21 12:04:35.001318+0530 utrapp[2837:225450] [native] target database location: nosync
2024-06-21 12:04:35.001528+0530 utrapp[2837:225450] [native] Opening db in mode READ_WRITE, full path: /var/mobile/Containers/Data/Application/D57ECA84-BFB0-4D69-9A3E-C82C8C92709B/Library/LocalDatabase/utr_app.db3
2024-06-21 12:04:35.007139+0530 utrapp[2837:225450] [native] Database opened
2024-06-21 12:04:35.007212+0530 utrapp[2837:225450] [native] Good news: SQLite is thread safe!
2024-06-21 12:04:35.007234+0530 utrapp[2837:225450] [native] open cb finished ok
2024-06-21 12:04:35.153955+0530 utrapp[2837:225462] [javascript] 'TimezoneDate is', '2024-06-21T12:04:35.148000+05:30'
2024-06-21 12:04:35.200857+0530 utrapp[2837:225462] [javascript] Database opened successfully.
2024-06-21 12:04:35.210591+0530 utrapp[2837:225462] [javascript] 'test333', 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJodHRwOi8vc2NoZW1hcy5taWNyb3NvZnQuY29tL2FjY2Vzc2NvbnRyb2xzZXJ2aWNlLzIwMTAvMDcvY2xhaW1zL2lkZW50aXR5cHJvdmlkZXIiOiJVVFIiLCJodHRwOi8vc2NoZW1hcy54bWxzb2FwLm9yZy93cy8yMDA1LzA1L2lkZW50aXR5L2NsYWltcy9uYW1laWRlbnRpZmllciI6IjRiMWI5YzFkLTljNGEtNDViMy04Y2ExLWI5NTFhNThiYjlhOSIsImh0dHA6Ly9zY2hlbWFzLnhtbHNvYXAub3JnL3dzLzIwMDUvMDUvaWRlbnRpdHkvY2xhaW1zL25hbWUiOiJEZW5ueSBEaWNraW5zb24iLCJMb2dpbk1ldGhvZCI6Im1pbGl0YXJ5IiwiaHR0cDovL3NjaGVtYXMubWljcm9zb2Z0LmNvbS93cy8yMDA4LzA2L2lkZW50aXR5L2NsYWltcy9yb2xlIjoic2VydmljZV9tZW1iZXIiLCJuYmYiOjE3MTg5NDk2OTIsImV4cCI6MTcyMTU3NzY5MiwiaXNzIjoiaHR0cHM6Ly90ZXN0LnV0cmFwcC5vcmciLCJhdWQiOiJhbnkifQ.Blot-2r-IOM-fEAIwzU7O5KARi9gpK3PZj8wUwJ88To'
2024-06-21 12:04:35.446177+0530 utrapp[2837:225462] [javascript] VideoSourceModel created successfully
2024-06-21 12:04:35.446243+0530 utrapp[2837:225462] [javascript] account created successfully
2024-06-21 12:04:35.446262+0530 utrapp[2837:225462] [javascript] book_detail created successfully
2024-06-21 12:04:35.446280+0530 utrapp[2837:225462] [javascript] contact created successfully
2024-06-21 12:04:35.446298+0530 utrapp[2837:225462] [javascript] image created successfully
2024-06-21 12:04:35.446315+0530 utrapp[2837:225462] [javascript] share created successfully
2024-06-21 12:04:35.446335+0530 utrapp[2837:225462] [javascript] story created successfully
2024-06-21 12:04:35.446353+0530 utrapp[2837:225462] [javascript] video created successfully
2024-06-21 12:04:35.446370+0530 utrapp[2837:225462] [javascript] video_frame created successfully
2024-06-21 12:04:35.446387+0530 utrapp[2837:225462] [javascript] video_source created successfully
2024-06-21 12:04:35.446405+0530 utrapp[2837:225462] [javascript] IX_VideoSourceModel_Videoid created successfully
2024-06-21 12:04:35.446423+0530 utrapp[2837:225462] [javascript] IX_account_idme_uuid created successfully
2024-06-21 12:04:35.446444+0530 utrapp[2837:225462] [javascript] IX_account_profile_image_id created successfully
2024-06-21 12:04:35.446466+0530 utrapp[2837:225462] [javascript] IX_book_detail_ResourcePath created successfully
2024-06-21 12:04:35.446489+0530 utrapp[2837:225462] [javascript] IX_contact_account_id created successfully
2024-06-21 12:04:35.446509+0530 utrapp[2837:225462] [javascript] IX_contact_contact_account_id created successfully
2024-06-21 12:04:35.446527+0530 utrapp[2837:225462] [javascript] IX_image_account_id created successfully
[... 87 lines cut ...]
  "hasTorch": true,
  "maxZoom": 16,
  "hardwareLevel": "full"
}
```

### Device

iPhoneX

### VisionCamera Version

16.7.8

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

laxminarayana-capgemini, 2024-06-22:
Hi @sw-tt-hitendrachavda, in new React-native-vision-camera package(4.1.0), they have fixed with the prop 'outputOrientation'. Kindly check it.

mrousavy, 2024-06-24:
> then camera is not showing proper.

No screenshots videos or whatsoever attached. Closing

closed 2024-06-24 by mrousavy