margelo/react-native-vision-camera #3015: 🐛 Unable to record video more than 30 fps

laxminarayana-capgemini, 2024-06-21:
### What's happening?

Hi @mrousavy. The recent update for orientation is working fine. Thanks.
I am trying to record the video in 60 fps. But the output video is only 30 fps. I am attaching the log and code. Kindly suggest me a solution for this.

`<Camera
            style={StyleSheet.absoluteFill}
            device={device}
            isActive={isAppForeground}
            video={true}
            audio={true}
            ref={camera}
            format={format}
            exposure={0}
            zoom={0}
            // orientation={currentOrientation}
            outputOrientation='device'
          />`
`const format = useCameraFormat(device, [
    { videoResolution: { width: 720, height: 1280 } },
    { videoStabilizationMode: 'auto' },
    { fps: 60 },
  ])`
The issue which I am facing is only in Android, in IOS it is working fine. While logging I got the device info, I got this.

### Reproduceable Code

```tsx
<Camera
            style={StyleSheet.absoluteFill}
            device={device}
            isActive={isAppForeground}
            video={true}
            audio={true}
            ref={camera}
            format={format}
            exposure={0}
            zoom={0}
            // orientation={currentOrientation}
            outputOrientation='device'
          />

const format = useCameraFormat(device, [
    { videoResolution: { width: 720, height: 1280 } },
    { videoStabilizationMode: 'auto' },
    { fps: 60 },
  ])
```

### Relevant log output

```shell
This log, I have got when running,

const device = useCameraDevice(isBack ? 'back' : 'front')
console.log("device fps***", device?.formats);

[{"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 1836, "photoWidth": 4080, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}].
```

### Camera Device

```json
[{"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 1836, "photoWidth": 4080, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 1808, "photoWidth": 4000, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 2160, "photoWidth": 3840, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 2448, "photoWidth": 3264, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 1840, "photoWidth": 3264, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 1472, "photoWidth": 3264, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 1728, "photoWidth": 2304, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 1080, "photoWidth": 1920, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 1200, "photoWidth": 1600, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 720, "photoWidth": 1600, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 1080, "photoWidth": 1440, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 960, "photoWidth": 1280, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 720, "photoWidth": 1280, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 768, "photoWidth": 1024, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 720, "photoWidth": 960, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 540, "photoWidth": 960, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 600, "photoWidth": 800, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 480, "photoWidth": 800, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427, "maxFps": 30, "maxISO": 3200, "minFps": 10, "minISO": 50, "photoHeight": 480, "photoWidth": 720, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.85308320798427,  [... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

j-jonathan, 2024-06-24:
On your device format, you can see "maxFps": 30, so it's normal that it doesn't record a 60 fps video.

mrousavy, 2024-06-24:
1. Please don't tag me. 
2. Jonathan is right, your `format` doesn't even support 60 FPS. Please read the docs before posting issues.

closed 2024-06-24 by mrousavy