margelo/react-native-vision-camera #3016: outputOrientation not working in android but working properly in iOS.

gauravrana0-nagarro, 2024-06-24:
### What's happening?

if I add outputOrientation prop it always takes images in the orientation of mobile  but works properly in iOS.
outputOrientation="landscape-right" is not rotating image in android

### Reproduceable Code

```tsx
<Camera
                ref={camera}
                style={StyleSheet.absoluteFill}
                device={device}
                outputOrientation="landscape-right"
                isActive={isFocused}
                photo={true}
                onError={onError}></Camera>
```

### Relevant log output

```shell
06-24 15:53:09.953  6318 23920 I CameraLightSensor_Service: Open camera failed: error 2
06-24 15:53:09.953  6318 23920 I CameraLightSensor_Service: stopTask called, isCameraStopPending : false
06-24 15:53:09.954  6318 23920 I CameraLightSensor_Service: stopBackgroundThread : mBackgroundHandler is set to null
06-24 15:53:10.240  5362  5760 D NetdEventListenerService: DNS Requested by : 100, 10716, 0
06-24 15:53:10.468  5166 23637 E Camera2-FrameProcessorBase: FrameProcessorBase: Error waiting for new frames: Connection timed out (-110)
06-24 15:53:10.506  9219 10881 I SDHMS:C : SIOP:: AP:418(472,80) BAT:386(386,0) USB:0(0,0) CHG:458(458,0) WIFI:433(433,0) LRP:432(432,0) LRF:418(418) LRB:432(432) 
06-24 15:53:10.867  5026 23840 I ExynosCameraFrameManager: [CAM(0)][RUNNING FRAME WORKER]-(workerMain[983]): wait and pop fail, ret(-110)
06-24 15:53:11.028 18022 18053 I .msil.truevalu: Background young concurrent copying GC freed 338042(19MB) AllocSpace objects, 230(4600KB) LOS objects, 34% free, 45MB/69MB, paused 694us,121us total 104.740ms
06-24 15:53:11.494  5362  9315 I EDMNativeHelperService: isCameraEnabled
06-24 15:53:11.494  5362  9315 D EDMNativeHelper: isCameraEnabled
06-24 15:53:11.497  5362  9315 D EDMNativeHelper: checking for camera in EnterpriseDeviceManagerService
06-24 15:53:11.497  5362  9315 D EnterpriseDeviceManagerService: isCameraEnabledNative
06-24 15:53:11.497  5362  9315 D EnterpriseDeviceManagerService: checking for camera in restr
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
```

### Device

samsung m21

### VisionCamera Version

4.3.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

SonuGitCfcs, 2024-06-24:
same issue on Redmi Note 7 Pro device & Some Samsung device

gauravrana0-nagarro, 2024-06-24:
I just installed shadow lens app on my android device. the same issue is in Shadow Lens app also. It stores image in landscape mode if clicked closely and in portrait if clicked from some distance

Marius-Adam, 2024-07-06:
I have same issue on IOS

gabrielbull, 2024-07-11:
We're also experiencing this issue on at least these devices:

-Realme C21-Y, Android Version 11
-Vivo Y27s, Android Version 14

@mrousavy, I believe this bug was introduced in version 4 as well as we didn't have it before.

jkaufman, 2024-07-11:
The ability to set `outputOrientation` to a specific orientation [was removed](https://github.com/mrousavy/react-native-vision-camera/pull/3025).

mrousavy, 2024-07-11:
Set `outputOrientation="preview"`, that should work.

Anahita-Fakhravar, 2024-07-12:
Hi @mrousavy, I tried your solution, but I'm still having issues. After taking a photo, the image preview orientation is correct, but when I save it to the device gallery, it appears sideways. This only happens with pictures taken using the front camera. Any suggestions?
Thanks!

duonghungit24, 2024-11-20:
> Set `outputOrientation="preview"`, that should work.

 not working, please support me