margelo/react-native-vision-camera #3017: 💭 Is it possible to decrease the video resolution on v4?

fontesrp, 2024-06-24:
### Question

We upgraded to Vision Camera version 4 and noticed that Android videos have significantly higher resolution and size than version 3. We've put together a relatively simple setup to demonstrate the difference:

```js
const device = useCameraDevice('front')

const sortedFormats = (device?.formats || [])
  .filter(deviceFormat => deviceFormat.videoWidth)
  .sort(
    (format1, format2) =>
      format1.videoWidth - format2.videoWidth ||
      format1.videoHeight - format2.videoHeight ||
      format1.photoWidth - format2.photoWidth ||
      format1.photoHeight - format2.photoHeight ||
      format1.minFps - format2.minFps
  )

const deviceWithSortedFormats = { ...(device || {}), formats: sortedFormats }

console.log(JSON.stringify(deviceWithSortedFormats))
```

On **version 3**, the output is:

```json
{
  "formats": [
    {
      "autoFocusSystem": "none",
      "fieldOfView": 82.07262253221181,
      "maxFps": 30,
      "maxISO": 3200,
      "maxZoom": 4,
      "minFps": 1,
      "minISO": 32,
      "photoHeight": 240,
      "photoWidth": 320,
      "pixelFormats": ["yuv","native"],
      "supportsDepthCapture": false,
      "supportsPhotoHdr": true,
      "supportsVideoHdr": false,
      "videoHeight": 144,
      "videoStabilizationModes": ["off","off"],
      "videoWidth": 176
    },
    ...
  ],
  "hardwareLevel": "limited",
  "hasFlash": false,
  "hasTorch": false,
  "id": "1",
  "isMultiCam": false,
  "maxExposure": 20,
  "maxZoom": 4,
  "minExposure": -20,
  "minFocusDistance": 0,
  "minZoom": 1,
  "name": "FRONT (1)",
  "neutralZoom": 1,
  "physicalDevices": ["wide-angle-camera"],
  "position": "front",
  "sensorOrientation": "landscape-right",
  "supportsFocus": false,
  "supportsLowLightBoost": false,
  "supportsRawCapture": false
}
```

On version 4, the same device outputs:

```json
{
  "formats": [
    {
      "autoFocusSystem": "contrast-detection",
      "fieldOfView": 82.07262253221181,
      "maxFps": 30,
      "maxISO": 3200,
      "minFps": 8,
      "minISO": 32,
      "photoHeight": 240,
      "photoWidth": 320,
      "supportsDepthCapture": false,
      "supportsPhotoHdr": false,
      "supportsVideoHdr": false,
      "videoHeight": 480,
      "videoStabilizationModes": ["off"],
      "videoWidth": 640
    },
    ...
  ],
  "hardwareLevel": "limited",
  "hasFlash": false,
  "hasTorch": false,
  "id": "1",
  "isMultiCam": false,
  "maxExposure": 20,
  "maxZoom": 4,
  "minExposure": -20,
  "minFocusDistance": 0,
  "minZoom": 1,
  "name": "1 (FRONT) androidx.camera.camera2",
  "neutralZoom": 1,
  "physicalDevices": ["wide-angle-camera"],
  "position": "front",
  "sensorOrientation": "landscape-right",
  "supportsFocus": true,
  "supportsLowLightBoost": false,
  "supportsRawCapture": false
}
```

As you can see, this specific device went from supporting video recordings in 176x144 to a minimum resolution of 640x480. Is it possible to re-enable the smaller resolution or resize the frame to get smaller videos?

### What I tried

We tried reducing the video resolution by playing around with the `useCameraDevice` and `useCameraFormat` hooks and using frame processors with the `vision-camera-resize-plugin` plug-in, but we couldn't achieve smaller videos.

### VisionCamera Version

4.3.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-06-27:
Hi! No, I don't think this is possible in V4 anymore.

CameraX is responsible for providing me with all safe resolutions, and if CameraX doesn't give us a specific resolution then it just doesn't consider it safe to use for the media encoding (recording).

You could create a feature request in the CameraX issue tracker to support more resolutions, I'd be curious what they reply as well.

mrousavy, 2024-06-27:
This is how I get the video sizes for a specific device:
https://github.com/mrousavy/react-native-vision-camera/blob/9ca6643748c0b39aa0d3e4fd4ebfb9c377adc0c8/package/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt#L117

closed 2024-06-27 by mrousavy