margelo/react-native-vision-camera #3018: 🐛 Opening camera in landscape shows rotated preview on some tablets

jaumereg, 2024-06-24:
### What's happening?

On some tablets (specially the ones that have the selfie camera on the longer side like phones) when you open the camera in landscape orientation, the preview is rotated 90 degrees, but if you rotate the tablet to portrait, and rotate it back to landscape, it gets fixed itself.

Here is a video of the issue:
https://github.com/mrousavy/react-native-vision-camera/assets/27198255/8022ef80-c4b6-4e6b-9dde-0301b9094aa1

I have tried to fix it in many ways, like downgrade the version, set orientation manually, remounting the camera, using `format` or `resizeMode` but none of them worked. I believe this is a bug, I have reported a similar issue one week ago, but in this case the orientation output is correct, but the preview is not.

Log output was too long, so I've shortened it, here is the full output:
[log.txt](https://github.com/user-attachments/files/15956788/log.txt)

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

// ...

return (
    <Camera
        style={{ flex: 1 }}
        device={device}
        isActive
        ref={cameraRef}
        photo
    />
)
```

### Relevant log output

```shell
06-24 15:47:31.621   597  1349 I EDMNativeHelperService: isCameraEnabled
06-24 15:47:31.621   404  1521 W QCamera2HWI: put_parameters : X
06-24 15:47:31.622   404  1521 D ShotSingle: startPreview : PreviewWidth=640, PreviewHeight=480
06-24 15:47:31.622   404  1521 I ShotCommon: enablePreviewMsgBy : msg(In:0x200, Out:0x200)
06-24 15:47:31.622   404  1521 W QCamera2HWI: msg_type_enabled : E
06-24 15:47:31.622   404  1521 W QCamera2HWI: msg_type_enabled : X ret 0
06-24 15:47:31.622   404  1521 W QCamera2HWI: enable_msg_type : E, msg type 16
06-24 15:47:31.623   404  1521 W QCamera2HWI: enable_msg_type : X
06-24 15:47:31.623   404  1521 I ShotCommon: enableMsgType : msg(In:0x10, Out:0x81d)
06-24 15:47:31.623   404  1521 V ShotSingle: startPreview(0)
06-24 15:47:31.623   404  1521 W QCamera2HWI: [KPI Perf] start_preview: E PROFILE_START_PREVIEW
06-24 15:47:31.623   404  9671 W QCamera2HWI: preparePreview: E mCameraId=0
06-24 15:47:31.623   404  9671 W QCameraHWI_Mem: alloc: use ION_SYSTEM_HEAP_ID memory: (33554432)
06-24 15:47:31.624   404  9671 W QCameraParameters: getFlipMode: No flip mode for stream type 7
06-24 15:47:31.624   404  9671 W QCameraParameters: [PARM_DBG] getFlipMode : the filp mode of stream type 7 is 0 .
06-24 15:47:31.624   431   431 E mm-camera: server_process_hal_event:__DBG__ E event id=3
06-24 15:47:31.624   431   431 E mm-camera: server_process_hal_event:__DBG__ X sucess event id=3
06-24 15:47:31.624   431  9692 E mm-camera: mct_pipeline_process_set:command=800000a
06-24 15:47:31.625   431   431 E mm-camera: server_process_hal_event:__DBG__ E event id=3
06-24 15:47:31.625   431   431 E mm-camera: server_process_hal_event:__DBG__ X sucess event id=3
06-24 15:47:31.625   431  9692 E mm-camera: mct_pipeline_process_set:command=8000012
06-24 15:47:31.625   431  9692 E mm-camera: mct_stream_start_link:CAM_STREAM_TYPE_METADATA
06-24 15:47:31.625   404  9671 E mm-camera-intf: mm_stream_get_v4l2_fmt: Unknown fmt=101
06-24 15:47:31.626   431   431 E mm-camera: server_process_hal_event:__DBG__ E event id=3
06-24 15:47:31.626   431   431 E mm-camera: server_process_hal_event:__DBG__ X sucess event id=3
06-24 15:47:31.626   431  9692 E mm-camera: mct_pipeline_process_set:command=8000004
06-24 15:47:31.626   404  9671 W QCameraHWI_Mem: alloc: use ION_SYSTEM_HEAP_ID memory: (33554432)
06-24 15:47:31.626   404  9671 E QCamera2HWI: [BeautyShot_dbg] getBufNumRequired =8
06-24 15:47:31.626   404  9671 W QCameraParameters: [PARM_DBG] getFlipMode : the filp mode of stream type 1 is 0 .
06-24 15:47:31.627   404  9671 E QCamera2HWI: [BeautyShot_dbg] getBufNumRequired =8
06-24 15:47:31.627   431   431 E mm-camera: server_process_hal_event:__DBG__ E event id=3
06-24 15:47:31.627   431   431 E mm-camera: server_process_hal_event:__DBG__ X sucess event id=3
06-24 15:47:31.627   431  9692 E mm-camera: mct_pipeline_process_set:command=800000a
06-24 15:47:31.628   431   431 E mm-camera: server_process_hal_event:__DBG__ E event id=3
06-24 15:47:31.628   431   431 E mm-camera: server_process_hal_event:__DBG__ X sucess event id=3
06-24 15:47:31.628   431  9692 E mm-camera: mct_pipeline_process_set:command=8000012
06-24 15:47:31.628   431  9692 E mm-camera: mct_stream_start_link: PREVIEW
[... 264 lines cut ...]
  "hasFlash": false,                               
  "maxExposure": 20,                               
  "name": "0 (BACK) androidx.camera.camera2.legacy"
}
```

### Device

Samsung Galaxy Tab A

### VisionCamera Version

4.3.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

davidburson, 2024-06-25:
On my iPhone X iOS 16.7.8, the preview is correctly landscape when the Camera is started while the phone is sideways, but the picture it takes is rotated 90 degrees.

mars-lan, 2024-06-26:
> On my iPhone X iOS 16.7.8, the preview is correctly landscape when the Camera is started while the phone is sideways, but the picture it takes is rotated 90 degrees.

Couldn't get my hand on a device to verify yet, but several users of ours have reported the same issue on iOS 16 as well.

zzz08900, 2024-07-29:
Confirming the same issue with version 4.5.0 and Xiaomi Pad 6/Android 13.

If the app is launched while the tablet is in landscape position, the preview is rotated by 90 degrees and is stretched - it seems the width/height of the preview was swapped.

Will try the fix proposed by OP tomorrow.

I remember I saw this problem before with RNVC v2. We were stuck with v2.14 for a really long time because of this problem - well technically we are still stuck since the problem appeared again on v4.5.0.

Just on a side note, for v4 it is possible to build an ugly hack that rotates the preview by 270 degrees using styling + texture-view so everything appears to be working fine as long as the user does not rotate the tablet.

Disclaimer: I'm not responsible for any consequence of using the mentioned hack above.

angelos3lex, 2024-08-21:
On my Samsung SM T590, Android 10 this is also happening, sometimes the preview is wrong or the output. Even if i have `outputOrientation='preview'` sometimes preview and recorded video have different orientation.

boiboif, 2024-10-28:
> Confirming the same issue with version 4.5.0 and Xiaomi Pad 6/Android 13.确认4.5.0版本和小米Pad 6/Android 13存在同样的问题。
> 
> If the app is launched while the tablet is in landscape position, the preview is rotated by 90 degrees and is stretched - it seems the width/height of the preview was swapped.如果在平板电脑处于横向位置时启动应用程序，预览将旋转 90 度并拉伸 - 预览的宽度/高度似乎已交换。
> 
> Will try the fix proposed by OP tomorrow.明天将尝试 OP 提出的修复方案。
> 
> I remember I saw this problem before with RNVC v2. We were stuck with v2.14 for a really long time because of this problem - well technically we are still stuck since the problem appeared again on v4.5.0.我记得我之前在 RNVC v2 中看到过这个问题。由于这个问题，我们在 v2.14 上被困了很长一段时间——从技术上来说，我们仍然被困住，因为这个问题在 v4.5.0 上再次出现。
> 
> Just on a side note, for v4 it is possible to build an ugly hack that rotates the preview by 270 degrees using styling + texture-view so everything appears to be working fine as long as the user does not rotate the tablet.顺便说一句，对于 v4，可以构建一个丑陋的 hack，使用样式 + 纹理视图将预览旋转 270 度，因此只要用户不旋转平板电脑，一切似乎都工作正常。
> 
> Disclaimer: I'm not responsible for any consequence of using the mentioned hack above.免责声明：我对使用上述黑客行为造成的任何后果不承担任何责任。
@zzz08900 Any updates?

meleffendi, 2024-12-26:
Can confirm the same issue on Android 14 tablet Lenovo P12 vision camera version 4.5.2. The patch above did not work for me.