margelo/react-native-vision-camera #3027: 🐛 Unable to set minimum zoom on first render

RenaudAubert, 2024-06-25:
### What's happening?

Hi!
I'm trying to have the back camera to have the minimum zoom from the start.
I've tried using `device.minZoom` but I still have the neutral zoom even on subsequent renders the zoom is still neutral.
When I try to pinch to change zoom using the `enableZoomgesture` prop I can only zoom in as if the the camera was already at min zoom and the preview instantly jump to display the zoomed out view

I've managed to reproduce this behavior in the example app. I've added `zoom={device.minZoom}` to the camera component but when opening the app I have the neutral zoom

The logs from logcat includes the moment the camera is initialized up to me pinching to zoom in/out to reset the zoom
Please let me know if the log is too long I will update and only provide the camera initialization.

Cheers

### Reproduceable Code

```tsx
// Also tried adding `{ physicalDevices: ["ultra-wide-angle-camera"]}` filter
const device = useCameraDevice("back");

{device != null &&
<Camera
    device={device}
    isActive={isActive}
    ref={camera}
    style={StyleSheet.absoluteFill}
    photo
    video={false}
    audio={false}
    onError={onError}
    onInitialized={onInitialized}
    zoom={device.minZoom}
    enableZoomGesture
    enableLocation
 />}
```

### Relevant log output

```shell
Camera Lifecycle changed to CREATED!
Updating CameraSession...
Updating CameraSession...
Updating CameraSession...
configure { ... }: Waiting for lock...
A new configure { ... } call arrived, aborting this one...
CameraView attached to window!
Initializing without READ_DEVICE_CONFIG permission. enabled=false, interval=1, missedFrameThreshold=3, frameTimeThreshold=64, package=com.cropscan
configure { ... }: Waiting for lock...
A new configure { ... } call arrived, aborting this one...
configure { ... }: Waiting for lock...
configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true, orientationChanged=true, locationChanged=true)
Creating new Outputs for Camera #0...
Using FPS Range: null
Creating Preview output...
Creating Photo output...
Successfully created new Outputs for Camera #0!
Binding Camera #0...
Binding 2 use-cases...
getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
Resolved dynamic range for use case androidx.camera.core.Preview-7e1c6b3b-60a0-454c-ba60-04cc5618e348 to no compatible HDR dynamic ranges.
DynamicRange@8af89a7{encoding=UNSPECIFIED, bitDepth=0}
->
DynamicRange@8c44266{encoding=SDR, bitDepth=8}
getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
Surface created[total_surfaces=1, used_surfaces=0](androidx.camera.core.processing.SurfaceEdge$SettableSurface@d3c5bc0}
Surface created[total_surfaces=2, used_surfaces=0](androidx.camera.core.SurfaceRequest$2@f6b73ec}
New surface in use[total_surfaces=2, used_surfaces=1](androidx.camera.core.SurfaceRequest$2@f6b73ec}
use count+1, useCount=1 androidx.camera.core.SurfaceRequest$2@f6b73ec
[... 519 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 6 Pro

### VisionCamera Version

4.3.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-07-10:
Hey - I created a PR to potentially fix this: https://github.com/mrousavy/react-native-vision-camera/pull/3066

Can you test if this fixes the issue for you?

mrousavy, 2024-07-10:
Thanks for the logs btw. A lot of people simply don't get this, but I just read the logs and I can understand what went wrong.

RenaudAubert, 2024-07-10:
Hi @mrousavy,
Thanks a lot for taking the time to look into this issue.
I apologize for not getting back to you sooner but I just tested with version 4.4.2 and the problem is still occurring.

Here's the log from the camera initialization (the "test" log is a `console.log` I added ensure I have the minimum zoom set)
The code for the Camera is the same as in my original post
```
profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=72000000, frameRate=60, width=3840, height=2160, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=33000000, frameRate=60, width=1920, height=1080, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=12000000, frameRate=30, width=1280, height=720, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=6000000, frameRate=30, width=720, height=480, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
Advanced Extensions Implemented 
Advanced Extensions Implemented 
profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=48000000, frameRate=30, width=3840, height=2160, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=33000000, frameRate=60, width=1920, height=1080, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=12000000, frameRate=30, width=1280, height=720, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
profiles = ImmutableEncoderProfilesProxy{defaultDurationSeconds=60, recommendedFileFormat=2, audioProfiles=[AudioProfileProxy{codec=3, mediaType=audio/mp4a-latm, bitrate=96000, sampleRate=48000, channels=1, profile=1}], videoProfiles=[VideoProfileProxy{codec=2, mediaType=video/avc, bitrate=6000000, frameRate=30, width=720, height=480, profile=-1, bitDepth=8, chromaSubsampling=0, hdrFormat=0}]}
getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=270, isOppositeFacing=false, result=270
Advanced Extensions Implemented 
Advanced Extensions Implemented 
'test: ', 0.5304937958717346
Cannot get virtualdevice_native service
Camera Lifecycle changed to CREATED!
Updating CameraSession...
Updating CameraSession...
Updating CameraSession...
configure { ... }: Waiting for lock...
A new configure { ... } call arrived, aborting this one...
CameraView attached to window!
Initializing without READ_DEVICE_CONFIG permission. enabled=false, interval=1, missedFrameThreshold=3, frameTimeThreshold=64, package=com.cropscan
configure { ... }: Waiting for lock...
A new configure { ... } call arrived, aborting this one...
configure { ... }: Waiting for lock...
configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true, orientationChanged=true, locationChanged=true)
Creating new Outputs for Camera #0...
Using FPS Range: null
Creating Preview output...
Creating Photo output...
Photo size: 4624x3472
Successfully created new Outputs for Camera #0!
Binding Camera #0...
Binding 2 use-cases...
getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
[... cut ...]
Attempting to send capture request onConfigured
Issuing request for session.
createCaptureRequest
CameraCaptureSession.onConfigured() mState=OPENED
CameraCaptureSession.onReady() OPENED
Update Preview stream state to STREAMING
PreviewView Stream State changed to STREAMING
invokeOnPreviewStarted()
view not autofillable - not passing ime action check
view not autofillable - not passing ime action check
tagSocket(214) with statsTag=0xffffffff, statsUid=-1
invokeOnAverageFpsChanged(0.0)
tagSocket(218) with statsTag=0xffffffff, statsUid=-1
```

AdamWP, 2024-10-30:
I have this same issue with v4.5.3 on a Pixel 7A.  Here's a work-around that zooms out when recording starts. I used minZoom + .01 in a few places as minZoom didn't work (rounding down issue?)

`import React, { useState, useRef, useEffect, useCallback } from 'react';
import { View, Text, StyleSheet, SafeAreaView, TouchableOpacity, StatusBar, Alert, Button, useWindowDimensions } from 'react-native';
import { Camera, CameraPermissionStatus, useCameraDevice, getCameraFormat } from 'react-native-vision-camera';
import { Dropdown } from 'react-native-element-dropdown';
import RNFS from 'react-native-fs';
import Reanimated, { useAnimatedProps, useSharedValue } from 'react-native-reanimated';
import { Gesture, GestureDetector, GestureHandlerRootView } from 'react-native-gesture-handler';

Reanimated.addWhitelistedNativeProps({ zoom: true });
const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera);

export default function CameraScreen() {
  const [minZoom, setMinZoom] = useState(1);
  const device = useCameraDevice('back', {
    physicalDevices: ['ultra-wide-angle-camera'],
  });
  const cameraRef = useRef<Camera>(null);

  const zoom = useSharedValue(device?.neutralZoom || 1);
  const zoomOffset = useSharedValue(0);

  const pinchGesture = Gesture.Pinch()
    .onBegin(() => {
      console.log('Pinch gesture started x');
      zoomOffset.value = zoom.value;
    })
    .onUpdate((event) => {
      console.log('Pinch gesture updated');
      console.log('minZoom:', minZoom);
      const newZoom = zoomOffset.value * event.scale;
      zoom.value = minZoom + .01; // must add a small value otherwise it doesn't work. possible rounding down issue?
      console.log('zoom.value:', zoom.value);
    });

  const animatedProps = useAnimatedProps(() => ({
    zoom: zoom.value,
  }));

  const [cameraPermissionStatus, setCameraPermissionStatus] = useState<CameraPermissionStatus>('not-determined');
  const [microphonePermissionStatus, setMicrophonePermissionStatus] = useState<CameraPermissionStatus>('not-determined');
  const [isRecording, setIsRecording] = useState(false);
  const [resolution, setResolution] = useState<string>('1920x1080');
  const [fps, setFps] = useState<string>('30');

  const { width, height } = useWindowDimensions();
  const isLandscape = width > height;

  const resolutionOptions = [
    { label: '720p', value: '1280x720' },
    { label: '1080p', value: '1920x1080' },
    { label: '4K', value: '3840x2160' },
  ];

  const fpsOptions = [
    { label: '30 FPS', value: '30' },
    { label: '60 FPS', value: '60' },
  ];

  const requestCameraPermission = useCallback(async () => {
    const status = await Camera.requestCameraPermission();
    setCameraPermissionStatus(status);
  }, []);

  const requestMicrophonePermission = useCallback(async () => {
    const status = await Camera.requestMicrophonePermission();
    setMicrophonePermissionStatus(status);
  }, []);

[... 215 lines cut ...]
  },
  dropdown: {
    flex: 1,
    marginHorizontal: 10,
    borderWidth: 1,
    borderColor: '#fff',
  },
  landscapeSideControls: {
    position: 'absolute',
    top: 50,
    right: 20,
    justifyContent: 'space-between',
    height: '60%',
  },
  sideButton: {
    backgroundColor: 'transparent',
    padding: 10,
  },
  sideButtonText: {
    color: 'white',
    fontSize: 16,
    textAlign: 'center',
  },
});
`

AdamWP, 2024-10-30:
I have this same issue with v4.5.3 on a Pixel 7A.  Here's a work-around that zooms out when recording starts. I used minZoom + .01 in a few places as minZoom didn't work (rounding down issue?)

import React, { useState, useRef, useEffect, useCallback } from 'react';
import { View, Text, StyleSheet, SafeAreaView, TouchableOpacity, StatusBar, Alert, Button, useWindowDimensions } from 'react-native';
import { Camera, CameraPermissionStatus, useCameraDevice, getCameraFormat } from 'react-native-vision-camera';
import { Dropdown } from 'react-native-element-dropdown';
import RNFS from 'react-native-fs';
import Reanimated, { useAnimatedProps, useSharedValue } from 'react-native-reanimated';
import { Gesture, GestureDetector, GestureHandlerRootView } from 'react-native-gesture-handler';

Reanimated.addWhitelistedNativeProps({ zoom: true });
const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera);

export default function CameraScreen() {
  const [minZoom, setMinZoom] = useState(1);
  const device = useCameraDevice('back', {
    physicalDevices: ['ultra-wide-angle-camera'],
  });
  const cameraRef = useRef<Camera>(null);

  const zoom = useSharedValue(device?.neutralZoom || 1);
  const zoomOffset = useSharedValue(0);

  const pinchGesture = Gesture.Pinch()
    .onBegin(() => {
      console.log('Pinch gesture started x');
      zoomOffset.value = zoom.value;
    })
    .onUpdate((event) => {
      console.log('Pinch gesture updated');
      console.log('minZoom:', minZoom);
      const newZoom = zoomOffset.value * event.scale;
      zoom.value = minZoom + .01; // must add a small value otherwise it doesn't work. possible rounding down issue?
      console.log('zoom.value:', zoom.value);
    });

  const animatedProps = useAnimatedProps(() => ({
    zoom: zoom.value,
  }));

  const [cameraPermissionStatus, setCameraPermissionStatus] = useState<CameraPermissionStatus>('not-determined');
  const [microphonePermissionStatus, setMicrophonePermissionStatus] = useState<CameraPermissionStatus>('not-determined');
  const [isRecording, setIsRecording] = useState(false);
  const [resolution, setResolution] = useState<string>('1920x1080');
  const [fps, setFps] = useState<string>('30');

  const { width, height } = useWindowDimensions();
  const isLandscape = width > height;

  const resolutionOptions = [
    { label: '720p', value: '1280x720' },
    { label: '1080p', value: '1920x1080' },
    { label: '4K', value: '3840x2160' },
  ];

  const fpsOptions = [
    { label: '30 FPS', value: '30' },
    { label: '60 FPS', value: '60' },
  ];

  const requestCameraPermission = useCallback(async () => {
    const status = await Camera.requestCameraPermission();
    setCameraPermissionStatus(status);
  }, []);

  const requestMicrophonePermission = useCallback(async () => {
    const status = await Camera.requestMicrophonePermission();
    setMicrophonePermissionStatus(status);
  }, []);

[... 214 lines cut ...]
    paddingBottom: 10,
  },
  dropdown: {
    flex: 1,
    marginHorizontal: 10,
    borderWidth: 1,
    borderColor: '#fff',
  },
  landscapeSideControls: {
    position: 'absolute',
    top: 50,
    right: 20,
    justifyContent: 'space-between',
    height: '60%',
  },
  sideButton: {
    backgroundColor: 'transparent',
    padding: 10,
  },
  sideButtonText: {
    color: 'white',
    fontSize: 16,
    textAlign: 'center',
  },
});

AdamWP, 2024-10-30:
I have this same issue with v4.5.3 on a Pixel 7A.  Here's a work-around that zooms out when recording starts. I used minZoom + .01  in a few places as minZoom didn't work (rounding down issue?)
```
import React, { useState, useRef, useEffect, useCallback } from 'react';
import { View, Text, StyleSheet, SafeAreaView, TouchableOpacity, StatusBar, Alert, Button, useWindowDimensions } from 'react-native';
import { Camera, CameraPermissionStatus, useCameraDevice, getCameraFormat } from 'react-native-vision-camera';
import { Dropdown } from 'react-native-element-dropdown';
import RNFS from 'react-native-fs';
import Reanimated, { useAnimatedProps, useSharedValue } from 'react-native-reanimated';
import { GestureHandlerRootView } from 'react-native-gesture-handler';

Reanimated.addWhitelistedNativeProps({ zoom: true });
const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera);

export default function CameraScreen() {
  const [minZoom, setMinZoom] = useState(1);
  const device = useCameraDevice('back', {
    physicalDevices: ['ultra-wide-angle-camera'],
  });
  const cameraRef = useRef<Camera>(null);

  const zoom = useSharedValue(device?.neutralZoom || 1);
  const zoomOffset = useSharedValue(0);

  const animatedProps = useAnimatedProps(() => ({
    zoom: zoom.value,
  }));

  const [cameraPermissionStatus, setCameraPermissionStatus] = useState<CameraPermissionStatus>('not-determined');
  const [microphonePermissionStatus, setMicrophonePermissionStatus] = useState<CameraPermissionStatus>('not-determined');
  const [isRecording, setIsRecording] = useState(false);
  const [resolution, setResolution] = useState<string>('1920x1080');
  const [fps, setFps] = useState<string>('30');

  const { width, height } = useWindowDimensions();
  const isLandscape = width > height;

  const resolutionOptions = [
    { label: '720p', value: '1280x720' },
    { label: '1080p', value: '1920x1080' },
    { label: '4K', value: '3840x2160' },
  ];

  const fpsOptions = [
    { label: '30 FPS', value: '30' },
    { label: '60 FPS', value: '60' },
  ];

  const requestCameraPermission = useCallback(async () => {
    const status = await Camera.requestCameraPermission();
    setCameraPermissionStatus(status);
  }, []);

  const requestMicrophonePermission = useCallback(async () => {
    const status = await Camera.requestMicrophonePermission();
    setMicrophonePermissionStatus(status);
  }, []);

  useEffect(() => {
    (async () => {
      await requestCameraPermission();
      await requestMicrophonePermission();
      const availableDevices = await Camera.getAvailableCameraDevices();
      const backDevice = availableDevices.find(d => d.position === 'back');
      if (backDevice) {
        setMinZoom(backDevice.minZoom);
        zoom.value = backDevice.minZoom + 0.01;
        console.log('backDevice.minZoom:', backDevice.minZoom);
      }
    })();
  }, []);
[... 187 lines cut ...]
  dropdown: {
    flex: 1,
    marginHorizontal: 10,
    borderWidth: 1,
    borderColor: '#fff',
  },
  landscapeSideControls: {
    position: 'absolute',
    top: 50,
    right: 20,
    justifyContent: 'space-between',
    height: '60%',
  },
  sideButton: {
    backgroundColor: 'transparent',
    padding: 10,
  },
  sideButtonText: {
    color: 'white',
    fontSize: 16,
    textAlign: 'center',
  },
});

```

rodrigobertolotti, 2025-02-06:
Same issue here, zoom value regardless is minimum zoom or not is not working on first camera render, it starts always in neutral zoom value.

tomasperezdev, 2025-02-12:
Same issue here. In my scenario, I would activate/deactivate the QRScanner so it would not be reading multiple entries; once one is observed, I would hide the camera component and then show it again after it finishes processing so it can scan again.

Im experiencing the same issue, both at camera render or whenever it comes back from its "isScannig" state.

I have done the following workaround, which is working; I am still trying to get a cleaner implementation for the use case, but the zoom resets every time the component appears on the screen.

```
const resetZoom = (currentZoom: number) => {
    setZoom(currentZoom - 0.01);
    setTimeout(() => {
      setZoom(currentZoom);
    }, 100);
  };

  const reloadCamera = () => {
    setIsScanning(true);
    setTimeout(() => {
      setIsScanning(false);
      resetZoom(zoom);
    }, 500);
  };

  useEffect(() => {
    reloadCamera();
  }, [device]);
```

This approach currently serves as a workaround but is less than ideal.

I also tried using Zoom as a user instead of a useState, and it had the same issue.

ShivamKJJW, 2025-05-15:
Hi, I created a PR to potentially fix this. #3527 
It will be great if anyone can check this too and see if it works flawlessly.

closed 2026-04-21 by mrousavy