margelo/react-native-vision-camera #3028: react-native-vision-camera:buildCMakeRelWithDebInfo[arm64-v8a] FAILED

FazilMuhammed, 2024-06-26:
### Question

when i try make a Apk.
  
  cd android && ./gradlew app:assembleRelease && cd ..

 getting error 

"Execution failed for task ':react-native-vision-camera:buildCMakeRelWithDebInfo[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/node_modules/react-native-vision-camera/android/.cxx/RelWithDebInfo/6vrrw3b6/arm64-v8a'"

react native version 0.74.2
react native vision camera 4.3.2
react-native-worklets-core 1.3.3

how to fix this CMakeFile issue already try with troubleshoot no change happens

### What I tried

when try to generate APK

### VisionCamera Version

4.3.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

dankocher, 2024-06-26:
I have the same problem. Any solution?

dankocher, 2024-06-26:
Deleting folder `android/.gradle` worked for me

mrousavy, 2024-06-27:
wrong issue template.

closed 2024-06-27 by mrousavy

LuizFBrisighello, 2024-09-09:
Related to #1913 probably

vaibhavdhake123, 2025-02-13:
Same iusse Here