margelo/react-native-vision-camera #3029: Focus not working on some Android devices when setting FPS

rodrigobertolotti, 2024-06-26:
### What's happening?

If i set my fps in the camera to fps={15} and tap to focus on certain android devices, i get a crash or error where the camera eventually recovers but it leads to a 3 o 4 seconds of UI freezed. If i remove the fps prop, the error does not happen. 

Just to clarify, the current camera device indicates focusing is supported and although on my use case im setting fps to 15, i can reproduce it on the ShadowLens and on the example app for that same device. 

This happens only on some Androd devices (Pixel 2 and Pixel 5), this does not happen on my iPhone  15 Pro or other android devices (Xiaomi 13c with Android 14 works fine for example).

This same crash happens on the ShadowLens app while recording and tapping to focus. 

This is the error on the js side: 

 { [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]
name: 'session/recoverable-error',
_code: 'session/recoverable-error',
_message: 'An unknown error occurred while creating the Camera Session, but the Camera can recover from it.',
_cause: undefined }

### Reproduceable Code

```tsx
<Camera
              fps={15}
              photo={true}
              photoQualityBalance="quality"
              video={true}
              photoHdr={true}
              outputOrientation="device"
              enableZoomGesture={true}
              device={currentDevice}
              format={format}
              torch={torchOn ? 'on' : 'off'}
              isActive={cameraActive}
              audio={audioEnabled}
              photoQualityBalance={'quality'}
              onInitialized={onCameraReady}
              exposure={exposure}
              aspectRatio={aspectRatio}
              videoMode={videoMode}
              enableHighQualityPhotos={highQualityPhotos}
              style={cameraStyle}
              codeScanner={codeScanner && scanCodes}
              videoStabilizationMode={videoStabilization}
              onError={processCameraError}
            ></Camera>
```

### Relevant log output

```shell
E  Unable to access camera: CAMERA_ERROR (3): createDefaultRequest:1618: Camera 0: Error creating default request for template 3: Function not implemented (-38)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  java.lang.Exception: Stack trace
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at java.lang.Thread.dumpStack(Thread.java:1529)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.CaptureSession.issueBurstCaptureRequest(CaptureSession.java:796)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.CaptureSession.issuePendingCaptureRequest(CaptureSession.java:659)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.CaptureSession.issueCaptureRequests(CaptureSession.java:546)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.Camera2CameraImpl.submitCaptureRequests(Camera2CameraImpl.java:1645)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.Camera2CameraImpl$ControlUpdateListenerInternal.onCameraControlCaptureRequests(Camera2CameraImpl.java:2363)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.Camera2CameraControlImpl.submitCaptureRequestsInternal(Camera2CameraControlImpl.java:640)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.FocusMeteringControl.cancelAfAeTrigger(FocusMeteringControl.java:527)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.FocusMeteringControl.cancelFocusAndMeteringInternal(FocusMeteringControl.java:828)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.FocusMeteringControl.cancelFocusAndMeteringWithoutAsyncResult(FocusMeteringControl.java:815)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.FocusMeteringControl.lambda$executeMeteringAction$10$androidx-camera-camera2-internal-FocusMeteringControl(FocusMeteringControl.java:790)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.camera2.internal.FocusMeteringControl$$ExternalSyntheticLambda6.run(Unknown Source:4)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.workOnQueue(SequentialExecutor.java:231)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:173)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
2024-06-26 12:46:18.701  8783-8847  System.err              com.mrousavy.camera.example          W  	at java.lang.Thread.run(Thread.java:923)
2024-06-26 12:46:18.702  8783-8847  UseCaseAttachState      com.mrousavy.camera.example          D  Active and attached use case: [androidx.camera.video.VideoCapture-5d589c8b-abae-4cdb-bb29-e961f854f21e137680087, androidx.camera.core.Preview-1694c683-ccda-428d-bcc2-a294807f7b4e261039729, androidx.camera.core.ImageCapture-c30a4f7f-1d88-4457-af5a-a206df9de3b4168819798] for camera: 0
[... 44 lines cut ...]
### Camera Device

```json
{"hardwareLevel": "full", "hasFlash": true, "hasTorch": true, "id": "0", "isMultiCam": false, "maxExposure": 12, "maxZoom": 7, "minExposure": -12, "minFocusDistance": 10, "minZoom": 1, "name": "0 (BACK) androidx.camera.camera2", "neutralZoom": 1, "physicalDevices": ["wide-angle-camera"], "position": "back", "sensorOrientation": "landscape-left", "supportsFocus": true, "supportsLowLightBoost": false, "supportsRawCapture": false}
```

### Device

Google Pixel 2 Android 11

### VisionCamera Version

4.1.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-06-27:
So this is definitely a CameraX bug then.
Can you report it in the CameraX issue tracker?

mrousavy, 2024-06-27:
(thanks for the detailed testing & logs btw)

mrousavy, 2024-06-28:
Did you post a CameraX issue? If yes, please link it here for visibility.

rodrigobertolottizinsp, 2024-07-02:
Hi, I was quite busy I'll try to do it today. I'll paste the link here as soon it's done. Thanks!

rodrigobertolotti, 2024-07-02:
Hi, I was quite busy I'll try to do it today. I'll paste the link here as soon it's done. Thanks!

closed 2026-04-21 by mrousavy