margelo/react-native-vision-camera #3032: Expose Camera Intrinsic Parameters✨

pvedula7, 2024-06-28:
### What feature or enhancement are you suggesting?

Exposing camera intrinsic parameters to JS. This thread: https://github.com/mrousavy/react-native-vision-camera/issues/2998 referenced grabbing calibration where @mrousavy said was lots of lines of code, but exposing only intrinsic should be easier since we do not need the calibration data part of classes.

Android exposes it in `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` [here](https://developer.android.com/reference/android/hardware/camera2/CameraCharacteristics#LENS_INTRINSIC_CALIBRATION)
IOS has an `isCameraIntrinsicMatrixDeliveryEnabled` on `AVCaptureConnection` and available in the [CMBuffer](https://developer.apple.com/documentation/coremedia/cmsamplebuffer/attachmentkey/3242294-cameraintrinsicmatrix) `cameraIntrinsicMatrix`

Why?
Intrinsic parameters are necessary for any kind of 2D to 3D CV techniques such as point clouds. 

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

N/A at the moment

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-06-28:
Hey - good feature request.

As Hans said, I work on this in my free time and at the moment Camera intrinsic data delivery has a very low priority for me.

If you want this prioritized consider contacting me thru my consultancy agency, otherwise this stays open until I decide to tackle it in my free time in a few weeks or months or so.

I had a working prototype where I delivered the matrix thru the CMSampleBuffer, but it was a lot of (Objective-)C code that was required for this. On Android, it's also a bit different as it's not available per Frame, but on the device already and I don't want two different solutions.

shivams2i, 2024-12-24:
Any updates on this? Will this feature be available in upcoming releases?