margelo/react-native-vision-camera #3035: Is it possible to buffer frames while Vision Camera is running and append them to a video recording after the recording finishes?

bigg-S, 2024-07-01:
### Question

Hi everyone,

I'm using Vision Camera to record short videos based on the occurrence of an event. However, there is often a delay between when frame processing stops and when recording starts, causing some frames to be missed.

I was considering maintaining a buffer of frames prior to recording and then appending these frames to the recorded video before saving it. Is this feasible with Vision Camera? If so, could someone guide me on how to implement this?

Thanks in advance for your help!

### What I tried

I have been able to maintain a buffer of 30 frames prior to occurrence of the event.

### VisionCamera Version

4.0.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-07-02:
Hey - yes this is possible, but you would need to create your own recording session in your custom native Frame Processor Plugin. In this custom FP Plugin you can then simply append the `CMSampleBuffer`s to your video writer (`AVAssetWriter`), or just re-use the `RecordingSession` from VisionCamera.

On Android, this is infinitely more complex as you need to write your completely own and custom video recorder (`MediaEncoder`) which can write `Image`s (or `HardwareBuffer`s) to a video file.

In both cases, you'd also have to manually add an audio input somehow.

Essentially you'd just be re-building the whole video recording feature in your own frame processor plugin. It is possible, but very complex.

---

So instead my ideal solution would be to change the recording API.

This is how VisionCamera's recording API currently works:

```ts
console.log('starting...')
camera.current.startRecording({
  onRecordingFinished: (video) => console.log(video),
  videoCodec: 'h265'
})
// We cannot console.log('started!') here, because it is not an awaitable call.

setTimeout(async () => {
  console.log('stopping...')
  await camera.current.stopRecording()
  console.log('stopped!')
}, 5000)
```

As highlighted in my comment, the problem is that `startRecording` is **not awaitable** because Native Modules have a limitation where you cannot have both a callback (`onRecordingFinished`) and a Promise in a single method. So this method returns `void`, and is not awaitable.

Because of this limitation, we don't know when the recording actually started. It might be after a few milliseconds, it might be after a second, we don't know.

Also, another problem with this approach is that we do not have a warmed up recording instance in memory. The recording instance has to be re-created each time, which costs between 50ms to 250ms on iOS - a lot of time!

My idea is to build a new API -  Instead of just `startRecording(...): void`, I'd create a `prepareRecording(...): Promise<Recording>` method:

```ts
const recording = await camera.current.prepareRecording({
  onRecordingFinished: (video) => console.log(video),
  videoCodec: 'h265'
})
// `recording` is now created and warmed up. We can keep that in memory (e.g. useRef)

// later:
console.log('starting...')
await recording.start()
console.log('started!')
setTimeout(() => {
  console.log('stopping...')
  await recording.stop()
  console.log('stopped!')
}, 5000)
```

This API would be perfect, but is unfortunately impossible to build with Native- or Turbo- Modules, because we cannot have custom types (here: `Recording`, return value of `prepareRecording`) in ObjC/Java.

I am working on a prototype new React Native Framework which allows us to use such types, but it is far from finished - this is a huge project.

So long story short; Right now your only option is a custom Frame Processor Plugin. Or a patch which we can create for you through my consultancy agency.

In the future we might have custom types and warmed up recording sessions - that'd definitely be really really cool.

closed 2024-07-02 by mrousavy

bigg-S, 2024-07-02:
Thank you for the detailed and insightful response! I truly appreciate the thorough explanation of the complexities involved in creating a custom native Frame Processor Plugin for video recording. Your breakdown of the challenges on both iOS and Android, as well as the current limitations of VisionCamera's recording API, was incredibly helpful.

mrousavy, 2024-07-02:
Thanks! Just for reference, [this is the `RecordingSession.swift` from VisionCamera](https://github.com/mrousavy/react-native-vision-camera/blob/main/package/ios/Core/RecordingSession.swift)

bigg-S, 2024-07-02:
As a quick solution though. Is there a way I can save the frames as jpg images(in the frame processor) in a temporary folder and use ffmpeg to create a video and append to the recorded video before saving onRecordingFinished?

bigg-S, 2024-07-02:
Thanks so much. I'll try that