margelo/react-native-vision-camera #3038: unresolved supertypes: FrameProcessorPlugin

sabirbugti9, 2024-07-02:
### How were you trying to build the app?

I tried to run the application on android its give me this error

### Full build logs

```tsx
Supertypes of the following classes cannot be resolved. Please make sure you have the required dependencies in the classpath:
    class com.simple.PaletteFrameProcessorPlugin, unresolved supertypes: FrameProcessorPlugin
Adding -Xextended-compiler-checks argument might provide additional information.
```

### Project dependencies

```json
"@datadog/mobile-react-navigation": "^2.3.6",
    "@react-native-community/blur": "^4.4.0",
    "@react-native-community/cli": "^13.6.9",
    "@react-navigation/stack": "^6.4.0",
    "react": "18.2.0",
    "react-native": "0.74.3",
    "react-native-animateable-text": "^0.12.1",
    "react-native-gesture-handler": "^2.17.1",
    "react-native-haptic-feedback": "^2.2.0",
    "react-native-reanimated": "^3.12.1",
    "react-native-safe-area-context": "^4.10.7",
    "react-native-screens": "^3.32.0",
    "react-native-static-safe-area-insets": "^2.2.0",
    "react-native-vision-camera": "^4.3.2",
    "react-native-worklets-core": "^1.3.3"
```

### VisionCamera Version

4.3.2

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

sabirbugti9, 2024-07-02:
@mrousavy i am trying to update your color waver app i am getting this issue

mrousavy, 2024-07-02:
Colorwaver? I don't have `"@datadog/mobile-react-navigation": "^2.3.6",` in colorwaver

mrousavy, 2024-07-02:
> unresolved supertypes: FrameProcessorPlugin

This sounds like you just didn't import the `FrameProcessorPlugin` type from VisionCamera.

closed 2024-07-02 by mrousavy

sabirbugti9, 2024-07-02:
package com.simple;

import android.annotation.SuppressLint;
import android.content.Context;
import android.graphics.Bitmap;
import android.graphics.Color;
import android.media.Image;
import androidx.annotation.NonNull;
import androidx.annotation.Nullable;
import androidx.camera.core.ImageProxy;
import androidx.palette.graphics.Palette;
import com.simple.utils.YuvToRgbConverter;
import com.facebook.react.bridge.WritableNativeMap;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

public class PaletteFrameProcessorPlugin extends FrameProcessorPlugin {
    private final YuvToRgbConverter yuvToRgbConverter;
    @SuppressWarnings("FieldCanBeLocal")
    private static final int DEFAULT_COLOR = Color.BLACK;

    private int getBitmapQualityArea(Bitmap bitmap, String qualityString) {
        int higher = Math.max(bitmap.getWidth(), bitmap.getHeight());
        int lower = Math.min(bitmap.getWidth(), bitmap.getHeight());

        switch (qualityString) {
            case "lowest": {
                int maxWidth = 50;
                return (lower / higher * maxWidth) * maxWidth;
            }
            case "low": {
                int maxWidth = 100;
                return (lower / higher * maxWidth) * maxWidth;
            }
            case "high": {
                int maxWidth = 250;
                return (lower / higher * maxWidth) * maxWidth;
            }
            case "highest":
            default: {
                // full area
                return bitmap.getHeight() * bitmap.getWidth();
            }
        }
    }

    @SuppressLint("UnsafeOptInUsageError")
    @Nullable
    @Override
    public Object callback(@NonNull ImageProxy imageProxy, @NonNull Object[] params) {
        Image image = imageProxy.getImage();
        if (image == null) {
            // Image Proxy is empty!
            return null;
        }

        Bitmap bitmap = Bitmap.createBitmap(image.getWidth(), image.getHeight(), Bitmap.Config.ARGB_8888);
        yuvToRgbConverter.yuvToRgb(image, bitmap);

        Palette.Builder builder = new Palette.Builder(bitmap);

        if (params.length > 0 && params[0] instanceof String) {
            int area = getBitmapQualityArea(bitmap, (String) params[0]);
            builder.resizeBitmapArea(area);
        }

        Palette palette = builder.generate();

        int average = palette.getLightVibrantColor(DEFAULT_COLOR);
        int vibrant = palette.getVibrantColor(DEFAULT_COLOR);
        int darkVibrant = palette.getDarkVibrantColor(DEFAULT_COLOR);
        int dominant = palette.getDominantColor(DEFAULT_COLOR);

        WritableNativeMap result = new WritableNativeMap();
        result.putString("primary", rgbIntToHexString(vibrant));
        result.putString("secondary", rgbIntToHexString(darkVibrant));
        result.putString("detail", rgbIntToHexString(average));
        result.putString("background", rgbIntToHexString(dominant));
        return result;
    }

    private String rgbIntToHexString(int rgb) {
        return String.format("#%06X", (0xFFFFFF & rgb));
    }

    PaletteFrameProcessorPlugin(Context context) {
        super("getColorPalette");
        yuvToRgbConverter = new YuvToRgbConverter(context);
    }
}

sabirbugti9, 2024-07-02:
i am getting issue from here @mrousavy