margelo/react-native-vision-camera #3040: 'UIWindowScene' is only available in iOS 13.0 or newer 🐛

Romick2005, 2024-07-02:
### What's happening?

[image]

Please add version checks.

### Reproduceable Code

```tsx
No code needed.
```

### Relevant log output

```shell
It does not build on iOS for relevant output, but works on Android.
```

### Camera Device

```json
{
  device: "Any iPhone device with iOS > 13"
}
```

### Device

iPhone XS

### VisionCamera Version

4.4.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-07-02 by maintenance-hans[bot]

rodrigobertolottizinsp, 2024-07-03:
Project is set to run for iOS 13.4 and higher, maybe you are modifying your pods target iOS deployment version in the Podfile?

Romick2005, 2024-07-04:
> Project is set to run for iOS 13.4 and higher, maybe you are modifying your pods target iOS deployment version in the Podfile? 

I already have that. Latest react native do that for me in:
/node_modules/react-native/scripts/cocoapods/helpers.rb
min_ios_version_supported 13.4

geekabhinavp, 2024-07-04:
+1 facing same issue!

geekabhinavp, 2024-07-04:
I fixed it!
Step 1: Check the deployment versions using this command:
```
grep -r 'IPHONEOS_DEPLOYMENT_TARGET' ios/
```
Step 2: If any version lower than 13 is found, run the following:
```
find ios/Pods/Pods.xcodeproj -name project.pbxproj -print0 | xargs -0 sed -i '' 's/IPHONEOS_DEPLOYMENT_TARGET = [0-9]*\.[0-9]*/IPHONEOS_DEPLOYMENT_TARGET = 13.0/g'
```

TeraBlitz, 2024-07-12:
Is there any reason this is closed? this is still an issue.

SarjuHansaliya, 2024-07-16:
Fixed it with this code in Podfile and then run `pod install`

```ruby
platform :ios, '14.2'

target 'YourAppName' do
  #...some code...

  post_install do |installer|
    installer.pods_project.targets.each do |target|
      target.build_configurations.each do |config|
        config.build_settings['IPHONEOS_DEPLOYMENT_TARGET'] = '14.2' # replace here with your minimum deployment 
      end
    end
  end
end
```

You might be facing this issue because one of the pod in your project is using <13.0 target. You can check that by searching `IPHONEOS_DEPLOYMENT_TARGET` in `ios` directory.

w4t3r-45, 2024-09-02:
same issue even with testing to set the minimum deployment version , is there any other solutions guys ?

iliapnmrv, 2024-09-19:
hey! I've just reinstalled pods using `npx pod-install` and app build worked for me

parnekov, 2024-10-29:
it happened after install react-native-permissions. 

Fix it!

**Close Xcode -> Open -> Command +K (Clean) -> Run -> Works!**

jay10895, 2024-11-05:
[image]

I fixed it by changing Deployment iOS Deployment Target in Pods > react-native-vision-camera > Build Settings iOS Deployment Target

praveenflymeout, 2025-02-09:
[image] I followed above all suggestion like update iOS deployment target inside pod file need to update(14.2), but still i am facing same issue in my project. Could you please anybody help me in this.

"react-native": "0.72.3",
"react": "18.2.0",
"react-native-vision-camera": "4.6.3",