margelo/react-native-vision-camera #3041: App unable to go to sleep once camera loads🐛

FoundersApproach, 2024-07-02:
### What's happening?

Once camera library loads in one screen and going back to previous screen in stack navigation then app unable to goes to sleep. May be camera reference object is unable to let app sleep ?

I did some object reference clearance in component unmount but no luck. Please help! 

### Reproduceable Code

```tsx
import React, { useState, useEffect, useCallback } from 'react';
import { View, Text, TouchableOpacity, Alert, Platform, StyleSheet, PermissionsAndroid, Image, AppState } from 'react-native';
import {Camera, useCameraDevices, useCameraDevice,useCameraFormat, Templates, useCodeScanner } from 'react-native-vision-camera';
import TextRecognition from '@react-native-ml-kit/text-recognition';
import AppConstants from '../module/constantVairable'
import { getCameraFace, getCurrentOrientation, getDeviceHeight, getDeviceWidth, getSafeAreaInsetBottom, getSafeAreaInsetTop, getTorchOn, isOrientationPortrait, setCameraRef, setScanLog, verifyGTINChecksum } from '../api/api';
import { Gesture, GestureDetector } from 'react-native-gesture-handler'
import ImageResizer from '@bam.tech/react-native-image-resizer';
import { RFValue } from 'react-native-responsive-fontsize';
import RNFS from 'react-native-fs';
import Animated, { runOnJS } from 'react-native-reanimated';

let cameraRef = null
let capturePhotoInteval = false
let focusPointTimer = false

const ScannerFunction = (props) => {
    const device = useCameraDevice(getCameraFace());
    const [permissionGranted, setPermissionGranted] = useState(false);
    const [isInitialized, setIsInitialized] = useState(false);
    const [torch, setTorch] = useState(false);
    const [restartCamera, setRestartCamera] = useState(false);
    const [cameraActive, setCameraActive] = useState(true);

    const [tapLocation, setTapLocation] = useState(null);

    useEffect(() => {
        // let appStateListener = AppState.addEventListener('change', appStateChangeListener);
        if (Platform.OS === 'android') {
          askCameraPermission();
        }
        else {
          setPermissionGranted(true)
          if (device && device.hasTorch && getTorchOn()) {
            setTorch(true)
          }
        }
        startCaptureTimer(2)
        
        return () => {
          cameraRef = null
          console.log("ScannerFunction unmount>>>>")
          setCameraActive(false)
          setTorch(false)
          setPermissionGranted(false)
          stopCaptureTimer()
          setCameraRef(null)
        }
    }, []);

    const startCaptureTimer = (sec=1) => {
      // console.log("startCaptureTimer>>>>")
      if (capturePhotoInteval) {
        clearInterval(capturePhotoInteval);
      }

      capturePhotoInteval = setInterval(() => {
        // console.log("capturePhotoInteval")
        capturePhoto()
      }, 2 * 1000);
    }
[... 230 lines cut ...]
              photo={true}
              ref={(ref) => cameraRef = ref}
              
            />
```

### Device

iPhone 14 plus (ios 17.1.1)

### VisionCamera Version

"react-native-vision-camera": "^3.9.2"

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

kevinranks, 2024-08-01:
@FoundersApproach We're seeing a similar issue on iOS 16+. On iOS <= 15 the green dot showing the camera is active never goes away. Can you try capturing the "goBack" and then using setTimeout to set the camera in-active prior to going back? 

Something like this:

```
const delayedGoBack = useCallback(() => {
     setCameraActive(false)
     setTimeout(() => navigation.goBack(), 10)
}, [])
```

Note: Would need to disable back gestures

kevinranks, 2024-08-01:
https://github.com/mrousavy/react-native-vision-camera/blob/77e98178f84824a0b1c76785469413b64dc96046/package/ios/React/CameraView.swift#L281

Looks like this specific line keeps the phone from going to sleep. It doesn't appear that this is implemented on Android for VC.

jkaufman, 2024-08-22:
Perhaps this should just be removed, with the idle management left to library consumers?

jkaufman, 2024-08-22:
Perhaps this should just be removed? Idle management could be left to library consumers. This would benefit Expo users, especially, which uses [`keep-awake`](https://docs.expo.dev/versions/latest/sdk/keep-awake/) to allow stacked calls to hold and release wake lock / idle timer setting.

msageryd, 2024-12-11:
I'm finally migrating from react-native-camera to get rid of some old problems.. only to run into new ones =(

Leaving the green camera dot on and prevent the app from sleeping is a no-go for me. The delayedGoBack approach solves this, but only when I'm in charge of the back navigation. Do I really have to disable both the back button and the "swipe to go back" function to get this to work?

I have postponed the migration for many years and I was quite confident that vision-camera would be a really mature product by now. Everything else in vision-camera seems excellent, but this bug is a bummer. I'm hoping that I have misunderstood something or that I have misconfigured something. Is it really the case that this bug affects everyone who uses react-navigation?

msageryd, 2024-12-11:
For anyone else ending up here:
I tapped into the "beforeRemove" event in order to handle this delay even if the back navigation is not explicitly performed by me via goBack.

With the below code I can use the regular `navigation.goBack()` and the user can go back via the back-button. This should also work for the Android physical back button, but I haven't tried.

```
import {
  useNavigation,
  useFocusEffect,
} from '@react-navigation/native';
```

```
  useFocusEffect(() => {
      const unsubscribe = navigation.addListener('beforeRemove', async e => {
        e.preventDefault();
        setIsCameraActive(false);
        await new Promise(resolve => setTimeout(resolve, 10));
        navigation.dispatch(e.data.action);
      });
      return unsubscribe;
    }, [navigation]);
```

ChristopherGabba, 2025-02-09:
+1 on this. I commented out the line:

```swift
UIApplication.shared.isIdleTimerDisabled = isActive
```

milon27, 2025-09-02:
is it fixed? i still face the issue

closed 2026-04-21 by mrousavy