margelo/react-native-vision-camera #3043: 💭 Orientation and the equivalent degree

xulihang, 2024-07-03:
### Question

Previous vision camera versions (<4.0.3), I use 90 for LANDSCAPE_LEFT and 270 for LANDSCAPE_RIGHT.

For the current version, the degree for LANDSCAPE_LEFT becomes 270. So is the current one correct and the previous versions wrong?

```java
public static int getRotationDegreeFromOrientation(Orientation orientation) {
    if (orientation.name().equals(Orientation.PORTRAIT.name())) {
        return 0;
    }else if (orientation.name().equals(Orientation.LANDSCAPE_LEFT.name())) {
        return 270;
    } else if (orientation.name().equals(Orientation.LANDSCAPE_RIGHT.name())) {
        return 90;
    }else if (orientation.name().equals(Orientation.PORTRAIT_UPSIDE_DOWN.name())) {
        return 180;
    }
    return 0;
}
```

### VisionCamera Version

4.4.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-07-03:
Are you talking about Frame Processors? As in, `Frame.orientation`?

xulihang, 2024-07-03:
Yes. The frame.orientation.

mrousavy, 2024-07-03:
`LANDSCAPE_LEFT` means that the Frame is in a landscape orientation where the home button is on the left side. So like this:
[image]

(that would be it's natural orientation, so you don't need to rotate anything to make it display upright).

Now if you hold your phone upright in portrait mode (aka home button at the bottom), then the image would be displayed wrong as it is still in it's landscape orientation.

In this case, I _think_ you would need to rotate by -90 (or +270) degeres to make it appear upright. So landscape left should be 270, yes.

This behaviour should be identical on both iOS and Android.

xulihang, 2024-07-03:
Thanks for the clarification.

closed 2024-07-03 by xulihang

mrousavy, 2024-07-03:
Please let me know if everything works out, and if iOS and Android are now consistent. Thanks 😄