margelo/react-native-vision-camera #3044: 🐛 Android Compile Error: `Unresolved reference: jsCallInvokerHolder`

coder-xiaomo, 2024-07-03:
### What's happening?

When build with react native 0.75.0-rc.3, the following error has occurred:

```
e: file:///home/runner/work/xxxxx/xxxxx/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:33:56 Unresolved reference: jsCallInvokerHolder
```

It works fine with react native 0.74.3, but when i try to upgrade to react native 0.75.0-rc.3, this problem occurred.

my environment:
    "react": "19.0.0-rc-fb9a90fa48-20240614",
    "react-native": "0.75.0-rc.3",
    "react-native-vision-camera": "4.4.0",

### Reproduceable Code

```tsx
this is a complie problem, just yarn add react-native-vision-camera 4.4.0 with react-native 0.75.0-rc.3, the problem will occurred
```

### Relevant log output

```shell
> Task :jcore-react-native:extractProguardFiles
e: file:///home/runner/work/xxxxx/xxxxx/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:33:56 Unresolved reference: jsCallInvokerHolder

> Task :react-native-vision-camera:compileReleaseKotlin FAILED
```

### Camera Device

```json
this is a complie problem, so there is nothing i can input...
```

### Device

Android

### VisionCamera Version

4.4.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

coder-xiaomo, 2024-07-03:
Where the error was reported:

node_modules\react-native-vision-camera\android\src\main\java\com\mrousavy\camera\frameprocessors\VisionCameraProxy.kt:33:56

```
  init {
    //                                                👇👇 this is where the error is reported
    val jsCallInvokerHolder = context.catalystInstance.jsCallInvokerHolder as CallInvokerHolderImpl
    val jsRuntimeHolder =
      context.javaScriptContextHolder?.get() ?: throw Error("JSI Runtime is null! VisionCamera does not yet support bridgeless mode..")
    mScheduler = VisionCameraScheduler()
    mContext = WeakReference(context)
    mHybridData = initHybrid(jsRuntimeHolder, jsCallInvokerHolder, mScheduler)
  }
```

[image]

mrousavy, 2024-07-03:
See https://github.com/mrousavy/react-native-vision-camera/issues/2614 - VisionCamera does not fully support the new architecture yet due to a lot of different reasons (mainly it's heavy use of JSI in the Frame Processor runtime).

Closing this issue as #2614 will be used for tracking this.

closed 2024-07-03 by mrousavy