margelo/react-native-vision-camera #3045: Torch default ON but not working.

HadiKhalifeh, 2024-07-03:
### What's happening?

The torch cannot be turned ON by default when opening the camera.

### Reproduceable Code

```tsx
useEffect(() => {
    const timeoutId = setTimeout(() => {
      if (isActive) {
        setTorch(true);
      } else {
        setTorch(false);
      }
    }, 300);

    return () => clearTimeout(timeoutId);
  }, [isActive]);

 <Camera
            style={StyleSheet.absoluteFill}
            device={device}
            codeScanner={codeScanner}
            torch={torch ? 'on' : 'off'}
            isActive={isActive && isCameraInitialized}
            onInitialized={onInitialized}
          />
```

### Relevant log output

```shell
no error
```

### Camera Device

```json
-
```

### Device

Android

### VisionCamera Version

v4.3.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-07-03 by maintenance-hans[bot]

mrousavy, 2024-07-03:
Can't help with no log output.

HadiKhalifeh, 2024-07-03:
> Can't help with no log output.

There are no logs. The torch is enabled by default but not working. Additionally, cameraRef.current?.props.torch returns undefined instead of "on" or "off".

mrousavy, 2024-07-03:
> There are no logs.

There are logs, and I need those logs. Without logs I don't know what went wrong.

When creating an issue I added a detailed explanation on how you can get logs. You should read that and paste the logs here.

jslok, 2025-04-24:
I have the same issue on Android. If torch="on" at first render, the torch will not be on. I have to start the camera with torch off then toggle it on. Hopefully this is the relavent log area. There's some lines about the thermal status but my phone is not overheating.

```
04-23 17:32:19.230  1194  1760 I Lyric   : page_release_controller.cc:109: Disabling page releasing.
04-23 17:32:19.232  1381  1447 D Camera3-Device: Set real time priority for request queue thread (tid 32226)
04-23 17:32:19.232  1381  1447 D Camera3-Utils: Set regular priority for thread (tid 1447)
04-23 17:32:19.235 31833 32041 D SyncCaptureSessionImpl: [androidx.camera.camera2.internal.SynchronizedCaptureSessionImpl@67964c8] Session onConfigured()
04-23 17:32:19.236 31833 32041 D CaptureSession: Attempting to send capture request onConfigured
04-23 17:32:19.236 31833 32041 D CaptureSession: Issuing request for session.
04-23 17:32:19.236 31833 32041 D Camera2CaptureRequestBuilder: createCaptureRequest
04-23 17:32:19.238  1381 32225 D CameraService: startCameraStreamingOps: startOp for uid 10391
04-23 17:32:19.239  1194  1754 I GCH_CameraDeviceSession: ProcessCaptureRequest: [sbc] Stream 0 1st req arrived, notify SBC Manager.
04-23 17:32:19.239  1194  1754 I GCH_CameraDeviceSession: ProcessCaptureRequest: [sbc] Stream 1 1st req arrived, notify SBC Manager.
04-23 17:32:19.239  2076  2076 I pixel-thermal: VIRTUAL-SKIN:36.6098 raw data: VIRTUAL-SKIN:36609.8
04-23 17:32:19.239  1194  1754 I Lyric   : camera_powerhal.cc:174: SetPowerBoost Trigger CAMERA_LAUNCH_EXTENDED for 2000ms
04-23 17:32:19.239  1194  1754 I Lyric   : multicam_node.cc:1178: ********************dpm: put camera: 2 on NORMAL, frame number: 0, zoom ratio: 1.00 ********************
04-23 17:32:19.239  1194  1754 I Lyric   : multicam_node.cc:1355: ******************** new lead: camera: 2(Wide), frame number: 0, zoom ratio: 1.00 *******************
04-23 17:32:19.239  1194  1754 I Lyric   : multicam_node.cc:2102: ====== Multicam status update ======
04-23 17:32:19.239  1194  1754 I Lyric   : multicam_node.cc:2103: Frame number: 0, Lead cam: 2, Zoom ratio: 1.00, APP requested streams:  (0 -> 32769, 2) (1 -> 32771, 2)
04-23 17:32:19.239  1194  1754 I Lyric   : multicam_node.cc:2110: cam: 2, dpm_mode: Normal, activation_flags: None
04-23 17:32:19.239  1194  1754 I Lyric   : retiming_node.cc:1339: ******** dpm: Rear(hal id: 2) change DPM mode from Undefined to Normal, is virtual transition: 0, framework request id 0, frame number: 0
04-23 17:32:19.239  1194 26180 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: global frame rate at 60fps for camera 2 at frame 0
04-23 17:32:19.240  1194 26180 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: 3A at 30fps for camera 2 at frame 0
04-23 17:32:19.240  1194 26180 I Lyric   : three_a_request_node.cc:634: cam2_three_a_request: updating 3A max FPS to 30
04-23 17:32:19.240  1194 21601 I Lyric   : multicam_node.cc:2102: ====== Multicam status update ======
04-23 17:32:19.240  1194 21601 I Lyric   : multicam_node.cc:2103: Frame number: 1, Lead cam: 2, Zoom ratio: 1.00, APP requested streams:  (0 -> 32769, 2) (1 -> 32771, 2)
04-23 17:32:19.240  1194 21601 I Lyric   : multicam_node.cc:2110: cam: 2, dpm_mode: Undefined, activation_flags: None
04-23 17:32:19.240  1194 26195 I Lyric   : conditional_dpm_node.cc:65: cam2_conditional_dpm turn to dpm mode: Normal, framte number: 0
04-23 17:32:19.240  1194 26191 I Lyric   : focus_lens_actuator_driver_impl_base.cc:226: act-rear-normal save Active to pending mode
04-23 17:32:19.241 31833 32041 D CaptureSession: CameraCaptureSession.onConfigured() mState=OPENED
04-23 17:32:19.241  1194 26173 W Lyric   : generic_power_throttle_controller.cc:764: 240 calls in 370 sec(s):  Unsupported thermal policy: LDAF
04-23 17:32:19.241  1194 26173 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: LDAF NoOp for camera 2 at frame 0
04-23 17:32:19.241  1194 26173 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: RoiTracker NoOp for camera 2 at frame 0
04-23 17:32:19.241  1194 26173 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: PhaseDetection NoOp for camera 2 at frame 0
04-23 17:32:19.242  1194 26190 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: AE Flicker Detection NoOp for camera 2 at frame 0
04-23 17:32:19.242  1194 26190 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: AE True Light NoOp for camera 2 at frame 0
04-23 17:32:19.242  1194 26190 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: AE Topshot Motion Calculation NoOp for camera 2 at frame 0
04-23 17:32:19.242  1194 26190 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: AE Debug Data NoOp for camera 2 at frame 0
04-23 17:32:19.242  1194 26190 I Lyric   : generic_power_throttle_controller.cc:768: camera thermal status: AE Flash LED NoOp for camera 2 at frame 0
04-23 17:32:19.242  1194 26173 W Lyric   : external_auto_focus_controller_proxy.cc:539: 2 calls in 496 sec(s): ExternalAutoFocusControllerProxy: camera_id = 2; The total number of frames missing CDAF, HW PDAF, SW PDAF and Sparse PDAF statistics is 1
04-23 17:32:19.242  1194 26173 W Lyric   : external_auto_focus_controller_proxy.cc:884: 4 calls in 503 sec(s): Not able to get shutter notification timestamp
[... cut ...] tRanges={physical=[60.00 Hz, 90.00 Hz], render=[60.00 Hz, 60.00 Hz]} idleScreenConfig=nullptr}
04-23 17:32:19.245   566   566 I RefreshRateSelector: 33 mode changes were performed under the previous policy
04-23 17:32:19.245   566   566 D DisplayModeController: setDesiredMode 4619827677550801152 {mode={fps=60.00 Hz, modePtr={id=0, vsyncRate=60.00 Hz, peakRefreshRate=60.00 Hz}}, emitEvent=true, force=false}
04-23 17:32:19.245 31833 32041 D CaptureSession: CameraCaptureSession.onReady() OPENED
```