margelo/react-native-vision-camera #3050: 🔧     SwiftVerifyEmittedModuleInterface normal arm64 (in target 'VisionCamera' from project 'Pods')

sabirbugti9, 2024-07-04:
### How were you trying to build the app?

i tried to use vision camera in ios react native i am getting this error 

### Full build logs

```tsx
The following build commands failed:

        SwiftVerifyEmittedModuleInterface normal arm64 Verifying\ emitted\ module\ interface\ VisionCamera.swiftinterface /Users/user/Library/Developer/Xcode/DerivedData/sample-bhtfjuhyykmkwtclkfcjkldqxstp/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/VisionCamera.build/Objects-normal/arm64/VisionCamera.swiftinterface (in target 'VisionCamera' from project 'Pods')

        SwiftVerifyEmittedModuleInterface normal arm64 Verifying\ emitted\ module\ interface\ VisionCamera.private.swiftinterface /Users/user/Library/Developer/Xcode/DerivedData/sample-bhtfjuhyykmkwtclkfcjkldqxstp/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/VisionCamera.build/Objects-normal/arm64/VisionCamera.private.swiftinterface (in target 'VisionCamera' from project 'Pods')
```

### Project dependencies

```json
"@datadog/mobile-react-navigation": "^2.3.6",
    "@react-native-community/blur": "^4.4.0",
    "@react-native-community/cli": "^13.6.9",
    "@react-navigation/native": "^6.1.17",
    "@react-navigation/native-stack": "^6.10.0",
    "@react-navigation/stack": "^6.4.0",
    "react": "18.2.0",
    "react-native": "0.74.3",
    "react-native-animateable-text": "^0.12.1",
    "react-native-gesture-handler": "^2.17.1",
    "react-native-haptic-feedback": "^2.2.0",
    "react-native-reanimated": "^3.12.1",
    "react-native-safe-area-context": "^4.10.7",
    "react-native-screens": "^3.32.0",
    "react-native-sound": "^0.11.2",
    "react-native-static-safe-area-insets": "^2.2.0",
    "react-native-vector-icons": "^10.1.0",
    "react-native-vision-camera": "^4.3.2",
    "react-native-worklets-core": "^1.3.3",
    "react-navigation": "^5.0.0"
```

### VisionCamera Version

4.3.2

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

sabirbugti9, 2024-07-04:
Please help me with this @mrousavy

mrousavy, 2024-07-10:
This does not look like a VisionCamera error at all, either your Swift version is outdated, your Xcode is outdated, your MacOS is outdated, or you don't have Swift set up in your project. 

But either way - please stop tagging me in issues, I am not your free support hotline.

closed 2024-07-10 by mrousavy