margelo/react-native-vision-camera #3051: 🐛  Frame orientation is in landscape-right when phone is held in portrait mode

Marius-Adam, 2024-07-05:
### What's happening?

I want to scan text in portrait mode but the camera does not pick up anything only if the text is sideways or the phone is in landscape mode. 

When logging out the frame.orientation it comes back as landscape-right when the phone is held in portrait

### Reproduceable Code

```tsx
import React from "react";
import { StyleSheet } from "react-native";
import { Camera, useCameraDevice, useFrameProcessor } from "react-native-vision-camera";
import { useTextRecognition } from "react-native-vision-camera-text-recognition";

function App() {
  const device = useCameraDevice("back");
  const options = { language: "latin"};
  const { scanText } = useTextRecognition(options);
  const frameProcessor = useFrameProcessor((frame) => {
    "worklet";
    console.log(frame.orientation)
    const data = scanText(frame);
   
  }, []);
  return (
    <>
      {!!device && (
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive
          frameProcessor={frameProcessor}
        />
      )}
    </>
  );
}
export default App;
```

### Relevant log output

```shell
Running "main" with {"rootTag":1,"initialProps":{"concurrentRoot":false}}
Loading react-native-worklets-core...
Worklets loaded successfully
VisionCameraProxy: Creating Worklet Context...
VisionCameraProxy: Worklet Context Created!
🟢 Creating JS object for module 'ExpoKeepAwake'
Looking up Frame Processor Plugin "scanText"...
Frame Processor Plugin "scanText" found! Initializing...
Initialized TensorFlow Lite runtime.
INFO: Initialized TensorFlow Lite runtime.
13:27:21.231: [info] 📸 VisionCamera.didSetProps(_:): Updating 22 props: [onInitialized, cameraId, position, enableBufferCompression, onOutputOrientationChanged, onStarted, preview, top, onCodeScanned, right, isActive, isMirrored, onViewReady, onError, onStopped, onPreviewOrientationChanged, onPreviewStopped, enableFrameProcessor, onPreviewStarted, left, bottom, onShutter]
13:27:21.236: [info] 📸 VisionCamera.configurePreviewOrientation(_:): Updating Preview rotation: landscapeLeft...
13:27:21.237: [info] 📸 VisionCamera.configureOutputOrientation(_:): Updating Outputs rotation: landscapeLeft...
13:27:21.237: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...
13:27:21.246: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: true, locationChanged: true)
13:27:21.246: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Input Device...
13:27:21.246: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
13:27:21.261: [debug] 📸 VisionCamera.sensorOrientation: Sensor Orientation changed from landscapeLeft -> portrait
13:27:21.261: [info] 📸 VisionCamera.configurePreviewOrientation(_:): Updating Preview rotation: portrait...
13:27:21.261: [info] 📸 VisionCamera.configureOutputOrientation(_:): Updating Outputs rotation: portrait...
13:27:21.261: [info] 📸 VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
13:27:21.261: [info] 📸 VisionCamera.configureOutputs(configuration:): Configuring Outputs...
13:27:21.261: [info] 📸 VisionCamera.configureOutputs(configuration:): Adding Video Data output...
13:27:21.263: [info] 📸 VisionCamera.configurePreviewOrientation(_:): Updating Preview rotation: portrait...
13:27:21.263: [info] 📸 VisionCamera.configureOutputOrientation(_:): Updating Outputs rotation: portrait...
13:27:21.263: [info] 📸 VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
13:27:21.265: [info] 📸 VisionCamera.setTargetOutputOrientation(_:): Setting target output orientation from device to device...
13:27:21.266: [info] 📸 VisionCamera.getPixelFormat(for:): Available Pixel Formats: ["420v", "420f", "BGRA", "&8v0", "&8f0", "&BGA"], finding best match... (pixelFormat="yuv", enableHdr={false}, enableBufferCompression={false})
13:27:21.266: [info] 📸 VisionCamera.getPixelFormat(for:): Using PixelFormat: 420f...
13:27:21.606: [info] 📸 VisionCamera.init(frame:session:): Preview Layer started previewing.
13:27:21.609: [info] 📸 VisionCamera.configure(_:): Beginning AudioSession configuration...
13:27:21.609: [info] 📸 VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
13:27:21.609: [info] 📸 VisionCamera.configure(_:): Beginning Location Output configuration...
13:27:21.611: [info] 📸 VisionCamera.configure(_:): Committed AudioSession configuration!
13:27:21.624: [info] 📸 VisionCamera.configure(_:): Finished Location Output configuration!
```

### Camera Device

```json
{
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "maxZoom": 120,
  "formats": [],
  "position": "back",
  "hardwareLevel": "full",
  "hasTorch": true,
  "minZoom": 1,
  "name": "Back Camera",
  "hasFlash": true,
  "minExposure": -8,
  "sensorOrientation": "portrait",
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "neutralZoom": 1,
  "supportsRawCapture": false,
  "supportsFocus": true,
  "isMultiCam": false,
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "minFocusDistance": 15
}
```

### Device

iPhone 12 Pro Max (IOS 17.5.1)

### VisionCamera Version

4.4.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

christophemenager, 2024-07-05:
I do have the same issue, only on iOS

emilje, 2024-07-13:
Same, iPhone 11.

frabanca, 2024-07-15:
Same issue on iPhone 14 Pro and _"react-native-vision-camera": "^4.5.0"_.

Marius-Adam, 2024-07-15:
Blocking use of any frame processor with text recognition @mrousavy

mrousavy, 2024-07-15:
> Blocking use of any frame processor with text recognition @mrousavy
Please stop pinging me, I am not free 24h consultancy

---

The problem is with a Frame Processor Plugin you are using, not with my library.

I explained how `Frame.orientation` works in https://github.com/mrousavy/react-native-vision-camera/pull/3077, this is now finally the last change to the `Frame.orientation` prop and should give the plugin developer all they need to make it work.

In the case of react-native-vision-camera-text-recognition, he needs to update [this `switch` statement](https://github.com/gev2002/react-native-vision-camera-text-recognition/blob/758041042d6608368fe2a41c38e3368c04f6b9c0/ios/VisionCameraTextRecognition.swift#L146-L151) with the correct values now. For example `0` is `.up` (as in the documentation) - not `.right`. With the current code it wouldn't have worked in selfie cameras or iPads.

If a Frame Processor Plugin wants to actually support rotation, they just need to add a parameter to their plugin where you can pass the current `outputOrientation` of the Camera. 

cc @gev2002

closed 2024-07-15 by mrousavy

frabanca, 2024-07-15:
Hi @mrousavy , thanks for your reply and your time!

As often happens, your comments are very useful and allow us to understand the situation and in some cases manage it.

For example, your previous comment was very useful for me, because although I use another engine for OCR (this one: https://www.npmjs.com/package/@ismaelmoreiraa/vision-camera-ocr), it It was easy to find the file to edit in the plugin source (this file https://github.com/gev2002/react-native-vision-camera-text-recognition/blob/758041042d6608368fe2a41c38e3368c04f6b9c0/ios/VisionCameraTextRecognition.swift#L146-L151 a line 142) and insert the .right value instead of the .up value.

This small change - which I'm still testing - seems to be enough to solve the problem.

Thanks again for your time.

frabanca, 2024-07-15:
Seems a good idea to give the plugin the raw value. 
Maybe - to prevent incompatibility - you can pass both so you can allow legacy plugin to work like before.

mrousavy, 2024-07-15:
Well it's both the same name unfortunately.

I could rename it to something like `bufferOrientation` and keep the old one (`orientation`)? I'll think about it, but to be fair, no frame processor plugin truly supported orientation before anyways.

frabanca, 2024-07-15:
Considering how easy is to "fix" the plugin(s) code, maybe it's better to leave the property name "orientation" and let:
1. Plugin maintainers update their code
2. Plugin utilizers update/fix the code manually (as I did in less than 3 mins after your anware)

alamothe, 2024-07-22:
We are using `@ismaelmoreiraa/vision-camera-ocr` and it has the same problem

Flocurry, 2024-07-22:
Hello @mrousavy,

I've installed react-native-vision-camera-text-recognition@3.0.4

In the file android/src/main/java/com/visioncameratextrecognition/VisionCameraTextRecognitionModule.kt, I've modified the callback function and the getFrameRotation methods

override fun callback(frame: Frame, arguments: Map<String, Any>?): HashMap<String, Any>? {
        val data = WritableNativeMap()
        val mediaImage: Image? = frame.image
        mediaImage?.let {
            val rotation = getFrameRotation(frame.orientation)
            Log.d("callback FrameProcessor", "rotation: $rotation")
            val image = InputImage.fromMediaImage(it, rotation)
            val task: Task<Text> = recognizer.process(image)
            try {
                val text: Text = Tasks.await(task)
                if (text.text.isEmpty()) {
                    return WritableNativeMap().toHashMap()
                }
                data.putString("resultText", text.text)
                data.putArray("blocks", getBlocks(text.textBlocks))
                it.close()
                return data.toHashMap()
            } catch (e: Exception) {
                it.close()
                e.printStackTrace()
                return null
            }
        }
        return null
    }

override fun callback(frame: Frame, arguments: Map<String, Any>?): HashMap<String, Any>? {
        val data = WritableNativeMap()
        val mediaImage: Image? = frame.image
        mediaImage?.let {
            val rotation = getFrameRotation(frame.orientation)
            Log.d("callback FrameProcessor", "rotation: $rotation")
            val image = InputImage.fromMediaImage(it, rotation)
            val task: Task<Text> = recognizer.process(image)
            try {
                val text: Text = Tasks.await(task)
                if (text.text.isEmpty()) {
                    return WritableNativeMap().toHashMap()
                }
                data.putString("resultText", text.text)
                data.putArray("blocks", getBlocks(text.textBlocks))
                it.close()
                return data.toHashMap()
            } catch (e: Exception) {
                it.close()
                e.printStackTrace()
                return null
            }
        }
        return null
    }

    private fun getFrameRotation(orientation: Orientation): Int {
        Log.d("FrameProcessor", "Orientation: $orientation")
        return when (orientation) {
            Orientation.PORTRAIT -> 270
            Orientation.LANDSCAPE_LEFT -> 90
            Orientation.PORTRAIT_UPSIDE_DOWN -> 180
            Orientation.LANDSCAPE_RIGHT -> 270
        }
    }

You can see logs below

Shubham0850, 2025-03-04:
Facing the same issue.

anyabreus, 2025-03-12:
The same issue. But I do not use frame processors. I can see it when I take a photo and log its orientation (on IPhone X; haven't tested it on Android device)

mingca, 2025-05-18:
@Flocurry Did you manage to solve the issue?

eduardorfreitas93, 2025-06-26:
I had the same issue on Android when using the `frameProcessor`.
The first thing I did was set `androidPreviewViewType="texture-view"` and `outputOrientation="preview"`, and I also added the code below.

```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/OrientationManager.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/OrientationManager.kt
index 5326d6b..8aadc93 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/OrientationManager.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/OrientationManager.kt
@@ -1,47 +1,98 @@
 package com.mrousavy.camera.core

+import kotlinx.coroutines.CoroutineScope
+import kotlinx.coroutines.Dispatchers
+import kotlinx.coroutines.SupervisorJob
+import kotlinx.coroutines.launch
+import kotlinx.coroutines.delay
+import kotlinx.coroutines.Job
+
+import kotlinx.coroutines.flow.StateFlow
+import kotlinx.coroutines.flow.MutableStateFlow
+import kotlinx.coroutines.flow.asStateFlow
+
 import android.content.Context
 import android.hardware.display.DisplayManager
 import android.util.Log
 import android.view.OrientationEventListener
 import android.view.Surface
+import android.os.Handler;
+import android.os.Looper;
 import com.mrousavy.camera.core.types.Orientation
 import com.mrousavy.camera.core.types.OutputOrientation

 class OrientationManager(private val context: Context, private val callback: Callback) {
   companion object {
     private const val TAG = "OrientationManager"
+    private const val DISPLAY_ROTATION_DELAY = 150L
+    private const val DEVICE_ROTATION_DELAY = 100L
   }

   private var targetOutputOrientation = OutputOrientation.DEVICE
   private var lastOutputOrientation: Orientation? = null
   private var lastPreviewOrientation: Orientation? = null

-  // Screen Orientation Listener
-  private var screenRotation = Surface.ROTATION_0
-  private val displayManager = context.getSystemService(Context.DISPLAY_SERVICE) as DisplayManager
+  private var displayRotationJob: Job? = null
+  private var deviceRotationJob: Job? = null
+
+  private val displayManager by lazy {
+    context.getSystemService(Context.DISPLAY_SERVICE) as DisplayManager
+  }
+  private val scope = CoroutineScope(Dispatchers.Main + SupervisorJob())
+
+  private val _screenRotation = MutableStateFlow(Surface.ROTATION_0)
+  val screenRotationFlow: StateFlow<Int> = _screenRotation.asStateFlow()
+
+  var screenRotation: Int
+    get() = _screenRotation.value
+    private set(value) {
+      _screenRotation.value = value
+    }
+
   private val displayListener = object : DisplayManager.DisplayListener {
     override fun onDisplayAdded(displayId: Int) = Unit
     override fun onDisplayRemoved(displayId: Int) = Unit
     override fun onDisplayChanged(displayId: Int) {
-      // Display rotated!
-      val display = displayManager.getDisplay(displayId) ?: return
-      screenRotation = display.rotation
-      maybeNotifyOrientationChanged()
+      displayRotationJob?.cancel()
+
+      displayRotationJob = scope.launch {
+        delay(DISPLAY_ROTATION_DELAY)
+        displayManager.getDisplay(displayId)?.rotation?.let { rotation ->
+          if (_screenRotation.value != rotation) {
+            Log.d(TAG, "Screen rotation changed: ${_screenRotation.value} -> $rotation")
+            _screenRotation.value = rotation
+            maybeNotifyOrientationChanged()
+          }
+        }
+      }
     }
   }

   // Physical Device Orientation listener
   private var deviceRotation = Surface.ROTATION_0
   private val orientationListener = object : OrientationEventListener(context) {
     override fun onOrientationChanged(rotationDegrees: Int) {
-      // Phone rotated!
       if (rotationDegrees == OrientationEventListener.ORIENTATION_UNKNOWN) {
-        // phone is laying flat - orientation is unknown! Avoid sending out event.
         return
       }
-      deviceRotation = degreesToSurfaceRotation(rotationDegrees)
-      maybeNotifyOrientationChanged()
+
+      val newDeviceRotation = degreesToSurfaceRotation(rotationDegrees)
+
+      if (newDeviceRotation != deviceRotation) {
+        deviceRotationJob?.cancel()
+
+        deviceRotationJob = scope.launch {
+          delay(DEVICE_ROTATION_DELAY)
+          Log.d(TAG, "Device rotation changed: $deviceRotation -> $newDeviceRotation")
+          deviceRotation = newDeviceRotation
+          maybeNotifyOrientationChanged()
+        }
+      }
     }
   }

@@ -72,6 +123,8 @@ class OrientationManager(private val context: Context, private val callback: Cal
   }

   fun stopOrientationUpdates() {
+    displayRotationJob?.cancel()
+    deviceRotationJob?.cancel()
     displayManager.unregisterDisplayListener(displayListener)
     orientationListener.disable()
   }
```

Library version: "react-native-vision-camera": "4.6.4"

Example code:

```
<Camera
        ref={refVisionCamera}
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={hasPermission && enabledCam}
        androidPreviewViewType="texture-view"
        outputOrientation="preview"
        frameProcessor={frameProcessor}
      />
```