margelo/react-native-vision-camera #3052: 🐛 Frame toArrayBuffer output is the wrong size. Missing depth

nicotroia, 2024-07-07:
### What's happening?

Running on my Samsung Galaxy S9
"react-native-vision-camera": "~4.3.2",
"react-native-worklets-core": "^1.3.3",

Trying to read the raw pixels within the following frame
```
width=640, height=480, pixelFormat=rgb, orientation=landscape-right

const buffer = frame.toArrayBuffer();
const data = new Uint8Array(buffer);
```

The data.length is `307200` which is exactly 640*480 indicating only 1 byte per pixel (grayscale?). But if the pixelFormat were actually RGB it would be 640*480*3 . 

Am I doing something wrong here? I'm not able to grab pixels from the edges of the screen because the indices keep going out of bounds. 

### Reproduceable Code

```tsx
const device = useCameraDevice(CameraType.Back);

const frameProcessor = useFrameProcessor(frame => {
  "worklet";

  try {
    runAtTargetFps(4, () => {
      const { width, height, orientation, bytesPerRow, pixelFormat } = frame;

      const buffer = frame.toArrayBuffer();
      const data = new Uint8Array(buffer);
      const bytesPerPixel = 3;

      const getPixel = (data, x, y, bytesPerRow, bytesPerPixel) => {
        const index = y * bytesPerRow + x * bytesPerPixel;

        return {
          r: data[index],
          g: data[index + 1],
          b: data[index + 2],
          a: data[index + 3],
        };
      };

        // Top-Left Corner
        const topLeftPixel = getPixel(data, 0, 0, bytesPerRow, bytesPerPixel);
        // Top-Right Corner
        const topRightPixel = getPixel(data, width - 1, 0, bytesPerRow, bytesPerPixel);
        // Bottom-Left Corner
        const bottomLeftPixel = getPixel(data, 0, height - 1, bytesPerRow, bytesPerPixel);
        // Bottom-Right Corner
        const bottomRightPixel = getPixel(data, width - 1, height - 1, bytesPerRow, bytesPerPixel);
```

### Relevant log output

```shell
LOG  Frame dimensions: width=640, height=480, bytesPerRow=2560

 LOG  Top-Left Pixel: 14, 12, 15, A=255
 LOG  Top-Right Pixel: R=37, G=32, B=29, A=255
 LOG  Index out of bounds: 1226240
 LOG  Bottom-Left Pixel: R=undefined, G=undefined, B=undefined, A=undefined
 LOG  Index out of bounds: 1228796
 LOG  Bottom-Right Pixel: R=undefined, G=undefined, B=undefined, A=undefined
```

### Camera Device

```json
{
  "formats": [],
  "hardwareLevel": "full",
  "hasFlash": true,
  "hasTorch": true,
  "id": "0",
  "isMultiCam": false,
  "maxExposure": 20,
  "maxZoom": 8,
  "minExposure": -20,
  "minFocusDistance": 0,
  "minZoom": 1,
  "name": "0 (BACK) androidx.camera.camera2",
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "position": "back",
  "sensorOrientation": "landscape-left",
  "supportsFocus": true,
  "supportsLowLightBoost": false,
  "supportsRawCapture": false
}
```

### Device

Galaxy S9

### VisionCamera Version

4.3.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

nicotroia, 2024-07-19:
@mrousavy just an update. I got my hands on a newer android (Galaxy S21) and rebuit it for that device however the same issue is occurring as on the S9

```
const frameProcessor = useFrameProcessor(frame => {
    "worklet";

    try {
      runAtTargetFps(3, () => {
        const { width, height, orientation, bytesPerRow, pixelFormat } = frame;

        const buffer = frame.toArrayBuffer();
        const data = new Uint8Array(buffer);
        
        const bytesPerPixel = 4;

        console.log(`Data length: ${data.length}`);
        console.log(`Expected length: ${width * height * bytesPerPixel}`);
        
        if (data.length !== width * height * bytesPerPixel) {
          console.error(
            "Data length does not match expected size. Check the frame format and conversion method."
          );
        }
        
      });
    } catch (err) {
      console.log("frameProcessor err", err, (err as Error).message);
    }
  }, []);
```

the data.length is always `frame.width * frame.height` which doesn't make sense. Is the frame compressed in some way? Am i doing something wrong?

Thanks for your help

mrousavy, 2024-07-19:
> Is the frame compressed in some way

No, but it might have multiple planes and `toArrayBuffer()` might only return the first plane (Y) or (R)

omedkane, 2025-02-04:
Same issue on android also

AdrianStojanovic, 2025-03-19:
Yeah same Problem. Some plane is missing for sure because the image has wrong coloring.

bfar97, 2025-06-30:
Update from my side I managed to fix this for my specific case in which my pixel format is RGBA by patching the c++ toArrayBuffer function
I changed:
`size_t size = bufferDescription.height * bufferDescription.stride;`
to
`size_t size = bufferDescription.height * bufferDescription.stride * 4;`

I think a proper fix would be to parse the frame's pixelFormat and adjust the buffer size accordingly; Leaving this info in case anyone wants to work on it. I might do it later if I find the time;

mrousavy, 2025-06-30:
interesting find