margelo/react-native-vision-camera #3054: 🔧 Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'

Nithin-Kamineni, 2024-07-08:
### How were you trying to build the app?

I got this error in my main project so I recreated the error by creating a smaller project with minimal possible libraries and code to recreate the bug/error. The project is working fine and running properly without any errors when I run the application using 
'npm react-native start' 

But the application. I started to throw this error after I did a grade clean using './gradlew clean' command on my project and this error is shown.

main-project git repo: https://github.com/Nithin-Kamineni/visionDropAIopenCV3

small-project git repo: https://github.com/Nithin-Kamineni/visioncameraIssue

The way to recreate the bug is to

1. clone the small-repo to your PC
2. Install libraries and dependencies 
3. command: `cd android` and `./gradlew clean` in 
4. `run npm react-native run` (android)

The project ran without any error before I performed './gradlew clean'.
These are the **logs** before doing './gradlew clean' (no error)

```
[VisionCamera] Thank you for using VisionCamera ??
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at C:\visioncameraIssue\node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core found, Frame Processors are enabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!
Directory 'C:\java\main' (Windows Registry) used for java installations does not exist
...
> Task :react-native-gesture-handler:configureCMakeDebug[arm64-v8a]
> Task :react-native-gesture-handler:buildCMakeDebug[arm64-v8a]
> Task :react-native-gesture-handler:configureCMakeDebug[armeabi-v7a]
> Task :react-native-gesture-handler:buildCMakeDebug[armeabi-v7a]
> Task :react-native-gesture-handler:configureCMakeDebug[x86]
> Task :react-native-gesture-handler:buildCMakeDebug[x86]
> Task :react-native-gesture-handler:configureCMakeDebug[x86_64]
> Task :react-native-gesture-handler:buildCMakeDebug[x86_64]
> Task :react-native-gesture-handler:mergeDebugJniLibFolders UP-TO-DATE
> Task :react-native-gesture-handler:mergeDebugNativeLibs UP-TO-DATE
> Task :react-native-gesture-handler:copyDebugJniLibsProjectOnly UP-TO-DATE
> Task :react-native-reanimated:mergeDebugJniLibFolders UP-TO-DATE
> Task :react-native-reanimated:mergeDebugNativeLibs UP-TO-DATE
> Task :react-native-reanimated:copyDebugJniLibsProjectOnly UP-TO-DATE
> Task :react-native-static-safe-area-insets:mergeDebugJniLibFolders UP-TO-DATE
> Task :react-native-static-safe-area-insets:mergeDebugNativeLibs NO-SOURCE
> Task :react-native-static-safe-area-insets:copyDebugJniLibsProjectOnly UP-TO-DATE
> Task :react-native-vision-camera:mergeDebugJniLibFolders UP-TO-DATE
> Task :react-native-vision-camera:mergeDebugNativeLibs UP-TO-DATE
> Task :react-native-vision-camera:copyDebugJniLibsProjectOnly UP-TO-DATE
> Task :app:mergeDebugNativeLibs UP-TO-DATE
> Task :app:stripDebugDebugSymbols UP-TO-DATE
> Task :app:validateSigningDebug UP-TO-DATE
> Task :app:writeDebugAppMetadata UP-TO-DATE
> Task :app:writeDebugSigningConfigVersions UP-TO-DATE
> Task :app:packageDebug UP-TO-DATE
> Task :app:createDebugApkListingFileRedirect UP-TO-DATE

> Task :app:installDebug
Installing APK 'app-debug.apk' on 'Pixel 7 - 13' for :app:debug
Installed on 1 device.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

For more on this, please refer to https://docs.gradle.org/8.5/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.

BUILD SUCCESSFUL in 46s
200 actionable tasks: 52 executed, 148 up-to-date
[... 443 lines cut ...]
```

### VisionCamera Version

4.4.1

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

SidhNor, 2024-07-08:
Same issue here. Occurs only in debug build on M1 processor. Release build is compiling fine.

bfar97, 2024-07-10:
Same issue here, after upgrading from react-native 0.71.3 to 0.74.3

neel-simform, 2024-07-11:
This issue mostly occurs due to frame processing being enabled. When frame processing is disabled, it works fine, but when enabled, the issue arises. Based on my understanding, this problem is related to both worklet-core and vision-camera.

neel-simform, 2024-07-12:
@mrousavy 
Just want to confirm that both the buildCMakeDebug[arm64-v8a] Failed and configureCMakeDebug[arm64-v8a] Failed errors are related to a caching issue. I need to run ./gradlew clean && rm -rf .gradle every time to get it to work.

fynally, 2024-08-21:
Am still getting this error even after cleaning the project and removing the .gradle folder

Febriansyah11, 2024-08-28:
> @mrousavy Just want to confirm that both the buildCMakeDebug[arm64-v8a] Failed and configureCMakeDebug[arm64-v8a] Failed errors are related to a caching issue. I need to run ./gradlew clean && rm -rf .gradle every time to get it to work.

thanks sir, you save my day 🙌🏻

developerdanx, 2024-08-28:
Cleaning the `gradle cache` is doing the trick. Thanks

mrousavy, 2024-08-29:
Awesome!

closed 2024-08-29 by mrousavy

jdicami, 2024-09-30:
I still get this, I have literally nuked my projected, cloned to separate folders, and even tried using a docker container to build the job. Every time linking errors with vision camera. It's the only library that causes a problem.

gpbaculio, 2024-10-24:
Follow the installation guide:
https://react-native-vision-camera.com/docs/guides

match versions here if you have similar package:
https://github.com/mrousavy/react-native-vision-camera/blob/main/package/package.json

laurensk, 2024-10-24:
> Am still getting this error even after cleaning the project and removing the .gradle folder

Same for me. I've had this issue since upgrading to react-native 0.76.0

RicardoGuiliani, 2024-10-24:
> > Am still getting this error even after cleaning the project and removing the .gradle folder
> 
> Same for me. I've had this issue since upgrading to react-native 0.76.0

Same issue here after upgrading from react-native 0.75.4 to 0.76

viatsyshyn, 2024-10-27:
> > > Am still getting this error even after cleaning the project and removing the .gradle folder
> > 
> > 
> > Same for me. I've had this issue since upgrading to react-native 0.76.0
> 
> Same issue here after upgrading from react-native 0.75.4 to 0.76

same here

maheshmuttintidev, 2024-10-29:
> > > > Am still getting this error even after cleaning the project and removing the .gradle folder
> > > 
> > > 
> > > Same for me. I've had this issue since upgrading to react-native 0.76.0
> > 
> > 
> > Same issue here after upgrading from react-native 0.75.4 to 0.76
> 
> same here

same here, please resolve asap

laurensk, 2024-10-29:
See https://github.com/mrousavy/react-native-vision-camera/issues/3260#issuecomment-2438632177

heart2friend, 2024-11-26:
> @mrousavy Just want to confirm that both the buildCMakeDebug[arm64-v8a] Failed and configureCMakeDebug[arm64-v8a] Failed errors are related to a caching issue. I need to run ./gradlew clean && rm -rf .gradle every time to get it to work.

Is there permanent solution for this?  evrytime need to run this command.

Huy-Gc, 2024-12-12:
https://github.com/Shopify/react-native-skia/issues/2692#issuecomment-2419848716
Upgrade @shopify/react-native-skia v1.5.0 will fix this

nathanfallet, 2024-12-24:
I have the same issue on my CI.
Building the first time for staging is working but then building for production fails with this error. The only difference between the builds is an environment variable.
Both are built one after another using `./gradlew clean assembleRelease`, on a clean ubuntu docker container (the same is used for both builds, but it's a new one for each version)

I'll try to reproduce clearing the cache manually to see if that works. But it's really weird that the cache makes the build fail.

camboYY, 2025-03-26:
not able to get it worked

alameen-47, 2025-05-01:
hey same issue sometimes it works sometimes it arises is there any permanent fix for this problem 
ANY EXPERTS ONLINE !!!!!!!!

kartiknadcab, 2025-05-21:
@neel-simform  thanks a lot

dev-hk22, 2025-06-13:
Try this ./gradlew clean && rm -rf .gradle
command to work always either than only ./gradlew clean it works for me.

Hemantkonda, 2025-06-13:
Try this ./gradlew clean && rm -rf .gradle
command to work always rather than only ./gradlew clean it works for me.

Gautham495, 2025-08-08:
As @BedoEbied, do not use Android Studio to build the app, if you are using bare react native cli.

Always use the command prompt to run and build the app.

nppull, 2025-12-19:
> [@mrousavy](https://github.com/mrousavy) Just want to confirm that both the buildCMakeDebug[arm64-v8a] Failed and configureCMakeDebug[arm64-v8a] Failed errors are related to a caching issue. I need to run ./gradlew clean && rm -rf .gradle every time to get it to work.

thanks bro, you save my life