margelo/react-native-vision-camera #3055: 🐛 camera freezing when pointing to a "Face"

marcosrjjunior, 2024-07-08:
### What's happening?

The camera is freezing when pointing to a face. Please check videos bellow.

[30 fps](https://www.youtube.com/watch?v=RHcQgXXmaCc)
[60 fps](https://youtu.be/JDwVI1lMBNg)

Found these issues pointing to the same error: https://github.com/mrousavy/react-native-vision-camera/issues/2313, https://github.com/mrousavy/react-native-vision-camera/pull/2329. Not sure if is direct related because I'm not even using any zoom option.

Seems like the camera is trying to perform some face identification. I'm guessing it is related to the iOS version.

**
- This issue is also happening when using a different camera library.
- I could not reproduce on the "ShadowLens 1.1.2 (17)" app.

### Reproduceable Code

```tsx
function App() {
  const device = useCameraDevice('back')

  if (device == null) return <View />

  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive
      enableZoomGesture={false}
    />
  )
}
```

### Relevant log output

```shell
output on the videos above.

[unknown/unknown: Error Domain=AVFoundationErrorDomain Code=-11819 "Cannot Complete Action" UserInfo={NSLocalizedDescription=Cannot Complete Action, NSLocalizedRecoverySuggestion=Try again later.}]
```

### Camera Device

```json
{
  "isMultiCam": false,
  "minFocusDistance": 10,
  "formats": [],
  "name": "Back Camera",
  "minExposure": -8,
  "hasFlash": false,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "neutralZoom": 1,
  "supportsFocus": true,
  "supportsRawCapture": false,
  "hasTorch": false,
  "maxZoom": 95.625,
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "sensorOrientation": "landscape-left",
  "minZoom": 1,
  "hardwareLevel": "full",
  "position": "back"
}
```

### Device

17.5.1

### VisionCamera Version

first identified on "2.16.7", also happening on 4.3.2 after update

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-07-10:
@arjunmenon1302 why are you bumping an issue on the same exact day it was created at?

mrousavy, 2024-07-10:
@marcosrjjunior sorry I cannot help you. I cannot reproduce the issue with the sample code you provided (tested on iPhone 15 Pro and iPhone 8), and if you didn't try to reproduce it in the example app then we don't know where it is coming from. Maybe you are using a Frame Processor Plugin that does something weird.

closed 2024-07-10 by mrousavy

mrousavy, 2024-07-11:
😂

mrousavy, 2024-07-11:
@marcosrjjunior AVFoundationErrorDomain code -11819 means that the daemon has crashed, I think you need to file this bug using Apple's Feedback Assistant. 

I don't think this is related to VisionCamera at all, but it doesn't hurt to also tell them that you use VisionCamera (provide a link to the github repo)

marcosrjjunior, 2024-08-08:
@mrousavy just an update on this issue.

we were able to workaround that by using [react-native-camera](https://github.com/react-native-camera/react-native-camera), with the option `autoFocus="off"`.

The issue was solved just by disabling that autoFocus prop.
To improve the experience a lit bit, we are using the `autoFocusPointOfInterest` with a "slider" to adjust the focus.

I couldn't find a way to force to disable the auto focus on the current vision camera. Thoughts?