margelo/react-native-vision-camera #3057: 🐛Frame Processor Plugin Build Failed: 'FrameHostObject.h' file not found

MatrixTurtle269, 2024-07-08:
### What's happening?

I was building my frame processor plugin (written in Objective-C++) in Xcode, but it fails with the error 'FrameHostObject.h' file not found. The issue seems to appear in `FrameProcessor.h`:

```h
// FrameProcessor.h

#pragma once

#import "Frame.h"
#import <AVFoundation/AVFoundation.h>
#import <Foundation/Foundation.h>

#ifdef __cplusplus
#import "FrameHostObject.h" >>> 'FrameHostObject.h' file not found
#import "WKTJsiWorklet.h"
#import <jsi/jsi.h>
#import <memory.h>
#endif

NS_ASSUME_NONNULL_BEGIN

// ...
```

It's really weird, since the Xcode editor seems to recognize that the file is there (and the file is physically there).

Reopening #1682 since it was marked as stale and no viable solutions were mentioned.

### Reproduceable Code

```tsx
// FrameProcessorPlugin.mm

#if __has_include(<VisionCamera/FrameProcessorPlugin.h>)
#import <opencv2/opencv.hpp>
#import <opencv2/calib3d/calib3d.hpp>

#import <Foundation/Foundation.h>
#import <CoreMedia/CoreMedia.h>
#import <CoreVideo/CoreVideo.h>

#import <VisionCamera/FrameProcessorPlugin.h> // file included here
#import <VisionCamera/Frame.h>
#import <VisionCamera/SharedArray.h>

#import "TFLTensorFlowLite.h"
```

### Relevant log output

```shell
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/user/Projects/project/ios/FrameProcessorPlugin.mm:9:
In file included from /Users/user/Projects/project/ios/Pods/Headers/Public/VisionCamera/FrameProcessorPlugin.h:12:
In file included from /Users/user/Projects/project/ios/Pods/Headers/Public/VisionCamera/FrameProcessorPluginRegistry.h:13:
In file included from /Users/user/Projects/project/ios/Pods/Headers/Public/VisionCamera/VisionCameraProxyHolder.h:10:
/Users/user/Projects/project/ios/Pods/Headers/Public/VisionCamera/FrameProcessor.h:16:9: fatal error: 'FrameHostObject.h' file not found
#import "FrameHostObject.h"
        ^~~~~~~~~~~~~~~~~~~
7 warnings and 1 error generated.
```

### Camera Device

```json
N/A
```

### Device

M2 MacBook Air

### VisionCamera Version

4.4.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-07-08 by maintenance-hans[bot]

creitz, 2025-04-16:
@MatrixTurtle269 were you able to resolve this? Seeing the same

Welinton-Hoff, 2025-08-01:
Same error

ngocthien0507, 2025-08-24:
React native 0.78.0 same error

luihugo, 2025-10-06:
+1 on react native 0.78.3